google-deepmind/mujoco #1657: Flex getting/setting position/velocity of each vertex

rpapallas, 2024-05-13:
Hello,

We would like to simulate a deformable object using the new “flex” type and need the following from MuJoCo:

1. Get the position and velocity of each vertex in the deformable object.
2. Set the position and velocity of each vertex in the deformable object.

I scanned through the docs and found some details about how this could be done, and I have something; however, there is a conflict which makes me think that my solution is incorrect.

Here is a model based on your Jelly Flex example:

```xml
<mujoco model="Jelly">
    <include file="scene.xml"/>

    <extension>
        <plugin plugin="mujoco.elasticity.solid"/>
        </extension>

    <option solver="CG" tolerance="1e-6" timestep=".001" integrator="implicitfast"/>

    <size memory="10M"/>

    <visual>
        <map stiffness="100"/>
        </visual>

    <default>
        <default class="wall">
            <geom type="plane" size=".5 .5 .05"/>
            </default>
        </default>

    <worldbody>
        <body mocap="true" pos="-.1 .05 0" zaxis=".5 0 1">
            <geom type="capsule" size=".1 .1" group="1" condim="1"/>
        </body>
        <geom name="+x" class="wall" zaxis="1 0 0"  pos="-.5 0 -.25"/>
        <geom name="-x" class="wall" zaxis="-1 0 0" pos=".5 0 -.25"/>
        <geom name="+y" class="wall" zaxis="0 1 0"  pos="0 -.5 -.25"/>
        <geom name="-y" class="wall" zaxis="0 -1 0" pos="0 .5 -.25"/>

        <flexcomp name="Jelly1" type="grid" count="8 8 8" spacing=".07 .07 .07" pos="0 0 1" dim="3" radius=".005" rgba="0 .7 .7 1" mass="5" >
            <edge damping="1"/>
            <contact selfcollide="none"/>
            <plugin plugin="mujoco.elasticity.solid">
                <config key="poisson" value="0"/>
                <!--Units are in Pa (SI)-->
                <config key="young" value="5e4"/>
                </plugin>
        </flexcomp>

        <flexcomp name="Jelly2" type="grid" count="8 8 8" spacing=".07 .07 .07" pos="0.7 0 1" dim="3" radius=".005" rgba="0 .7 .7 1" mass="5" >
            <edge damping="1"/>
            <contact selfcollide="none"/>
            <plugin plugin="mujoco.elasticity.solid">
                <config key="poisson" value="0"/>
                <!--Units are in Pa (SI)-->
                <config key="young" value="5e4"/>
                </plugin>
        </flexcomp>

        </worldbody>
    </mujoco>
```

Here is some code that tries to make sense of how to get the position of a specific vertex:

```cpp
    int flexId = mj_name2id(m, mjOBJ_FLEX, "Jelly2");
    int firstVertadr = m->flex_vertadr[flexId];
    int bodyId = m->flex_vertbodyid[firstVertadr];

    // If flex_vertbodyid gives the ID of that vertex as a normal MuJoCo body then this should work:
    float x = d->xpos[bodyId + 0];
    float y = d->xpos[bodyId + 1];
    float z = d->xpos[bodyId + 2];
    std::printf("Vertex %d xpos x=%f, y=%f, z=%f\n", 0, x, y, z);

   // However, the docs mention "flexvert_xpos" which, if my assumption above is true, it should be the same with xpos.
    x = d->flexvert_xpos[firstVertadr + 0];
    y = d->flexvert_xpos[firstVertadr + 1];
    z = d->flexvert_xpos[firstVertadr + 2];
    std::printf("Vertex %d xpos x=%f, y=%f, z=%f\n", 0, x, y, z);
```

I would like to get/set the position and velocity of the first vertex of “Jelly2”. My understanding is that `m->flex_vertbodyid[SOME_VERTEXID]` would map the flex vertex to a MuJoCo body? If that's the case, then I can treat that vertex as a normal MuJoCo body to set/get the position/velocity as any other rigid-body in MuJoCo.

However, I see in the documentation the `flexvert_xpos` and I tried to compare it with `xpos` (assuming `flex_vertbodyid` gives me a body id of that vertex), however the output is different, which makes me think that the `flex_vertbodyid` isn't really giving you a body ID corresponding to that vertex?

Although the documentation lists, `flexvert_xpos` I can't find how I could get the velocity and how to set the position and the velocity of that vertex similarly as `flexvert_xpos`.

rpapallas, 2024-05-14:
Sorry, it was an error on my part, I managed to figure this out. 

Still curious to know why there is `flexvert_xpos` when you can get the flex vertex xpos from `xpos` through `flex_vertbodyid`.

dblanm, 2024-05-15:
> Sorry, it was an error on my part, I managed to figure this out.
> 
> Still curious to know why there is `flexvert_xpos` when you can get the flex vertex xpos from `xpos` through `flex_vertbodyid`.

Hey! Would you mind sharing how you solved this?

rpapallas, 2024-05-17:
@dblanm 
It appears that my assumption is correct: the `flex_vertbodyid` maps a flex vertex to a mujoco body.

Here is a demonstration code:

Getting and setting all positions of the flex vertices:

```cpp
std::vector<std::tuple<float, float, float>> getFlexPositions(std::string flexName) {
    std::vector<std::tuple<float, float, float>> positions;

    int flexId = getFlexId(flexName);
    int firstVertadr = m->flex_vertadr[flexId];
    int numVertices = m->flex_vertnum[flexId];

    for(int i=0; i < numVertices; i++) {
        int bodyId = m->flex_vertbodyid[firstVertadr + i];
        float x = d->xpos[(bodyId * 3) + 0];
        float y = d->xpos[(bodyId * 3) + 1];
        float z = d->xpos[(bodyId * 3) + 2];

        /* // or the same with:
        x = d->flexvert_xpos[firstVertadr + i * 3 + 0];
        y = d->flexvert_xpos[firstVertadr + i * 3 + 1];
        z = d->flexvert_xpos[firstVertadr + i * 3 + 2];
        */
        positions.push_back(std::make_tuple(x, y, z));
    }

    return positions;
}

void setFlexPositions(std::string flexName, std::vector<std::tuple<float, float, float>> positions) {
    int flexId = getFlexId(flexName);
    int firstVertadr = m->flex_vertadr[flexId];
    int numVertices = m->flex_vertnum[flexId];

    for(int i=0; i < numVertices; i++) {
        int bodyId = m->flex_vertbodyid[firstVertadr + i];
        const int jointIndex = m->body_jntadr[bodyId];
        const int qposIndex = m->jnt_qposadr[jointIndex];
        float x = std::get<0>(positions[i]);
        float y = std::get<1>(positions[i]);
        float z = std::get<2>(positions[i]);
        d->qpos[qposIndex + 0] = x;
        d->qpos[qposIndex + 1] = y;
        d->qpos[qposIndex + 2] = z;
    }
}
```

Getting and setting the flex vertex velocities:

```cpp
std::vector<std::tuple<float, float, float>> getFlexVelocities(std::string flexName) {
    std::vector<std::tuple<float, float, float>> velocities;

    int flexId = getFlexId(flexName);
    int firstVertadr = m->flex_vertadr[flexId];
    int numVertices = m->flex_vertnum[flexId];

    for(int i=0; i < numVertices; i++) {
        int bodyId = m->flex_vertbodyid[firstVertadr + i];
        const int jointIndex = m->body_jntadr[bodyId];
        const int start = m->jnt_dofadr[jointIndex];
        double linearX = d->qvel[start + 0];
        double linearY = d->qvel[start + 1];
        double linearZ = d->qvel[start + 2];
        velocities.push_back(std::make_tuple(linearX, linearY, linearZ));
    }

    return velocities;
}

void setFlexVelocities(std::string flexName, std::vector<std::tuple<float, float, float>> velocities) {
    int flexId = getFlexId(flexName);
    int firstVertadr = m->flex_vertadr[flexId];
    int numVertices = m->flex_vertnum[flexId];

    for(int i=0; i < numVertices; i++) {
        int bodyId = m->flex_vertbodyid[firstVertadr + i];
        const int jointIndex = m->body_jntadr[bodyId];
        const int start = m->jnt_dofadr[jointIndex];
        float x = std::get<0>(velocities[i]);
        float y = std::get<1>(velocities[i]);
        float z = std::get<2>(velocities[i]);
        d->qvel[start + 0] = x;
        d->qvel[start + 1] = y;
        d->qvel[start + 2] = z;
    }
}
```

closed 2024-08-30 by kevinzakka