google-deepmind/mujoco #1665: Jittery/unstable contacts for a simple STL file of a box resting on a plane?

kurtenkera, 2024-05-15:
Hi,

I've created a simple STL file, `cqbox.stl`, of a box using [cadquery](https://cadquery.readthedocs.io/en/latest/). Below is the Python code that I used to create this box:

```
import cadquery as cq
from cadquery import exporters

result = cq.Workplane("front").box(1.0, 1.0, 1.0)
exporters.export(result, "cqbox.stl")
```

The simulation below shows that even when this box rests on a plane, two of the contact points are jittery/unstable? Why is this? In the video, I activate wireframe mode and toggle the convex hull off and on, and there doesn't appear to be anything majorly bad about the convex hull that suggests why these contact issues are occurring? I just want to clear up this issue as I need a reliable analysis of contacts for my project, and if this simple example is unstable/unreliable - I'm anticipating issues for more complicated STL files?

[box_stl_contacts.webm](https://github.com/google-deepmind/mujoco/assets/127472392/c34c2afd-fb95-4723-89af-67a4448aa24a)

Below is the xml file that I used for this simple example:

```
<mujoco>
 
  <compiler meshdir = "../mesh_files/" />

  <asset>
    <mesh file="cqbox.stl"/>
  </asset>
  
  <option gravity = "0 0 -9.8" />

  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
    <geom type="plane" size="5 5 0.1" rgba="1 0 1 1"/>
    <body pos="0 0 3" euler = "0 0 0">
      <joint type="free"/>
      <geom type="mesh" name="cqbox" mesh="cqbox" rgba = "0 1 0 1"/>
    </body>
  </worldbody>

</mujoco>
```

kurtenkera, 2024-05-16:
As a further addition to this issue, below is a video of a `cylinder` primitive `geom` with 3 stable contact points (these points remain stable over time):

[stable_cylinder_contacts.webm](https://github.com/google-deepmind/mujoco/assets/127472392/9bc2c4da-49e3-4861-96bf-8518d795e270)

However, when I rest the cylinder on top of a `box` geom, the contact points are unstable (see below)? This is worrying for me, as these experiments are all performed with MuJoCo's primitive `geom` objects (not with STL files). I'm worried that if contact forces are unstable for these native MuJoCo objects, there will be equal or increased unreliability for more complex geometries?

[unstable_cylinder_contacts.webm](https://github.com/google-deepmind/mujoco/assets/127472392/d92c2c5b-bb74-48c0-8ca5-e544f3e20bbb)

yuvaltassa, 2024-05-16:
1. This is working as intended. Jittery objects are a problem, jittery contact points are not. Mesh-plane collisions generate 3 contacts which is all you need for stability. Since there are 4 corners, they are bound to jump around.
2. If this really bothers you (though it shouldn't), try [multiccd](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-multiccd).

closed 2024-05-16 by yuvaltassa