google-deepmind/mujoco #1666: Actuator force visualization

alexhansson, 2024-05-15:
Hi,

I'm a student and I'm trying to use MuJoCo for the development of a simulation environment for space systems.

I'm trying to visualize the force each thruster of my space system generates. I know there is the built-in function which illustrates it by coloring the actuator geoms according to the thrust they are generating. Unfortunately in my case this is not clear enough. Is there a way to visualize the thrust with an arrow (scaled and colored according to thrust)?

It would probably be possible to do this with `mjv_initGeom`, but there is some overhead with the position and rotation of the arrows, which is why a built-in method is preferred.

Thank you for your help!

yuvaltassa, 2024-05-20:
This is possible, but I'm not sure we'll get to it. Are you using pure thrusters (site actuators with no `refsite`)?

Side note: I hope you know that the [size](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-scale-actuatorlength) and [color](https://mujoco.readthedocs.io/en/latest/XMLreference.html#visual-rgba-actuatornegative) of actuator visualization is modifiable? Perhaps this is good enough for you use case?

alexhansson, 2024-05-20:
Thanks for your answer @yuvaltassa.
I'm using thrusters, however with a `site` argument which specifies the position of each thruster on the body.  
I will investigate whether these two are enough. As of right now, the actuators are visualized as a small sphere which get re-colored according to the applied thrust. Sadly, you can barely see it, which is why a scalable arrow would be super nice.

yuvaltassa, 2024-05-20:
Oh that's just the site where you apply the force. Make the site bigger (it can be a cylinder if you want it to stick out).

But making an arrow is not hard, I'll try to add that at some point.

alexhansson, 2024-05-21:
Thank you.
Is the site's (cylinder's) size modifiable at runtime?

yuvaltassa, 2024-05-21:
Sure, that would have no side effects.