google-deepmind/mujoco #1667: mj_forward modifies jacobian when called after mj_step

wualbert, 2024-05-15:
# Summary
Calling `mj_forward` after a `mj_step` call unexpectedly modifies the Jacobian returned by `mj_jacBody`. By extension, other derived quantities may be affected as well. The generalized coordinates `qpos`, `qvel` are not modified by the call, which is expected.

# Expected behavior
According to the documentation, [`mj_forward` performs the same computations as `mj_step` but without the integration](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#main-simulation). The Jacobian, which only depends on `qpos`, should not change after calling `mj_forward` since `qpos` is the same and the Jacobian should have already been computed by `mj_step`.

# Code example
The following examples showcases how `mj_forward` changes the Jacobian. The code was tested on a M1 mac with macOS 14.5, Python 3.10.14, Mujoco 3.1.5.

```python
# Adapted from https://github.com/tayalmanan28/MuJoCo-Tutorial
import mujoco as mj
import numpy as np
from mujoco.glfw import glfw
import sys

xml = """
<mujoco>
	<option gravity="0 0 -9.81">
	</option>

	<worldbody>
		<body pos="0 0 2" euler="0 180 0">
			<joint name="pin" type="hinge" axis="0 -1 0" pos="0 0 0.5"/>
			<geom type="cylinder" size=".05 .5" rgba="0 .9 0 1" mass="1"/>
		</body>
	</worldbody>
	
	<actuator>
		<motor joint="pin" name="torque" gear="1" ctrllimited="true" ctrlrange="-100 100"/>
		<position name="position_servo" joint="pin" kp="10"/>
		<velocity name="velocity_servo" joint="pin" kv="0"/>
	</actuator>
</mujoco>
"""

def get_jac(model, data):
    jacp = np.zeros((3, model.nv))
    mj.mj_jacBody(model, data, jacp, None, 1)
    return jacp

def main(test_jacobian):
    model = mj.MjModel.from_xml_string(xml)
    data = mj.MjData(model)  # MuJoCo data
    cam = mj.MjvCamera()  # Abstract camera
    opt = mj.MjvOption()
    data.qpos[0] = np.pi / 2
    # Set camera configuration
    cam.azimuth = 90.0
    cam.distance = 5.0
    cam.lookat = np.array([0.012768, -0.000000, 1.254336])
    # Init GLFW, create window, make OpenGL context current, request v-sync
    glfw.init()
    window = glfw.create_window(1200, 900, "Demo", None, None)
    glfw.make_context_current(window)
    glfw.swap_interval(1)

    scene = mj.MjvScene(model, maxgeom=10000)
    context = mj.MjrContext(model, mj.mjtFontScale.mjFONTSCALE_150.value)
    while not glfw.window_should_close(window):
        mj.mj_step(model, data)
        # Compute the jacobian
        jac0 = get_jac(model, data)
        q0 = np.copy(data.qpos)
        v0 = np.copy(data.qvel)
        # Call mj_forward
        mj.mj_forward(model, data)
        if test_jacobian:
            # This test fails
            np.testing.assert_allclose(jac0, get_jac(model, data), atol=1e-4)
        np.testing.assert_allclose(q0, data.qpos, atol=1e-7)
        np.testing.assert_allclose(v0, data.qvel, atol=1e-7)
        viewport_width, viewport_height = glfw.get_framebuffer_size(window)
        viewport = mj.MjrRect(0, 0, viewport_width, viewport_height)
        # Update scene and render
        mj.mjv_updateScene(
            model,
            data,
            opt,
            None,
            cam,
            mj.mjtCatBit.mjCAT_ALL.value,
            scene,
        )
        mj.mjr_render(viewport, scene, context)
        glfw.swap_buffers(window)
        glfw.poll_events()

    glfw.terminate()

if __name__ == "__main__":
    test_jacobian = "--test-jacobian" in sys.argv
    if test_jacobian:
        print("Jacobian testing is enabled.")
    else:
        print("Jacobian testing is not enabled.")
    main(test_jacobian)
```

yuvaltassa, 2024-05-18:
This is working as expected, not sure why you are confused. After mj_step, everything in mjData **except for the state** is wrt **the previous timestep**. So after an mj_step there is an inconsistency between the state and everything else, since mj_step only does everything until it computes the new state and then it stops. After mj_forward all the quantities are wrt the current timestep, there is no inconsistency.

Let me know if something is still unclear.

closed 2024-05-18 by yuvaltassa

yuvaltassa, 2024-05-18:
Actually, this is probably a good opportunity to improve the documentation. If you now fully get it, tell me how you think we should improve the docs. If not, please keep asking questions until you are 100% confident you get it, and this conversation will help clarify how the docs should be improved...

wualbert, 2024-05-20:
@yuvaltassa Thank you for the reply. I understand the behavior now.

After spending more time reading the documentation, what you described is indeed detailed in [Programming/Simulation loop](https://mujoco.readthedocs.io/en/stable/programming/simulation.html#simulation-loop) section. 
> ... Jacobians, passive forces. None of these quantities are available before mj_step is called (or rather, they are available but outdated by one time step)

However, the documentation in [APIreference/APIfunctions/Main simulation](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#main-simulation) makes no mention of this explicitly. I also find the statement "`mj_step` is the only call needed" to be misleading. As far as I can tell most robotics use cases must call `mj_forward` separately to get forward kinematics results.
> Most users will only need to call [mj_step](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-step), which computes everything and advanced the simulation state by one time step...mj_forward performs the same computations as [mj_step](https://mujoco.readthedocs.io/en/stable/APIreference/APIfunctions.html#mj-step) but without the integration. It is useful after loading or resetting a model (to put the entire mjData in a valid state), and also for out-of-order computations that involve sampling or finite-difference approximations.

I think adding more emphasis on _all quantities except for the state itself is outdated after `mj_step`_ would be helpful, especially in the `APIreference` section.

commit 0d1c92b2 (2024-05-30): Add documentation on pipeline stages and mjData self-consistency. Fixes #1667

PiperOrigin-RevId: 638619787
Change-Id: Ic8dd0fbd178c4946757a07d74fca3b2c905f9080