google-deepmind/mujoco #1669: I have a question about the friction force experiment.

2-woo-10, 2024-05-16:
Hello.

I am conducting a test to check the sensor data provided by Mujoko.

What I want to do at the moment is to change the contact parameters and find the optimal contact parameters.

To this end, we are conducting the following experiment.

- This is an experiment to check the accelerometer, velocimeter, attached to the rectangular parallelepiped when a force is applied to the rectangular parallelepiped with xfrc_applied.

It is an experiment in which an object moves when a force greater than the friction force is applied by increasing the magnitude of the applied force.

In my opinion, if you apply about 10N to a rectangular parallelepiped with 1kg mass in the x-axis direction, the value will not be measured in the accelerometer.

This is because vertical drag x friction coefficient (coefficient of friction is 1) = approximately 10N.

However, when you apply force, the object appears to rotate without maintaining contact with the floor.
Also, when I look at the data on the sensor, it shows 10m/s^2, can you explain this?

Also, why does an object move more and more when you apply a force to 10N, and the object rolls around?

The following is an image of sensor data.
 d->xfrc_applied[(mj_name2id(m, 5, "object")*6)] = 10;

[image]

<details>
  <summary>minimal XML</summary>

<mujoco model="panda hand">
  <compiler angle="radian" meshdir="assets" autolimits="true"/>

  <option timestep="0.001"  cone="elliptic">
    <flag actuation="enable" gravity="enable" warmstart="enable" multiccd="enable"/>
  </option>

  <asset>

    
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
    markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="120" elevation="-20"/>
  </visual>

  <worldbody>

    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>

    <body name="object" pos="0.0 0.0 0.05">
      <joint type="free"/>
      <geom name="object" type="box" mass="1" size="0.20 0.20 0.050" rgba="0. 0 0. 1" condim="1" friction="1 0.005 0.0001" priority="1"/>
      <site name="sensor" type="box" size="0.00001 0.00001 0.00001 " />
    </body> 

  </worldbody>

  <sensor>
   <framepos name="object_pos" objtype="site" objname="sensor"/>
   <velocimeter name="vel_sensor" site="sensor"/>
   <accelerometer  name="acc_sensor" site="sensor"/>
   <force name="force_sensor" site="sensor"/>
  </sensor>

  <contact>
    <pair geom1="floor" geom2="object"/>
  </contact>
</mujoco>

</details>

Here is a screenshot / video, illustrating my question:

2-woo-10, 2024-05-16:
The second graph came out incorrectly, and a force is currently being applied to 10N in the x-axis direction.

yuvaltassa, 2024-05-21:
Can you post a video of what's happening in the scene?