google-deepmind/mujoco #1672: Camera rendering in MuJoCo - How to take a picture from a simulated camera and still keeping the GUI active

vmstavens, 2024-05-16:
Hello there,

I'm a research assistant and I'm trying to take pictures of my MuJoCo simulation using a simulated camera.

Currently, I have my main rendering loop running: 
<details>

  <summary>main render loop</summary>

```python

def run(self) -> None:
  """
  Method for running the MuJoCo viewer main loop.
  
  Returns:
  - None.
  """
  with mj.viewer.launch_passive(model=self._model, data=self._data, key_callback=self.keyboard_callback) as viewer:
  
	  # load initial state
	  load_mj_state(self._model, self._data, "home", self._lock)
	  mj.mj_step(self._model, self._data)
  
	  # Toggle site frame visualization.
	  if self._args.show_site_frames:
		  viewer.opt.frame = mj.mjtFrame.mjFRAME_SITE
  
	  while viewer.is_running():
		  step_start = time.time()
  
		  # Step the simulation.
		  mj.mj_step(self._model, self._data)
  
		  viewer.sync()
		  time_until_next_step = self.dt - (time.time() - step_start)
		  if time_until_next_step > 0:
			  time.sleep(time_until_next_step)
``` 

</details>

while I have my `keyboard_callback` function defined as

<details>

  <summary>`keyboard_callback`</summary>

```python

def keyboard_callback(self, key) -> None:
  """
  Method for the keyboard callback. This method should not be called explicitly.
  
  Parameters:
  - key: Keyboard input.
  
  Returns:
  - None.
  """
  if key == glfw.KEY_COMMA:
	  print(self.cam.image)
``` 

</details>
  
When I call the `cam.image` attribute, the following function is run

<details>
  <summary>get image attribute</summary>

```python
@property
def image(self) -> np.ndarray:
  """Return the captured RGB image."""
  self._renderer.update_scene(self._data, camera=self.name)
  self._img = self._renderer.render()
  self._renderer.update_scene(self._data)
  return self._img
```

</details>

The `self._renderer` is here a `mj.Renderer` object defined by `self._renderer = mj.Renderer(self._model, self._height, self._width)` where `self._model` is the model object loaded from the scene file.

After I call the image attribute, the GUI becomes unresponsive while the main rendering loop still seems to be running just fine. Ideally, I would want to control the GUI after i have rendered a picture from the camera.

All feedback and/or ideas for how I could progress/solve this problem of mine are very much appreciated!

qgallouedec, 2024-06-08:
Same issue.

vmstavens, 2024-06-15:
Hi @qgallouedec

I believe I found a solution, so to help out I have written the following that you hopefully can use to solve your problem :)

1. define the camera in the scene file

<details>
  <summary>Add camera to scene</summary>

```xml
	...
		<camera
		  	name="cam"
		  	pos="1 1 1"
			mode="targetbody"
			target="<insert name of a body>"
		/>
	</worldbody>
</mujoco>
```
</details>

2. Define camera class, here you can just name mine. In this class, I am using two utility functions: One for making a spatialmath SE3 homogeneous transformation matrices called `make_tf` and one for writing point clouds `pcwrite`
<details>
  <summary>Here is my camera class</summary>

```python
class Camera:
    def __init__(self, args, model, data, cam_name: str = "", save_dir="data/img/"):
        """Initialize Camera instance.

        Args:
        - args: Arguments containing camera width and height.
        - model: Mujoco model.
        - data: Mujoco data.
        - cam_name: Name of the camera.
        - save_dir: Directory to save captured images.
        """
        self._args = args
        self._cam_name = cam_name
        self._model = model
        self._data = data
        self._save_dir = save_dir + self._cam_name + "/"

        self._width = self._args.cam_width
        self._height = self._args.cam_height
        self._cam_id = self._data.cam(self._cam_name).id

        self._renderer = mj.Renderer(self._model, self._height, self._width)
        self._camera = mj.MjvCamera()
        self._scene = mj.MjvScene(self._model, maxgeom=10_000)

        self._image = np.zeros((self._height, self._width, 3), dtype=np.uint8)
        self._depth_image = np.zeros((self._height, self._width, 1), dtype=np.float32)
        self._seg_id_image = np.zeros((self._height, self._width, 3), dtype=np.float32)
        self._point_cloud = np.zeros((self._height, self._width, 1), dtype=np.float32)

        if not os.path.exists(self._save_dir):
            os.makedirs(self._save_dir)

    @property
    def height(self) -> int:
        """
        Get the height of the camera.

        Returns:
                int: The height of the camera.
        """
        return self._height

[... 394 lines cut ...]

		with mj.viewer.launch_passive(
			model=self._model,
			data=self._data,
			key_callback=lambda key: key_queue.put(key),
		) as viewer:
			self.cam._renderer.render()

			while viewer.is_running():
				step_start = time.time()

				while not key_queue.empty():
					self.keyboard_callback(key_queue.get())

				# step the simulation.
				mj.mj_step(self._model, self._data)
				viewer.sync()

				time_until_next_step = self.dt - (time.time() - step_start)
				if time_until_next_step > 0:
					time.sleep(time_until_next_step)
```
</details>

I hope this can assist you in solving your problem :)

closed 2024-08-30 by kevinzakka

Fima2003, 2025-01-06:
Hey @vmstavens where do you take pcwrite function? I do not see it in your utils

vmstavens, 2025-01-12:
Oh yeah, my bad, thanks for pointing that out @Fima2003! I hope this helps

<details>
    <summary>pcwrite</summary>
    ``` python

    from pathlib import Path
    from typing import Union
    
    import numpy as np
    import open3d as o3d

    def pcwrite(
        filename: str, point_cloud: Union[np.ndarray, list, o3d.geometry.PointCloud]
    ) -> None:
        """
        Write a point cloud to a PCD file.
    
        Parameters
        ----------
        filename : str
            File path where the PCD file will be saved.
        point_cloud : Union[np.ndarray, list, o3d.geometry.PointCloud]
            The point cloud data to write. It can be:
            - A numpy array representing the point cloud (nx3).
            - A list of points where each point is a list or tuple of 3 coordinates.
            - An Open3D point cloud object.
    
        Returns
        ----------
        None
    
        Raises
        ----------
        ValueError
            If the provided point cloud type is unsupported or the file cannot be written.
        """
        try:
            if isinstance(point_cloud, np.ndarray):
                # Convert numpy array to Open3D point cloud
                pcd = o3d.geometry.PointCloud()
                pcd.points = o3d.utility.Vector3dVector(point_cloud)
            elif isinstance(point_cloud, list):
                # Convert list to Open3D point cloud
                pcd = o3d.geometry.PointCloud()
                pcd.points = o3d.utility.Vector3dVector(np.array(point_cloud))
            elif isinstance(point_cloud, o3d.geometry.PointCloud):
                # Point cloud is already an Open3D point cloud object
                pcd = point_cloud
            else:
                raise ValueError("Unsupported point cloud type")
    
            path = Path(filename)
    
            # Create all parent directories if they don't exist
            path.parent.mkdir(parents=True, exist_ok=True)
    
            # Check if the file exists
            if path.exists():
                print(f"Open3D: Overwriting {filename}.")
            else:
                print(f"Open3D: Creating {filename}.")
    
            # Write the point cloud to a PCD file
            o3d.io.write_point_cloud(filename, pcd)
        except Exception as e:
            raise ValueError(f"Error writing point cloud to '{filename}': {e}")
    ```
</details>