google-deepmind/mujoco #1682: Support for Rendering with MJX for Enhanced Training Pipelines

markusheimerl, 2024-05-21:
```
import jax
import time
import mujoco
import functools
from brax import envs
from mujoco import mjx
from etils import epath
from jax import numpy as jp
from brax.io import mjcf, model
from brax.envs.base import PipelineEnv, State
from brax.training.agents.ppo import train as ppo
from jax import device_get

class Humanoid(PipelineEnv):

    def __init__(self, **kwargs):
        mj_model = mujoco.MjModel.from_xml_path((epath.Path(epath.resource_path("mujoco")) / ("mjx/test_data/humanoid") / "humanoid.xml").as_posix())
        mj_model.opt.solver = mujoco.mjtSolver.mjSOL_CG
        mj_model.opt.iterations = 6
        mj_model.opt.ls_iterations = 6

        sys = mjcf.load_model(mj_model)

        physics_steps_per_control_step = 5
        kwargs["n_frames"] = kwargs.get("n_frames", physics_steps_per_control_step)
        kwargs["backend"] = "mjx"

        super().__init__(sys, **kwargs)

    def reset(self, rng: jp.ndarray) -> State:
        """Resets the environment to an initial state."""
        rng, rng1, rng2 = jax.random.split(rng, 3)

        low, hi = -1e-2, 1e-2 # reset_noise_scale
        qpos = self.sys.qpos0 + jax.random.uniform(rng1, (self.sys.nq,), minval=low, maxval=hi)
        qvel = jax.random.uniform(rng2, (self.sys.nv,), minval=low, maxval=hi)

        data = self.pipeline_init(qpos, qvel)

        obs = self._get_obs(data, jp.zeros(self.sys.nu), None)
        reward, done, zero = jp.zeros(3)
        metrics = {
            "forward_reward": zero, "reward_linvel": zero, "reward_quadctrl": zero, "reward_alive": zero,
            "x_position": zero, "y_position": zero,"distance_from_origin": zero,"x_velocity": zero,"y_velocity": zero
        }
        return State(data, obs, reward, done, metrics)

    def step(self, state: State, action: jp.ndarray) -> State:
        """Runs one timestep of the environment's dynamics."""
        data0 = state.pipeline_state
        data = self.pipeline_step(data0, action)

        com_before = data0.subtree_com[1] # center of mass
        com_after = data.subtree_com[1]
        velocity = (com_after - com_before) / self.dt
        forward_reward = 1.25 * velocity[0]

        min_z, max_z = (1.0, 2.0) # healthy_z_range
        is_healthy = jp.where(data.q[2] < min_z, 0.0, 1.0)
        is_healthy = jp.where(data.q[2] > max_z, 0.0, is_healthy)
        healthy_reward = 5.0
        ctrl_cost = (0.1 * jp.sum(jp.square(action)))

        state.metrics.update(
            forward_reward=forward_reward, reward_linvel=forward_reward, reward_quadctrl=-ctrl_cost, 
            reward_alive=healthy_reward, x_position=com_after[0], y_position=com_after[1], 
            distance_from_origin=jp.linalg.norm(com_after), x_velocity=velocity[0], y_velocity=velocity[1]
        )
        return state.replace(
[... 103 lines cut ...]
When calling the `render` method within the custom `Humanoid` class, an error is thrown:

`jax.errors.TracerArrayConversionError: The numpy.ndarray conversion method __array__() was called on traced array with shape float32[28]`.

This error indicates that the conversion method is called on a traced array, which depends on the value of the argument `state.pipeline_state.q`. The current implementation of the MuJoCo renderer in mjx does not handle the conversion of traced arrays properly in this context.

**Importance of Camera Input in Reinforcement Learning:**
Using camera input is crucial when training robots with reinforcement learning. In real-world scenarios, robots rely on visual information captured by their cameras to perceive and interact with the environment. By incorporating camera pixels as part of the observation space during training, the learned policies can be more robust and adaptable to real-world conditions.

**Proposed Solution:**
To enable effective training with camera input using mjx, it is essential to address the compatibility issue between the MuJoCo renderer and JAX traced arrays. Possible solutions include:
1. Enhancing the mjx renderer to handle JAX traced arrays directly, ensuring seamless integration with the Brax environment.
2. Providing alternative rendering methods or workarounds specifically designed for mjx, allowing the retrieval of camera images during training.
3. Improving documentation and examples that demonstrate the proper usage of camera input with mjx and provide guidance on handling traced arrays in this context.

**Alternatives Considered:**
- Preprocessing the camera images outside of the JAX computation graph, but this approach may introduce additional overhead and complexity to the training pipeline.
- Using alternative rendering libraries or techniques, but this may limit the flexibility and compatibility with the existing MuJoCo and Brax ecosystem.

**Additional Context:**
Integrating camera input into reinforcement learning algorithms is crucial for developing intelligent and adaptable robots. By leveraging the power of JAX and mjx, researchers and developers can accelerate the training process and build more sophisticated models. However, the current compatibility issue between the MuJoCo renderer and JAX traced arrays hinders the effective utilization of camera input in this setup.

Addressing this issue and providing a seamless integration between the MuJoCo renderer and mjx will greatly benefit the robotics and reinforcement learning community. It will enable researchers to train models that can effectively process visual information, leading to more advanced and capable robots.

Thank you for considering this issue. Your support in resolving the compatibility problem and enhancing the usability of camera input with mjx will contribute to the advancement of robotics research and real-world applications.

yuvaltassa, 2024-05-21:
You are 100% right. 
We are working on option 2.

markusheimerl, 2024-05-21:
@yotam you have an estimate of when this might be done?

markusheimerl, 2024-05-22:
@yuvaltassa I think what I wanted to ask: Would you mind making the corresponding feature branch public in this repo? Id love to contribute.

yuvaltassa, 2024-05-22:
The moment we have something that works it will be OSS and we would love you to contribute!

@erikfrey is leading this effort, perhaps there is something he'd like to add.

erikfrey, 2024-05-22:
Hello!  Please see #1604 and #1485 for related discussions.  You can do visual observations today using `mjx.ray`, although this only works for toy environments.

We are working on integrating [Madrona](https://madrona-engine.github.io/) as a means for high throughput tiled rendering on GPU, but this is still very much a work in progress.  We'll share more once we have a good proof of concept - no ETA but this is actively under development.

markusheimerl, 2024-05-25:
@yuvaltassa @erikfrey thanks for your input! Have you considered https://github.com/JoeyTeng/jaxrenderer ?

``` python
import re
import jax
import numpy as onp
from PIL import Image
from jax import numpy as jp
from renderer import Model
from renderer import ModelObject
from renderer import LightParameters
from renderer.geometry import rotation_matrix
from renderer import CameraParameters
from renderer import ShadowParameters
from renderer import Renderer, transpose_for_display
from numpngw import write_apng

# Load model and textures
obj_path, texture_path, spec_path = "african_head.obj", "african_head_diffuse.tga", "african_head_spec.tga"
image = Image.open(texture_path)
width, height = image.size
texture = onp.zeros((width, height, 3))
for y in range(height):
    for x in range(width):
        texture[y, x] = onp.array(image.getpixel((x, y)))
texture = jp.array(texture, dtype=jp.single) / 255

image = Image.open(spec_path)
specular_map = onp.zeros((width, height, 3))
for y in range(height):
    for x in range(width):
        specular_map[y, x] = onp.array(image.getpixel((x, y)))
specular_map = jp.array(specular_map, dtype=jp.single)[..., 0]

verts, norms, uv, faces, faces_norm, faces_uv = [], [], [], [], [], []
_float, _integer, _one_vertex = re.compile(r"(-?\d+\.?\d*(?:e[+-]\d+)?)"), re.compile(r"\d+"), re.compile(r"\d+/\d*/\d*")

with open(obj_path, 'r') as file:
    for line in file:
        if line.startswith("v "):
            verts.append(tuple(map(float, _float.findall(line, 2)[:3])))
        elif line.startswith("vn "):
            norms.append(tuple(map(float, _float.findall(line, 2)[:3])))
        elif line.startswith("vt "):
            uv.append(tuple(map(float, _float.findall(line, 2)[:2])))
        elif line.startswith("f "):
            face, face_norm, face_uv = [], [], []
            vertices = _one_vertex.findall(line)
            assert len(vertices) == 3, f"Expected 3 vertices, got {len(vertices)}"
            for vertex in vertices:
                v, vt, vn = list(map(int, _integer.findall(vertex)))
                face.append(v - 1)
                face_norm.append(vn - 1)
                face_uv.append(vt - 1)
            faces.append(face)
            faces_norm.append(face_norm)
            faces_uv.append(face_uv)

model = Model(
    verts=jp.array(verts),
    norms=jp.array(norms),
    uvs=jp.array(uv),
    faces=jp.array(faces),
    faces_norm=jp.array(faces_norm),
    faces_uv=jp.array(faces_uv),
    diffuse_map=jax.numpy.swapaxes(texture, 0, 1)[:, ::-1, :],
    specular_map=jax.numpy.swapaxes(specular_map, 0, 1)[:, ::-1],
)

canvas_width, canvas_height, frames, rotation_axis = 1920, 1080, 30, "Y"
rotation_axis = dict(X=(1., 0., 0.), Y=(0., 1., 0.), Z=(0., 0., 1.))[rotation_axis]
degrees = jax.lax.iota(float, frames) * 360. / frames

eye, center, up = jp.array((0, 0, 3.)), jp.array((0, 0, 0)), jp.array((0, 1, 0))
camera = CameraParameters(viewWidth=canvas_width, viewHeight=canvas_height, position=eye, target=center, up=up)
light = LightParameters(direction=jp.array([0.57735, -0.57735, 0.57735]), ambient=0.1, diffuse=0.85, specular=0.05)
shadow = ShadowParameters(centre=center)

@jax.default_matmul_precision("float32")
def render_instances(instances, width, height, camera, light, shadow):
    img = Renderer.get_camera_image(objects=instances, light=light, camera=camera, width=width, height=height, shadow_param=shadow, colour_default=jp.zeros(3, dtype=jp.single))
    return jax.lax.clamp(0., img, 1.)

def rotate(model, rotation_axis, degree):
    instance = ModelObject(model=model)
    return instance.replace_with_orientation(rotation_matrix=rotation_matrix(rotation_axis, degree))

batch_rotation = jax.jit(jax.vmap(lambda degree: rotate(model, rotation_axis, degree))).lower(degrees).compile()
instances = [batch_rotation(degrees)]

@jax.jit
def render(batched_instances):
    def _render(instances):
        _render = jax.jit(render_instances, static_argnames=("width", "height"), inline=True)
        img = _render(instances=instances, width=canvas_width, height=canvas_height, camera=camera, light=light, shadow=shadow)
        return transpose_for_display((img * 255).astype(jp.uint8))

    return jax.jit(jax.vmap(_render))(batched_instances)

render_compiled = jax.jit(render).lower(instances).compile()
images = list(map(onp.asarray, jax.device_get(render_compiled(instances))))

write_apng('animation.png', images, delay=1/30.)
# ffmpeg -i animation.png intermediate.gif
# gifsicle --optimize=3 --delay=5 intermediate.gif > output.gif
``` 

All these views were rendered in parallel using jax as the only dependency:

[image]

markusheimerl, 2024-06-05:
Thank you for your input. Godspeed on integrating Madrona.

closed 2024-06-05 by markusheimerl

JeyRunner, 2024-12-07:
@markusheimerl in case you didn't see it already, here seems to be a working approach of using madrona for rendering in mjx:
https://github.com/shacklettbp/madrona_mjx