google-deepmind/mujoco #1686: Equation for the Scalar Length l_i in the Actuator Transmission for Hinge Joints

AdrianZw, 2024-05-22:
Hi,

I'm an engineer, and I'm trying to use MuJoCo for a research project.

I'm looking for some help with the model behind the actuation model for a hinge joint, to be precise.

In https://mujoco.readthedocs.io/en/stable/computation/index.html#force-generation, there is a reference to a function $l_i(q)$ (scalar legnth) and it's derivative $\dot{l}_i(q,v)$  in the scalar force $p_i$.  Can someone give me or point me into an equation for that function?

Besides, I would interest the correspondence of the XML parameters to that equation. 
I would guess that parameters described here https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-joint somehow affect this equation, is this correct?

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

This is xml-snippet out of the Sawyer model within https://github.com/google-deepmind/mujoco_menagerie and I think we could do some model identifaction for Sawyer or have done some model identification in the past, so I could be able to dig up our parameters.
However, I don't really understand the below parameters affect the model in MuJoCo. Thus, I can compare them to our models.

```XML
<mujoco>
  ...
      <default class="large_joint">
        <joint frictionloss="0.3" armature="1" damping="0.1"/>
        <general gainprm="800" biasprm="0 -800 -80" forcerange="-80 80"/>
      </default>
</mujoco>
```

</details>

yuvaltassa, 2024-05-22:
The length of a joint transmission is just the joint angle. Read the function mj_transmission for details regarding this and all other transmissions.

yuvaltassa, 2024-05-22:
For other details regarding actuation, (e.g. bias/gainprm), read mj_fwdActuation.

closed 2024-05-22 by yuvaltassa