google-deepmind/mujoco #1687: Estimate qvel from sequence of rotations

areiner222, 2024-05-22:
Hi,

I'm using mujoco(-mjx) for imitation learning and am trying to determine the best (right?) way to compute qvel from mocap data.

I have a humanoid model and using 3 hinge joints associated will all bodies in the kinematic chain besides the root free joint. I'm able to infer qpos from the sequence of kinematic chain relative rotations by converting those rotations into xyz euler angles (I say this with some confidence because the rendered sequence of positions after computing forward kinematics matches my reference motion), but I'm not sure I am computing qvel properly. As of now, I am just doing finite differencing on the qpos (euler) angles (mocap running at 60hz). Is this a good way to do it? I try to account for angle wrapping by computing the qpos / euler angles sequentially ensuring a continuous representation. Do I need to factor joint ranges into this as well?

Thanks!

yuvaltassa, 2024-05-22:
- Very specifically for the OP question, finite differencing is the correct thing. It is trivial for the hinges and sliders and only slightly less trivial for the quaternion in the free joint. 
- Side comment: 3 hinge joints everywhere is a bit much, no? Why give the elbow 3 dofs if you know it has 1?
- More generally, you are solving an Inverse Kinematics problem with retargeting. I recommend stepping through the [Least Squares tutorial notebook](https://github.com/google-deepmind/mujoco?tab=readme-ov-file#getting-started), which discusses IK with examples, and possibly @hartikainen's [MJPC](https://github.com/google-deepmind/mujoco_mpc) task (which solves a **much** harder problem: it finds the controls which achieve the desired tracking):

[image]

areiner222, 2024-05-23:
coming back here to say least_squares is magic - inverse kinematics with the previous state continuity regularization working really well

kevinzakka, 2024-05-24:
Nice @areiner222 !!

yuvaltassa, 2024-05-30:
Well 2 is just a better version of 1, right? You've got the right idea there. We have some code to do this in the mujoco_mpc repo, perhaps @erez-tom or @thowell can help you locate it, I think it's pretty straightforward to use.