google-deepmind/mujoco #1688: How to simulate glass window/door in mujoco?

0oook, 2024-05-23:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a student and I'm trying to use MuJoCo for learning.

I'm looking for some help with how to achieve a mirror effect glass window.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
    <asset>
        <material name="mirror_floor" rgba=".1 .2 .3 1" reflectance=".2"/>
        <material name="mirror_window" rgba="1 1 0 0.1" reflectance=".2"/>
    </asset>

    <worldbody>
        <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
        <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
        <geom type="plane" size=".5 .5 .01" material="mirror_floor"/>
        <geom type="box" size=".5 .01 .5" pos="0 0.1 0" material="mirror_window"/>
        <body name="ball" pos="0 0 0.05">
            <freejoint/>
            <geom size=".03"/>
        </body>
        <body name="ball_01" pos="0 0.2 0.05">
            <freejoint/>
            <geom size=".03"/>
        </body>
        <body name="hammer" pos="-.05 0 0.18">
            <joint axis="0 1 0"/>
            <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
            <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
        </body>
    </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:
    As you can see, the ground has a mirror effect, but the mirror_window does not.

https://github.com/google-deepmind/mujoco/assets/40737324/5f67f5e4-df9e-456a-8195-7bb3c724dfa3

yuvaltassa, 2024-05-23:
Did you read the [documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#asset-material-reflectance)?

"Only the first reflective geom in the model is rendered as such"

closed 2024-05-23 by yuvaltassa

0oook, 2024-05-24:
Sorry, I didn't read the document carefully. I read the document carefully again., I put mirror_window first, and put the Z axis towards the hammer,but it still had no effect.
<mujoco>
    <asset>
        <material name="mirror_floor" rgba=".1 .2 .3 1" reflectance=".2"/>
        <material name="mirror_window" rgba="1 1 0 0.1" reflectance=".2"/>
    </asset>

    <worldbody>
        <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
        <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
        <geom type="box" size=".5 .5 .01" pos="0 .1 .5" euler="90 0 0" material="mirror_window"/>
        <geom type="plane" size=".5 .5 .01" material="mirror_floor"/>
        <body name="ball" pos="0 0 0.05">
            <freejoint/>
            <geom size=".03"/>
        </body>
        <body name="ball_01" pos="0 0.2 0.05">
            <freejoint/>
            <geom size=".03"/>
        </body>
        <body name="hammer" pos="-.05 0 0.18">
            <joint axis="0 1 0"/>
            <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
            <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
        </body>
    </worldbody>
</mujoco>

yuvaltassa, 2024-05-24:
Does it work if you make the alpha of the window 1?

It might be the case that reflectance and transparency are not compatible... I realize this a problem for your use case.

0oook, 2024-05-24:
Thank you for your reply! 
Yes, I can achieve the mirror effect by adjusting alpha to 1, but there is no transparency effect.
So, how to make them compatible?

yuvaltassa, 2024-05-24:
I don't have a good answer for you with the native renderer. I guess you could do this in Unity? @Balint-H WDYT?

Balint-H, 2024-05-24:
Hello!

I checked quickly online, when using the HD Render Pipeline (HDRP) Unity can support reflections on transparent objects. You could consider using MuJoCo with Unity to achieve this effect, depending on how critical it is.

Alternatively you could always consider the hack of having a transparent copy of your scene behind the transparent window, and you could copy/mirror the state from the "real" hammer and ball.

yuvaltassa, 2024-05-24:
Ah ya that's a cute idea @Balint-H! Render two images, one with a perfect mirror, one without it. The average of these images will give you "translucent, reflective glass".

0oook, 2024-05-24:
Ok，thanks，i will try it later.