google-deepmind/mujoco #1694: [Unity plugin] Add kv property for position actuators

StafaH, 2024-05-23:
In Mujoco 3.1.0, the `kv` attribute was added for position actuators. This is currently still missing from the Unity plugin, which only looks for kp in the Custom Params for a position actuator. Can this be added in the plugin?

Fix should be minimal, adding `kv` as a GUI element when `position` actuator is selected, and adding the get/set for the attribute in PositionToMjcf and PositionFromMjcf. A variable for `kv` already exists for velocity controllers.

Balint-H, 2024-05-24:
Hello, thanks for reporting this! This is simple enough fix, I can do it in the next couple of days.

Balint-H, 2024-05-27:
Just submitted #1696 to address this, thanks for the patience!

closed 2024-06-05 by StafaH