google-deepmind/mujoco #1697: [<Actuator>] Custom Control Range Mapping for Actuators in MJCF XML

ozgurgulsuna, 2024-05-27:
**Summary**
I am suggesting a feature to support the ability to map the inherited joint control range to a custom range for the **_position controller_** within the <position> tag in the XML configuration file. This feature would be particularly useful for scenarios where actuators are procedurally generated, and each joint starts in a random position within its respective movement range.

here the joint position range is between -1.5 and -0.2
`<joint name="Linear-0-1" type="slide" axis="0.396823 0.917895 0.000000" range="-1.154032 -0.204032"/>`

The new feature could be implemented by adding attributes to the <actuator> tag that specify the desired control range. For example:
`<position name="Member-0-1" joint="Linear-0-1" kp="18000"  kv="10000" inheritrange="1" customrange="0 1" />`

**Problem Description**
In many simulations, especially those involving procedurally generated robots, joints initialize at random positions within their respective movement ranges. These ranges vary between each joint, with some joints having negative minimum range. As a result, each actuator ends up with a unique control range. Currently, there is no built-in functionality to standardize or map these varying control ranges to a consistent scale. This leads to a high number of actuators, each requiring manual adjustment to set their control ranges appropriately in python or c++.

**Benefits**
- Standardizing control ranges between two numbers simplifies the setup and management of multiple actuators, more suitable for learning applications.

- Allows for convenience in simulations where the random initial joint positions does not affect the actuator configurations.

Thank you for your attention and for your continuous efforts in improving MuJoCo.
Here is a procedurally generated truss robot model from my research.

[image]

yuvaltassa, 2024-05-29:
This is certainly possible. I think I prefer `rescalerange` to `customrange`, wdyt?

There is also the question of the semantics of `kp`, should they be wrt the natural units or the rescaled units?

yuvaltassa, 2024-05-30:
BTW [here](https://github.com/google-deepmind/dm_control/blob/04ff055d0b449742a3e075af6c0a711c2edfae89/dm_control/locomotion/walkers/scaled_actuators.py#L37) is a dm_control function that is basically doing what you want, correct?