google-deepmind/mujoco #1701: MuJoCo cup model penetrating floor/table

cherylwang20, 2024-05-28:
Hi,

I'm a student and I'm trying to use MuJoCo to simulate grabbing a concave object (e.g. beaker) with a UR10e Robotic Arm. I am trying to start with the cup model as a reference but it looks like the cup is penetrating the floor when positioned sideways. 

[image]

I tried putting it in the robohive env and it is also penetrating the table that I am positioning on. 

[image]

I'm looking for some help with potentially how to modify the collision type on the side so that it behaves correctly. I see that in this demo: https://github.com/google-deepmind/mujoco/discussions/357#discussioncomment-3022028, the cup is colliding well with the floor. 

Here is the link to the XML file that I am using:
[
robohive/envs/arms/ur10e/scene_gripper.xml](https://github.com/cherylwang20/mj_envs/blob/76784b850b37901cae7c9c2f6c94a47b1e129809/robohive/envs/arms/ur10e/scene_gripper.xml)

Any help is appreciated! Thank you.

cherylwang20, 2024-05-28:
I tried again and it looks like when the cup is positioned in the upright position and then letting the cup fall, it is not penetrating, it's only when the cup initially starts with a side-way position that this penetration is happening. 

[image]

Is there any possible explanation for this behavior?

Balint-H, 2024-05-29:
How do you set the rotation of the mug? Do you just apply pure rotation to the freejoint? 

Since the body frame of the mug starts at the bottom, rotation without position offset will cause it to penetrate the surface. Depending on the thickness of the surface, this can cause the mug to get stuck. You can either increase the thickness of the surface, disable the table collisions and add a plane instead, or offset the mug to prevent the penetration. 

If you need it on its side, you just need to move it upwards with a height equal to the mug's radius.

kevinzakka, 2024-05-30:
I think this has been addressed in https://github.com/kevinzakka/obj2mjcf/issues/31. @cherylwang20 Feel free to re-open if you run into further issues.

closed 2024-05-30 by kevinzakka