google-deepmind/mujoco #1710: `mj_copyData` not copying contact details?

marioprats, 2024-06-03:
Hi,
I have an issue with `mj_copyData` not copying contact details. Not sure if this is a bug or something intended.

Here is a minimal model:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
  <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.0">
      <freejoint/>
      <geom size=".03"/>
    </body>
  </worldbody>
</mujoco>
```

</details>

It has a sphere in contact with a ground plane.
I found the issue with `mj_copyData` in C++, but it can be reproduced in python too:

<details>
  <summary>minimal Python reproduction case</summary>

```python
import mujoco
import copy

model = mujoco.MjModel.from_xml_path("scene.xml")
data = mujoco.MjData(model)

mujoco.mj_forward(model,data)

print("Original data:")
print("  n_contacts: ", data.ncon)
print("  colliding geoms: ", data.contact.geom)

data2 = copy.copy(data)

print("Data copy:")
print("  n_contacts: ", data2.ncon)
print("  colliding geoms: ", data2.contact.geom)

```
</details>

The output is:

```
Original data:
  n_contacts:  1
  colliding geoms:  [[0 1]]
Data copy:
  n_contacts:  1
  colliding geoms:  [[0 0]]
```

So the copy preserves `ncon`, but doesn't seem to copy contact details over (e.g. colliding geoms).

yuvaltassa, 2024-06-03:
Hmmm, this should work. (That said, there is [no test for this function](https://github.com/google-deepmind/mujoco/blob/main/test/engine/engine_io_test.cc), so ya this could be a bug)

@saran-t is there a bug in [this line](https://github.com/google-deepmind/mujoco/blob/14a3ce5ffac0550a645ca391119014f489f70b55/src/engine/engine_io.c#L1268)?

saran-t, 2024-06-03:
It's certainly not deep-copying the arena. I can't remember whether that was a deliberate choice though, for MJPC?

yuvaltassa, 2024-06-03:
Oh damn you're right. So the comment `// copy arena memory` is basically a lie eh?
It should either be `// set arena pointers, don't copy memory` or we should actually copy the memory...

I have no recollection of this being a deliberate choice, seems like a strange thing to do. @nimrod-gileadi, @erez-tom, @thowell, any such recollection?

Marking this as a bug, even if we end up deciding that it's deliberate, the inline and API docs should be updated.

nimrod-gileadi, 2024-07-11:
We do have tests for this function in the Python bindings, but indeed we don't check for contacts: https://github.com/google-deepmind/mujoco/blob/14a3ce5ffac0550a645ca391119014f489f70b55/python/mujoco/bindings_test.py#L430

I think that copying the arena content is the right thing to do. Can't see why not.

closed 2024-07-11 by copybara-service[bot]

kurtenkera, 2024-11-01:
Hi @nimrod-gileadi, can I confirm that this bug occurs regardless of whether one accesses MuJoCo via the Python bindings or the C API? This bug is currently preventing me from rotating contact forces on a soft-gripper from the world frame to the rest frame of the gripper (see #2114). Do you guys intend on fixing this bug in one of the upcoming releases?

commit e7301edd (2024-07-11): Copy arena contents in mj_copyData.

Fixes #1710.

PiperOrigin-RevId: 651351384
Change-Id: I56e3204b30bb18f48278117e85e90464177da973