google-deepmind/mujoco #1720: Transforming contact forces from the contact frame to the local frame of a geom involved in said contact/collision?

kurtenkera, 2024-06-05:
Hi,

I am using MuJoCo via Python, and I am currently using `mj_contactForce` to extract the contact forces acting on arbitrary pairs of geoms during a collision. I understand the resulting contact forces are expressed in the **contact frame** of the contact forces, but I need to express them in the **local frame of a particular geom** involved in the contact. How can I express the resulting contact forces in the frame of a given geom that experiences said contact force?

I know that I can transform from the global/world frame to the local frame of a geom via the transpose of `geom_xmat`, which acts as a "global-to-local" transformation for a given geom. However, if I am to incorporate `geom_xmat` I need to first find a way to transform from the contact frame to the global/world frame?

So then, is there a general scheme that can transform contact forces from the contact frame to the frame of a geom involved in a collision?

yuvaltassa, 2024-06-09:
You're in the right direction but missing an important bit. A frame transform is both a translation and rotation, you're missing the translation. What you want is to go via the world frame like (contact_frame → world_frame → geom_frame). 

If `x` is in the geom frame and `y` is in the world frame then `y = geom_xmat.T @ x + geom_xpos`, and ` x = geom_xmat @ (y - geom_xpos)`. 

I think you should be able to get it from here, just note [the caveat](https://mujoco.readthedocs.io/en/latest/programming/simulation.html#contacts): `mjContact.frame`, uniquely in MuJoCo, is column major rather than row major (i.e. transposed). Let me know if you couldn't figure it out 🙂

closed 2024-06-09 by yuvaltassa

kurtenkera, 2024-06-09:
Ah so if `z` is in the contact frame and `x` is in the geom frame then `x = geom_xmat.T @ mjcontact.frame @ (z - contact_reference_pos)`, where `contact_reference_pos` is some reference position in the contact frame needed to capture the translation you mentioned?

yuvaltassa, 2024-06-09:
You're once again forgetting a translation.

Let's transpose and re-order as explained in the link above and get a regular xmat: `contact_xmat = contact.frame.reshape(3,3).T[:, [1, 2, 0]]`. Transpose to put the axes on the columns, reorder to get xyz order. So if `f` is the force we got from `mj_contactForce` then
```py
x = contact_xpos + contact_xmat.T @ f
y = geom_xmat @ (x - geom_xpos) 
```
note I flipped `x` and `y` above since `x` in the world frame is more MuJoCo-esque.

kurtenkera, 2024-06-12:
Thanks for the help @yuvaltassa, this makes sense.

For my purposes, I only need the correct **direction** and **magnitude** of `f` in the geom frame (it won't matter if `f` is a vector whose base is located at the correct contact position or not). So while I can see that your equations above for `x` and `y` would give `f` in the geom frame at the correct contact position, is it true that something like `f_geom = geom_xmat @ contact_xmat.T @ f` would give a contact force with the correct **direction** and **magnitude** (but obviously not the correct contact position in the geom frame)?

yuvaltassa, 2024-06-13:
I don't think so, but I might be wrong.

kurtenkera, 2024-06-19:
Hi @yuvaltassa. I believe the following rule you quoted above is wrong: 

> If y is in the geom frame and x is in the world frame then x = geom_xmat.T @ y + geom_xpos, and  y = geom_xmat @ (x - geom_xpos).

It should instead be:

If y is in the geom frame and x is in the world frame then x = geom_xmat @ y + geom_xpos, and  y = geom_xmat.T @ (x - geom_xpos).

I have played around with this extensively in MuJoCo and this has given me correct results, whereas the rule you quoted was giving me incorrect results. This also agrees with the rules for coordinate transforms explained [here](https://motion.cs.illinois.edu/RoboticSystems/CoordinateTransformations.html#Representation-of-coordinate-frames-and-coordinate-transforms) and Emo Todorov's comments in this old [thread](https://www.roboti.us/forum/index.php?threads/contact-locations-on-object.3459/) (i.e., "geom_xmat is a matrix that rotates from local to global coordinates").

Also, for a force `f` measured in the contact frame, **we do not want to apply translations** as doing so won't preserve the magnitude of `f`. Letting `contact_xmat = contact.frame.reshape(3,3).T` and applying the updated rule above, to obtain `f_geom` we would need to do `f_geom = geom_xmat.T @ contact_xmat @ f`. Since `geom_xmat` and `contact_xmat` are orthogonal matrices, it's easy to confirm the magnitude of `f_geom` equals the magnitude of `f` (but this isn't true if we include translations). Of course, we would need to apply the translations if we were transforming **a point** (not a force vector) in the contact frame.

I just wanted to include these comments (1) in case other users need to perform similar calculations and (2) in the off-chance I'm mistaken!

yuvaltassa, 2024-06-20:
Yes of course. Thanks for correcting!

The moral of the story is: believe the code that works, not the code written from memory 😛

If you find a good place to add this to the docs, feel free to send a PR!

Maybe here?

https://mujoco.readthedocs.io/en/latest/programming/simulation.html#coordinate-frames-and-transformations