google-deepmind/mujoco #1724: Using gmsh files in MuJoCo - XML Error: Error: Node tags must be sequential

sowmr5, 2024-06-07:
Hi,

I'm a student and I'm trying to use MuJoCo for robotic arm simulation.

I'm looking for some help with adding msh (gmsh) files to MuJoCo.

There is an error: "XML Error: Error: Node tags must be sequential" when using the file in MuJoCo.

I've loaded an stl file into gmsh and exported to gmsh format but the order is not sequential.

I just wondered if there was a way to reorder the nodes so that they are sequential and accepted by MuJoCo.

Any help is greatly appreciated, thanks in advance.

mohammad200h, 2024-06-07:
Can you share the stl and msh file please?

sowmr5, 2024-06-10:
Thanks for your reply, no problem, please see below.
[sample.zip](https://github.com/user-attachments/files/15758829/sample.zip)

mohammad200h, 2024-06-11:
1. Here is a video instruction of how to create a volume: 
https://youtu.be/C5BWRWCr7Ck
2. If you look at the video you see that there is a surface and volume entity. Mujoco will not be able to read this file.
3. You need to use GMSHConverter to clean the file (Official release coming soon). 
https://github.com/mohammad200h/GMSHConverter/tree/tree
4. Here is the end result 
[sample_41_ascii_out_vol.msh.zip](https://github.com/user-attachments/files/15791894/sample_41_ascii_out_vol.msh.zip)

Use that and let me know how it goes.

mstoelzle, 2024-07-29:
> @mohammad200h Many thanks for that, I used the converter on a few other meshes too and that ordered the nodes properly in the *_vol.msh file.
> 
> Only thing I thought worth mentioning is that the -i and -o flags don't seem to accept relative paths as it tries to write to the root. So I used full paths for both -i and -o and that was fine.
> 
> ```
> python converter.py -i reduced.msh -o reduced2.msh
> input_path::  reduced.msh
> Info    : Reading 'reduced.msh'...
> Info    : 1 entity
> Info    : 1889 nodes
> Info    : 3757 elements
> Info    : Done reading 'reduced.msh'
> Traceback (most recent call last):
>   File "converter.py", line 266, in <module>
>     main()
>   File "converter.py", line 257, in main
>     VolumeExtractor(args.input, args.output)
>   File "converter.py", line 132, in __init__
>     super().__init__(input_file_path, output_file_path)
>   File "converter.py", line 17, in __init__
>     self.process()
>   File "converter.py", line 20, in process
>     self.write41_mesh_format_41()
>   File "converter.py", line 84, in write41_mesh_format_41
>     with self.output_file_path.open('w') as f:
>   File "/usr/lib/python3.8/pathlib.py", line 1222, in open
>     return io.open(self, mode, buffering, encoding, errors, newline,
>   File "/usr/lib/python3.8/pathlib.py", line 1078, in _opener
>     return self._accessor.open(self, flags, mode)
> PermissionError: [Errno 13] Permission denied: '/reduced2_vol.msh'
> ```
> 
> Thanks again for your help.

@sowmr5 It should probably to replace:
- `gmsh.write(obj.name + ".msh")` with `gmsh.write(str(pathlib.Path(self.path) / (obj.name + ".msh")))`
- `gmsh.write(obj.name + ".msh")` with `gmsh.write(str(pathlib.Path(self.path) / (obj.name + ".msh")))`