google-deepmind/mujoco #1725: Rotation of 3 hinge joints with orthogonal axes.

KamatMayur, 2024-06-08:
Hi,

I'm a student and I'm trying to use MuJoCo for my final year engineering project.

I'm looking for some help with how to interpret the rotation of 3 hinge joints with orthogonal hinge axes.
If I use the simulate application to play with the values, it appears that the rotations aren't intrinsic or extrinsic euler angles. Neither does the order of rotation seem to affect the final orientation of the body.

The model that im using is taken from [here](https://github.com/google-deepmind/dm_control/blob/main/dm_control/suite/humanoid_CMU.xml) can someone please explain it to me? I have a bvh model with the same heirechy and im trying to transfer the animations from the bvh file to this mujoco model. But I'm unable to proceed because I do not understand it.

yuvaltassa, 2024-06-09:
It means just what it [says on the tin](https://github.com/google-deepmind/dm_control/blob/ed424509c0a0e8ddf7a43824924de483026ad9cc/dm_control/suite/humanoid_CMU.xml#L42). z, then y, then x (intrinsic, of course).

yuvaltassa, 2024-06-09:
(You should probably load this model in `simulate`, turn on joint visualization and play with the sliders)

closed 2024-06-09 by yuvaltassa