google-deepmind/mujoco #1730: Difference between training model in MJX and Evaluating in Mujoco with Height Map Activated

AlexS28, 2024-06-12:
Hi,

I wanted to ask if there was any analysis (or attempt) made to evaluate the model trained in MJX (specifically with the height map) and evaluate this trained model in Mujoco (with the height map applied in Mujoco as well)? I am mostly asking because I was successfully able to train our Biped robot on the height map in MJX, but this policy did not work in the regular Mujoco. However, a flat terrain in regular Mujoco did work. So I just want to double check if this is an error on my end, or if the height map in MJX vs Mujoco have some difference? Maybe on way to check if someone already has done is to see if the barkour example in MJX trained on the height map works in regular Mujoco with the height map.

Thanks,
Sincerely,
Alex

btaba, 2024-06-12:
Hi @AlexS28 , I have not tried this personally. The contacts between heightmaps in MJX vs MuJoCo are not equivalent. In MJX we use the separating axis test with one-shot contact manifolds (4 points), and cull these to a max of 4 points. In MuJoCo, we use MPR and can use up to 50 contact points FWIU. In the MJX [tutorial](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb), we do show transfer for the humanoid between MJX and MuJoCo, but the collision detection in that case is equivalent (capsules, planes, spheres). Hope that helps

AlexS28, 2024-06-12:
I see, thanks for the answer, I guess in that case it make sense why the behavior could be different between the two then.

closed 2024-06-12 by AlexS28