google-deepmind/mujoco #1736: Saving and restoring MuJoCo state for Model-Predictive Control

sanjainarainperspectalabs, 2024-06-13:
Hello, we are trying to program model-predictive control for a MuJoCo model. In MPC, in a given state S, we need to simulate forward along a large number of action sequences to find out the best one. Before we evaluate a new sequence, we need to restore the model's state to S. How should we store S? Just saving the qpos and qvel of all the joints does not seem to be enough. What else do we need to store and restore S? 

Concurrently, we are also evaluating the MJPC package where this problem must be addressed. But we'd also like to know how to do MPC from first principles. 

Thanks. -- Sanjai

yuvaltassa, 2024-06-13:
https://mujoco.readthedocs.io/en/latest/computation/index.html#the-state

closed 2024-06-13 by yuvaltassa

sanjainarainperspectalabs, 2024-06-13:
Thanks, Yuval. We'll look closely at this link. -- Sanjai