google-deepmind/mujoco #1738: Query on viewer/render/read_pixels API migration to Mujoco 3.x

girishsmurthyIntel, 2024-06-14:
Dear Team,

        I was unable to understand how to map mujoco2.x APIs **mujoco_py.MjViewerBasic**, corresponding **render and read_pixels** APIs in mujoco 3.0. Request your kind help on this.

girishsmurthyIntel, 2024-06-15:
I used the API mujoco.viewer.launch_passive(self.conveyor.model, self.conveyor.data, show_left_ui= False, show_right_ui= False), but I get the error:
ERROR: mjv_makeSceneState: mjvSceneState buffer is not fully used

Press Enter to exit ..

Please advise.

girishsmurthyIntel, 2024-06-18:
@yuvaltassa Can you kindly help here.

yuvaltassa, 2024-06-18:
We cannot help you without a minimal reproducible example.

girishsmurthyIntel, 2024-06-18:
Hi @yuvaltassa, Please find the model attached. The main xml file is warehouse_v2.xml
The segments of the code I have used are 
        self.model = mujoco.MjModel.from_xml_path(fullpath) # fullpath is the path for the file warehouse_v2.xml
        self.data = mujoco.MjData(self.model)
        mujoco.mj_forward(self.model, self.data)
        mujoco.viewer.launch_passive(self.model, self.data, show_left_ui= False, show_right_ui= False)
[assets.zip](https://github.com/user-attachments/files/15890240/assets.zip)

girishsmurthyIntel, 2024-06-20:
Dear @yuvaltassa, Can you kindly check this xml file and help us know of this error.

girishsmurthyIntel, 2024-06-20:
Hi @yuvaltassa , I figured out that if I remove the <size njmax="10000" nconmax="10000"></size> from the xml file, the error doesn't occur. Kindly provide your insights.

yuvaltassa, 2024-06-20:
I don't know what it means to "remove the from". No insights from me.

girishsmurthyIntel, 2024-06-20:
I am so sorry, the comment isn't rightly uploaded.
I figured out that if I remove the line below in the warehouse_v2.xml file, the error doesn't occur. kindly provide your insights
**<size njmax="10000" nconmax="10000"></size>**

yuvaltassa, 2024-06-20:
We have documentation, please read it. Specifically take note of the word **legacy**

https://mujoco.readthedocs.io/en/latest/modeling.html#csize

https://mujoco.readthedocs.io/en/latest/XMLreference.html#size-memory

closed 2024-06-20 by yuvaltassa

sIHURs, 2024-09-26:
> Hi @yuvaltassa , I am so sorry, the comment isn't rightly uploaded. I figured out that if I remove the line below in the warehouse_v2.xml file, the error doesn't occur. kindly provide your insights **njmax="10000" nconmax="10000"**

Hi, I would like to ask if a cause has been found for this issue, I have the same problem. My XML file runs fine in the simulator, but I encounter this error when using the Python API: ERROR: mjv_makeSceneState: mjvSceneState buffer is not fully used.