google-deepmind/mujoco #1739: Need help on how to specify the parameters for the contact model, or how to use customized contact model

ghweili, 2024-06-16:
Hi,

I am a researcher in mechanical engineering who is using Mujoco and ADAMS to simulate the collision process of a robot falling onto the ground. The robot comprises several rigid links (made of materials like  carbon fiber or aluminum) and springs. The main goal of using both software programs is to compare their results and validate the outcomes.

The biggest challenge I face is the difference in contact models used by ADAMS and Mujoco. 

In ADAMS, the contact model is defined by the formula:

$\text{Impact}=\text{Max}\left(0, k\left(x_{1}-x\right)^{e}-\text{STEP}\left(x, x_{1}-d, c_{\max }, x_{1}, 0\right) \cdot \dot{x}\right)  : x < x_{1} $

where $x_1$ indicates the collision start, and $d$ (the penetration depth) specifies at which point the damping coefficient reaches its maximum. It seems that this equation can be roughly understood as $f = kx + c\dot{x}$, except that the power of $x$ is not equal to 1. As can be seen, this power leads to inconsistency in the units of the three terms. However, the documentation explains how to handle this issue by multiplying this term by a constant term such as ``$1000^2$''. In this way,  all three terms become unit of force, e.g., Newtons.

As for Mujoco, it uses an indirect way to calculate the acceleration under collision, where  one first specifies solimp: ($d_0,d_{width} , width, midpoint, power$) to determine a function called $d(r)$,  then specify solref: (-stiffness, -damping) to assign the so called "stiffness" and "damping". Then, the real acceleration $a_1$  can be given by 

$a_{1} + d \cdot (b v + k r) = (1 - d) \cdot a_{0}$

where $a_0$ is the acceleration without considering the effects of the collision. If I understand it correctly, $d$ can be roughly regarded as a weight, considering that $a_1$ can be written in the form of a weighted sum, namely,

$a_{1} = d \cdot (-b v - k r) + (1 - d) \cdot a_{0}$

I have several questions in order to make the simulations in Mujoco and ADAMS behave similarly. 

1. Is it possible to use a user-defined contact model in Mujoco to replace the default model? If so, how can this be done? I have barely found any relevant information on this topic so far, although it is mentioned in the attached website that this is possible. Since contact is one of the most essential components of Mujoco (as suggested by its name),  this function could be of interest by many users, at least for the sake of making comparisons with other tools.

 

2. Which contact model (ADAMS or Mujoco) is more suitable for simulating a robot with rigid links (e.g., carbon fiber, aluminum) colliding with different ground types (e.g., cement road,  gravel road, grassland, soil land, beach)? 

3. How should I specify the relevant parameters for these cases? And generally, what are the differences (in terms of the collisions) between the simulation results produced by Mujoco and ADAMS? 

4. I am puzzled by the relationship $b = {\text{damping}}/{d_{\text{width}}}$ and $k = {\text{stiffness} \cdot d(r)}/({d_{\text{width}}^2)}$, where $d_{\text{width}}$ and $d(r)$ seem dimensionless. This implies $b$ and $k$ should have the same units as damping and stiffness, respectively. If this is true, the units in the equation $a_{1} + d \cdot (b v + k r) = (1 - d) \cdot a_{0}$ seem inconsistent. 

Additionally, when testing a simple cube falling from a given height onto the ground, we observed that the collision process remained the same regardless of the mass, which contradicts results from ADAMS.
We have also used mathematical software to solve the ODE function of a simple mass-spring-damper system oscillating vertically (subject to gravity) to double check the result, which, again, indicates that the mass should be relevant.  Although the latter experiment seems too simple for modeling collision, it demonstrates the fact that mass would make a difference even for such a simple dynamic process.

As a result, I'm not sure if the unexpected behavior in Mujoco is indeed correct, and whether it is caused by some mistakes we made when specifying the parameters, especially the stiffness and damping.  Although we found some references suggesting dividing stiffness and damping by inertia, it's unclear to us which inertia is meant, especially given that each rigid link has a six-dimensional inertia matrix which are hard to determine, and that we usually have many links connected to each other.

5.  So how should we exactly specify the relevant parameters for solimp: ($d_0,d_{width} , width, midpoint, power$) and solref: (-stiffness, -damping) based on the stiffness (N/m), damping (Ns/m) and penetration depth (m)?   After checking the documentation many times and looking at a number of references, we still can't figure it out the relation between these physical quantities and the required parameters.

Reference: The webpage suggesting the possibility of using customized contact model for Mujoco: https://people.tuebingen.mpg.de/mpi-is-software/pam/continuous_integration/18.04/docs/pam_documentation/doc/C1_handle.html

yuvaltassa, 2024-06-16:
1. No, though it can be added. No one has asked for this before, so I'm not sure it's as useful as you think.
2. I'm not familiar enough with ADAMS to answer that.
3. Ideally, you'd set these by doing proper System Identification.
4. MuJoCo normalizes by the contact inertia (inertia projected into the contact frame), so different masses have no effect on contact kinematics.
5. See 3.

closed 2024-06-16 by yuvaltassa

ghweili, 2024-06-16:
1. For the system identification you mentioned, does this mean that one has to develop a physical prototype first? If so, this would be quite inconvenient when one is doing optimal design in Mujoco, as no physical prototypes are available yet. In my case, I have to run numerous simulations to find the best parameters for the robot to use.

2. If you are proposing using some simplified preliminary physical experiments to conduct system identification for the contact parameters, this can also be inconvenient in certain cases, especially when these parameters are also to be optimized. For example, in the optimal design of hopping robots, one may want to find the best parameters for the contacts, based on which one could update the materials to achieve better performance. It seems that this inconvenience would occur for any applications where the robot's performance highly depends on the contact parameters, and one may want to optimize these parameters (such as hopping robots, moon landers, and robotic machining/assembling operations).

3. In ADAMS and many other software, a simple contact model with relatively clear physical meanings is used; moreover, these models can be easily used by general users because their input only involves physical properties with clear units such as stiffness (N/m), damping (Ns/m), and penetration depth (m). So, is there a formula or some general (rough) rules on how to specify the parameters within solimp and solref based on those physical properties with clear units?

Actually, stiffness (N/m), damping (Ns/m), and penetration depth (m) are used in most cases and are often available in the literature. Moreover, these parameters should be generally sufficient to characterize the collision process. As a result, offering a way to convert these physical quantities to the parameters within solimp and solref would be quite useful, especially in the preliminary stage of a design job. Currently, I have no clue how to set the parameters of solimp and solref, even if I have the aforementioned physical quantities at hand.

4. Regarding "MuJoCo normalizes by the contact inertia (inertia projected into the contact frame)," I suspect that the contact inertia would change when the robot configuration changes, right? If so, does this mean that one has to update the parameters for solimp and solref manually? If so, how can the users determine their values? If not, it seems that one still can't directly specify the two parameters of solref as the physical properties stiffness (N/m) and damping (Ns/m) directly; otherwise, cubes with different masses would behave the same, which contradicts reality. So what can we do if we want to achieve realistic behavior?

5. If one has no choice but to do system identification to get a rough value for the parameters for solimp and solref, what parameters are to be identified, and how? Are there any references for this?

6. Personally, I think that offering an approach to use user defined models for the contact would be the most convenient and advantageous way to solve all the issues mentioned above.

ghweili, 2024-06-16:
In case some one is interested in the performance of the cube with different masses, below is the snapshot of the tests we made Based on a demonstration file from Mathematica. The initial displacement of the ODE is set to 0, while the initial velocity is set as 2 for all experiments. Then, we change the mass and conducted several different tests. The curves in the right bottom would be different for different masses, although it seems that the shapes seems to remain the same, with the ``sizes'' of the curves "zoomed" in both axes simultaneously. 

I guess this is the reason why Mujoco chose to do the normalization, although it is not clear to me whether the same conclusion can be made for the more complex contact model in Mujoco.

Considering that the user would be interested in the realistic behavior in the simulations, personally I pro for the option of making the simulation mass dependent, or at least offer the users the option to choose it so, or offer the users a way to customize the contact model. 

[image]