google-deepmind/mujoco #1743: Theoretically calculated contact forces disagree with Mujocos contact forces in a simple example?

kurtenkera, 2024-06-18:
I'm trying to verify the accuracy of the contact forces calculated by MuJoCo using a simple example of a box falling (see video below) and Newton's laws.

https://github.com/google-deepmind/mujoco/assets/127472392/58253fd2-aebc-404a-b511-4fcc29d776da

Specifically, I have a $2 \times 2 \times 2$ metre box with a density $\rho = 65 \, kg/m^3$ whose initial position $\bm{x}_{0}$ at $t = 0$ is $(x_0, y_0, z_0) = (0, 0, 4)$. Note that since the centre of the box is located at this initial position $(0, 0, 4)$, the box free-falls $3m$ under the acceleration of gravity ($g = -9.81 \, m/s^2$) before impacting the ground. 

I calculated the velocity of the box prior to the collision as follows:

\begin{align}
v_f^2
&= v_i^2 + 2*a*d \\ 
&= 0 + 2*9.81*3 \nonumber
\end{align}

which implies that $v_f = 7.67 \, m/s$ prior to contact with the ground. Now, assuming the box decelerates at a constant rate (this is an approximation) when it comes into contact with the ground and comes to a stop (i.e., $v_f = 0 \, m/s$), we can calculate the magnitude of the box's deceleration using equation (1). In this case, $v_0 = 7.67 \, m/s$, $v_f = 0$, and $d = 0.05m$ is the distance over which the box decelerates (this is the minimum penetration distance between the box and the ground returned by MuJoCo). Substituting these numbers into (1) and solving for $a$, we get: 

\begin{align}
\|a\|
&= \|-(v_i)^2/2d\| \nonumber \\
&= 588.29 \, m/s^2 \nonumber
\end{align}

Finally, the **theoretical** net force $F$ on the box is given as $F = ma = 65*8* 588.29 = 305910.28N$. But MuJoCo tells me the forces on **each** corner upon impact is $F_i = 107302.75N$, so the net force (sum of all four forces) on the box is $F = 429211 N$. This is **about 100,000 Newtons greater** than what I calculated using my theoretical approach. **Which is correct?**

The plot below shows the velocity (magnitude) of the box over the simulation.

[image]

It contacts the ground at $t = 0.782s$, and judging by the graph it takes about 0.2 seconds to come to a complete stop. This provides another way to calculate the force on the box:

\begin{align}
F
&= m * (\Delta v / \Delta t) \nonumber \\
&= 520 * (7.67 / 0.2) \nonumber \\
&= 19942N \nonumber
\end{align}

But $19,942N$ is grossly **less than** what MuJoCo calculates $F = 429211 N$. Are my calculations just too rough/approximate here? How can I verify the accuracy of the contact forces in MuJoCo moving forward?

yuvaltassa, 2024-06-18:
Can you fix the LaTeX please? GH LaTeX is idiosyncratic but the preview button is helpful.

I haven't looked carefully yet, but could this be a duplicate of #1560 ?
(please try with sphere)

kurtenkera, 2024-06-18:
I just read [#1560](https://github.com/google-deepmind/mujoco/issues/1560) and the solutions you provided in that post were for a **equilibrium** case of a box/sphere **resting** on a plane. I have no issues with these scenarios, as it is simple to theoretically calculate the contact forces of a box/sphere **at rest** on a plane, and confirm that they sum up to `mass * gravity` (I have already done this with various different geoms).

My issue is that when I attempt to theoretically calculate contact forces in a **dynamical/non-equilibrium scenario** (e.g., a falling box, or a running robot), the more complex method of using Newton's laws (which is absolutely necessary) to sanity check Mujoco's **dynamical** contact forces does not work?

I will try with a sphere as you've asked, but I think it is still important to resolve (or better understand) the issue with the box as I will be scaling up to complex robots, and need to reliably calculate contact forces on these robots and have methods to sanity check these contact forces if possible!

yuvaltassa, 2024-06-19:
Okay, I read the OP a bit more carefully. I feel the biggest issue is in

> assuming the box decelerates at a constant rate

Why would this be a reasonable approximation? For MuJoCo's (and any other reasonable) contact model this is bound to be very, very wrong. If you like, you can calculate the total impulse (integral of force*time) during the collision, but looking at the force at a specific point in time, or the maximum force or whatever is not something that you can't recover without re-implementing the entire model. The force at a given time is a non-trivial function of the penetration depth and velocity.

closed 2024-06-26 by yuvaltassa