google-deepmind/mujoco #1747: Collision attribute of option element is referenced, but it does not exist

salehrayan, 2024-06-19:
I was trying to optimize my model for MJX on GPU according to [this](https://mujoco.readthedocs.io/en/stable/mjx.html#performance-tuning) by making pre defined contact pairs. 

The contact/pair section in XML Reference says:

> This element creates a predefined geom pair, which will be checked for collision if the collision attribute of [option](https://mujoco.readthedocs.io/en/stable/XMLreference.html#option) is set to “all” or “predefined”.

This attribute is referenced a few times but it does not exist in documentation. And parsing an XML with this attribute included will return the usual attribute-not-existing error.

Also the predefined collision pairs will not be the only collisions checked. 
If you have any alternatives for my task I would appreciate it.

yuvaltassa, 2024-06-20:
Ah ya, we need to fix that in the MJX docs. Thanks for bringing that to our attention.

This attribute has been removed, you can disable non-pair contacts in the usual way:

```xml
<default>
  <geom contype="0" conaffinity="0"/>
</default>
```

salehrayan, 2024-07-25:
Have to open this up again. 

the suggested fix worked for cases where a primitive shape is being fitted to my custom meshes, but in cases where I try to import my custom meshes and not a fitted primitive, **and make a mjx model out of them**, encountered the following error:

```
    model_jx = mjx.put_model(model)
  File "D:\rayan_bullet\lib\site-packages\mujoco\mjx\_src\io.py", line 124, in put_model
    fields['mesh_convex'][dataid] = mesh.convex(m, dataid)  # pytype: disable=unsupported-operands
  File "D:\rayan_bullet\lib\site-packages\mujoco\mjx\_src\mesh.py", line 264, in convex
    numvert, numface = graph[0], graph[1]
IndexError: index 0 is out of bounds for axis 0 with size 0
```

which results from having this in the XML:
```
<default>
  <geom contype="0" conaffinity="0"/>
</default>
```

if I remove this it won't give an error but contact pairs are ignored and everything is checked for contact.

yuvaltassa, 2024-09-04:
So the documentation bug from the issue title was fixed in f60ab4492df20ab7b51fc4b2097127cf7d5f4093, so I'm closing this, but I think you have a separate MJX-related issue? If yes, please open another issue with the appropriate description.

Thanks!

closed 2024-09-04 by yuvaltassa