google-deepmind/mujoco #1748: When to use the MuJoCo viewer or renderer, in particular in combination with MJX

davidbschwartz, 2024-06-20:
Hi,

I'm a student and I'm trying to use MuJoCo for a space robotics application.

I have a few questions about the MuJoCo viewer and renderer:

- Is it possible to add geometries to the renderer in a similar fashion to `viewer.user_scn.geoms`, or do you have to change the scene by adapting `renderer.scene.geoms`? If so, is the scene reset again after calling `renderer.update_scene(data, cam)`?
- Is there a way to save an image of what happens in the viewer, similar to `renderer.render().copy()`?
- Is it possible to use the MuJoCo viewer in combination with MJX? Because if I understand correctly, after using the step function in MJX, you would have to re-allocate `mujoco_data` using `mjx.get_data(mujoco_model, mjx_data)` (at least that is the way with the renderer), but the changes in the data do not seem to be transmitted to the passive viewer.
- What is the best way to change `mjx_data.ctrl` field since it is frozen? Would that be something along the lines of `mjx_data = mjx_data.replace(ctrl=jax.numpy.asarray(your_ctrl))`?

Thank you very much in advance.

erikfrey, 2024-06-20:
Hello - I can answer your last two questions:

* Is it possible to use the MuJoCo viewer in combination with MJX?

Yes, see an example of MJX hooked up to drive the simulate viewer here:

https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/viewer.py

You can start this via the command `mjx-viewer` which is installed in your path when you `pip install mujoco-mjx`.

* What is the best way to change mjx_data.ctrl field since it is frozen?

Yes, that's right.  JAX is designed to be [functionally pure](https://jax.readthedocs.io/en/latest/notebooks/Common_Gotchas_in_JAX.html#pure-functions) and that extends to its pytrees - `replace` ensures you're creating a copy rather than modifying the original.

yuvaltassa, 2024-06-20:
Regarding the first question: Yes. mjv_updateScene completely redraws the scene from scratch each time. It should really be called mjv_drawScene since there is no state from the previous call, "update" is confusing.

Regarding the second question, I'm not sure, maybe @kevinzakka knows?

davidbschwartz, 2024-06-21:
Thank you very much for your quick responses. Using `mjx.get_data_into(d, m, dx)` fixed the visualisation issue in MJX.

closed 2024-08-30 by kevinzakka

ucacaxm, 2024-10-29:
Hi,

I saw the example of viewer abow from Erikfrey, thank you.
I wanted to write a simpler code like this one:

```
import jax
import mujoco
from mujoco import mjx
import mujoco_viewer

mj_model = mujoco.MjModel.from_xml_path("data/cubes.xml")
mj_data = mujoco.MjData(mj_model)
viewer = mujoco_viewer.MujocoViewer(mj_model, mj_data)

mjx_model = mjx.device_put(mj_model)
mjx_data = mjx.device_put(mj_data)
jit_step = jax.jit(mjx.step)
mjx_data = mjx.put_data(mj_model, mj_data)

with_mjx = False
while viewer.is_alive:    
    if not with_mjx:
        # ok without MJX
        mujoco.mj_step(mj_model, mj_data)
    else:
        # does not work with MJX
        mjx_data = jit_step(mjx_model, mjx_data)
        mj_data = mjx.get_data(mj_model, mjx_data)
    viewer.render()
viewer.close()
```

It works without MJX but with MJX the scene is not displayed.
I transfer the Jax-MJX data to reguler mujoco data with get_data but that doesn't seem to be enough ?

Thanks,

davidbschwartz, 2024-10-29:
Hi! Not sure why the code using `mujoco_viewer` does not work, but this should:
```
import jax
import mujoco
from mujoco import mjx

mj_model = mujoco.MjModel.from_xml_path("data/cubes.xml")
mj_data = mujoco.MjData(mj_model)

mjx_model = mjx.put_model(mj_model)  # mjx.device_put was deprecated
mjx_data = mjx.put_data(mj_model, mj_data)
jit_step = jax.jit(mjx.step)

viewer = mujoco.viewer.launch_passive(mj_model, mj_data)
with_mjx = False
with viewer:
    if not with_mjx:
        mujoco.mj_step(mj_model, mj_data)
    else:
        mjx_data = jit_step(mjx_model, mjx_data)
        mj_data = mjx.get_data(mj_model, mjx_data)
viewer.close()
```