google-deepmind/mujoco #1750: How to Preserve Initial Shape of a Flexible Body in MuJoCo Simulation?

gleitn96, 2024-06-21:
Hi,

I'm a phd student and I'm trying to use MuJoCo for simulating a consignment with flexible packaging and rigid content.

I'm looking for some help with preserving the initial shape of a flexible body in my simulation.

My problem is that the flexible Packaging will not preserve its initial state and instead blows up like a balloon as shown in the attached picture. Is there a possibility to preserve the shape of the body until there is an outer force acting upon it, similar to how it's done in FEM?

[image]

Here is the XML for my issue:

<mujoco model="Consignment">
  <option timestep="0.000005" solver="CG" iterations="30" tolerance="1e-6" />

  <extension>
    <plugin plugin="mujoco.elasticity.shell" />
  </extension>

  <asset>
    <texture name="plane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8" />
    <material name="plane" reflectance="0.3" texture="plane" texrepeat="1 1" texuniform="true" />
    
    <mesh name="Content" file="Content.stl" />
    <material name="Content_material" rgba="0.8 0.6 0.4 1" />
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 0" size="0 0 .25" type="plane" material="plane" condim="3" />

    
    <flexcomp name="Packaging" type="mesh" radius="0.0002" dim="2" pos="0 0 0.2"
      file="Packaging.obj" scale="0.001 0.001 0.001" mass="0.01">
      <contact selfcollide="none" internal="false"/>
      <edge equality="false" damping="0.01" />
      <plugin plugin="mujoco.elasticity.shell">
        <config key="poisson" value="0.1" />
        <config key="thickness" value="0.001" />
        <config key="young" value="100e6" />       
      </plugin>
    </flexcomp>

    
    <body name="Content" pos="0 0.01 0.138" >
      <geom type="mesh" mesh="Content" material="Content_material" condim="3" mass="0.2" />
      <joint type="free" damping="1" />
    </body>

    
    <light directional="true" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1"
      castshadow="false" />
    <light directional="false" diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 0 4" dir="0 0 -1" />
  </worldbody>
</mujoco>

The full XML, the Packaging.obj, and Content.stl are in the attached .zip file.

Thank you for your help!

[Consignment.zip](https://github.com/user-attachments/files/15924401/Consignment.zip)

tina0425, 2024-07-18:
may you can try to chage the radius="0.0001"

quagla, 2024-07-31:
@gleitn96 is this solved?

gleitn96, 2024-07-31:
@quagla No, changing the radius hasn't resolved the issue. Despite this adjustment, the flexible packaging still fails to preserve its initial shape and expands like a balloon.

mxhrnxz, 2025-06-17:
@quagla  could you please give us your opinion on this? 
I am having the same issue.
thanks.