google-deepmind/mujoco #1751: Unexpected joint contact within a robot arm

edwardjjj, 2024-06-21:
Hi,

I'm a student and I'm trying to use MuJoCo for research.

I'm looking for some help with unexpected contact points between joints

Here is a model which explains my question:

<details>
  <summary>Robot XML</summary>

```XML
<mujoco model="RV4FRL">
  <compiler angle="radian" inertiafromgeom="true" texturedir="textures/" meshdir="meshes/"/>
  <default>
    <site rgba="1.0 0.0 0.7 1.0"/>
    <geom contype='1' conaffinity='1' condim='4' margin="1e-3" solref=".01 1" solimp=".95 .95 .01"/>
    <joint armature="0.01" damping=".1"/>
    <default class="RV4FRL">
        <joint damping='1e+2'/>
    </default>
  </default>
  <asset>
    <!--Meshes-->
    <mesh name="base" file="RV-4FR_BASE.stl" scale="0.001 0.001 0.001"/>
    <mesh name="shoulder" file="RV-4F_SHOULDER.stl" scale="0.001 0.001 0.001"/>
    <mesh name="upperarm" file="RV-4FL_NO1ARM.stl" scale="0.001 0.001 0.001"/>
    <mesh name="elbow" file="RV-4F_ELBOW.stl" scale="0.001 0.001 0.001"/>
    <mesh name="forearm" file="RV-4FL_NO2ARM.stl" scale="0.001 0.001 0.001"/>
    <mesh name="wrist1" file="RV-4F-SH_WRIST.stl" scale="0.001 0.001 0.001"/>
    <mesh name="wrist2" file="RV-4F-SH_HAND_FLANGE.stl" scale="0.001 0.001 0.001"/>
    <!--Materials-->
    <material name="rv4flr_mat" shininess="0.3" specular="0.75" reflectance="0.5" rgba="1.2 1.23 1.26 1"/>
    <material name="rv4flr_hand_mat" shininess="0.1" specular="0.5" reflectance="0.2" rgba="0.8 0.8 0.8 1"/>
    <material name="bench_mat" shininess="0.1" specular="0.5" reflectance="0.2" rgba="0.8 0.8 0.8 1"/>
  </asset>
  <worldbody>
    <!--Robot RV4FLR-->
    <body name="box_link" pos="-0.0325 0 0" euler="0 0 0">
        <geom size="0.1375 0.160 0.020" name="bench" type="box" material="bench_mat"/>
        <inertial pos="0 0 0" mass="1000" diaginertia="0 0 0"/>
        <body name="base_link" pos="0 0 0.02">
            <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
            <geom type="mesh" name="base_link" mesh="base" euler="1.5707963267948966 3.141592653589793 0" material="rv4flr_mat"/>
            <body name="shoulder_link" pos="0 0 .350">
                <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
                <joint name="shoulder_pan_joint" class="RV4FRL" pos="0 0 0" axis="0 0 1" limited="true"
                       range="-4.18799 4.18799"/>
                <geom type="mesh" mesh="shoulder" name="shoulder_link" euler="1.5707963267948966 3.141592653589793 0" material="rv4flr_mat"/>
                <body name="upper_arm_link" pos="0 0 0">
                    <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
                    <joint name="upperarm_joint" class="RV4FRL" pos="0 0 0" axis="0 1 0" limited="true" range="-2.0944 2.0944"/>
                    <geom type="mesh" mesh="upperarm" name="upper_arm_link" euler="1.5707963267948966 3.141592653589793 0" material="rv4flr_mat"/>
                    <body name="elbow" pos="0 0 .31">
                        <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
                        <joint name="elbow_joint" class="RV4FRL" pos="0 0 0" axis="0 1 0"
                               limited="true" range="0 2.86234"/>
                        <geom type="mesh" mesh="elbow" name="elbow" euler="1.5707963267948966 3.141592653589793 -1.5707963267948966" material="rv4flr_mat"/>
                        <body name="forearm" pos="-.050 0 0.335">
                            <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
                            <joint name="forearm_joint" class="RV4FRL" pos="0 0 0" axis="0 0 1" limited="true" range="-3.49066 3.49066"/>
                            <geom type="mesh" mesh="forearm" name="forearm" euler="1.5707963267948966 0 -1.5707963267948966" material="rv4flr_mat"/>
                            <body name="wrist_1_link" pos="0 0 0">
                                <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
                                <joint name="wrist_1_joint" class="RV4FRL" pos="0 0 0" axis="0 1 0" limited="true" range="-2.0944 2.0944"/>
                                <geom type="mesh" mesh="wrist1" name="wrist_1_link" euler="1.5707963267948966 0 -1.5707963267948966" material="rv4flr_mat"/>
                                <body name="wrist_2_link" pos="0 0 0.125">
                                    <site name="wrist_site"></site>
                                    <inertial pos="0 0 0" mass="1" diaginertia="0.0072 0.00443333 0.00443333"/>
                                    <joint name="wrist_2_joint" class="RV4FRL" pos="0 0 0" axis="0 0 1" limited="true" range="-6.28319 6.28319"/>
                                    <geom type="mesh" mesh="wrist2" name="wrist_2_link" euler="0 0 -1.5707963267948966" material="rv4flr_hand_mat"/>
                                </body>
                            </body>
                        </body>
                    </body>
                </body>
            </body>
        </body>
    </body>
  </worldbody>
  <!-- Actuators -->
  <actuator>
    <position gear="1" joint="shoulder_pan_joint" kp="3e+6" name="shoulder_pan_joint" ctrlrange="-4.18799 4.18799"></position>
    <position gear="1" joint="upperarm_joint" kp="2e+4" name="upperarm_joint" ctrlrange="-2.0944 2.0944"></position>
    <position gear="1" joint="elbow_joint" kp="3e+5" name="elbow_joint" ctrlrange="0 2.86234"></position>
    <position gear="1" joint="forearm_joint" kp="3e+5" name="forearm_joint" ctrlrange="-3.49066 3.49066"></position>
    <position gear="1" joint="wrist_1_joint" kp="1e+4" name="wrist_1_joint" ctrlrange="-2.0944 2.0944"></position>
    <position gear="1" joint="wrist_2_joint" kp="1e+4" name="wrist_2_joint" ctrlrange="-6.28319 6.28319"></position>
  </actuator>
  <!-- <contact> -->
  <!--   <exclude body1='base_link' body2='shoulder_link'/> -->
  <!--   <exclude body1='shoulder_link' body2='upper_arm_link'/> -->
  <!--   <exclude body1='upper_arm_link' body2='elbow'/> -->
  <!--   <exclude body1='elbow' body2='forearm'/> -->
  <!--   <exclude body1='forearm' body2='wrist_1_link'/> -->
  <!--   <exclude body1='wrist_1_link' body2='wrist_2_link'/> -->
  <!-- </contact> -->
</mujoco>

```
</details>
I tried add the contact section, but it did not resolve the issue, only makes the sim unstable.

Here is a screenshot / video, illustrating my question:
[image]
Here is the mesh files
[meshes.zip](https://github.com/user-attachments/files/15928431/meshes.zip)

HoangGiang93, 2024-06-26:
Because your meshes are not convex enough, check out the Convex Hull option to see the actual collision hull. You can make the link transparent to see the contact clearer.
https://github.com/google-deepmind/mujoco/assets/64316740/17563fe9-b6c2-4fab-b4fa-d6ac9b4abaa1

edwardjjj, 2024-06-27:
> Because your meshes are not convex enough, check out the Convex Hull option to see the actual collision hull. You can make the link transparent to see the contact clearer. https://github.com/google-deepmind/mujoco/assets/64316740/17563fe9-b6c2-4fab-b4fa-d6ac9b4abaa1

Thank you for the demonstration. So the easiest way is to just modify the stl files to make them more convex?

HoangGiang93, 2024-06-27:
Yes, I recommend to split it into multiple convex meshes.

Balint-H, 2024-06-27:
Alternatively you can just put the two offending geoms into an [`<exclude/>`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact-exclude) pair. Enable label of contacts to display which geoms are colliding, once contact rendering is also enabled.

edwardjjj, 2024-06-27:
> You can try to disable contacts with [`<exclude/>`](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact-exclude) between only the offending geoms, if you label contacts in simulate after rendering it will display which two geoms are touching.

Hi, I tried to add the following `<exclude/>`:
```
  <contact>
    <exclude body1='base_link' body2='shoulder_link'/>
    <exclude body1='upper_arm_link' body2='forearm'/>
  </contact>
```
This makes the simulator unstable, with the following error message:

`ARNING: Nan, Inf or huge value in QACC at DOF 0. The simulation is unstable. Time = 0.0160.`

Here is a video clip.
https://github.com/google-deepmind/mujoco/assets/76624665/2f382ab6-d442-4216-80fa-e9ccb082ac15

edwardjjj, 2024-06-28:
> Hi, I tried to add the following `<exclude/>`:
>
> ```
>   <contact>
>     <exclude body1='base_link' body2='shoulder_link'/>
>     <exclude body1='upper_arm_link' body2='forearm'/>
>   </contact>
> ```
>
> This [...]

I found it's the actuator parameters that is messing with the sim stability. I've reduced the kp of the actuators and now it works as expected.

closed 2024-06-28 by edwardjjj