google-deepmind/mujoco #1755: How to Fix Position and Orientation Error

binbinbai, 2024-06-23:
Hello,

I am using MuJoCo 210 to control a robotic hand by setting the target position as the control signal. However, I have noticed a consistent position and orientation error between the commanded and the actual positions.

Would it be better to upgrade to a newer version of MuJoCo or adjust the motor gains to reduce this error?

Thank you for your assistance!

yuvaltassa, 2024-06-24:
I'm assuming you're using position actuators. There will always be an error, position actuators are springs and the control signal sets the setpoint of the spring. When the target is achieved the force is 0 by definition, so you'll never get 0 error. Increasing the gains will reduce the error but never take it to 0.

That said, upgrading to the latest version is probably a good idea regardless.

closed 2024-06-24 by yuvaltassa

binbinbai, 2024-07-02:
Thank you so much! @yuvaltassa