google-deepmind/mujoco #1757: Decorative/Visual-only robot rendering for planning, debugging and rich visualizations.

Danfoa, 2024-06-25:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for learning legged locomotion using Koopman operator theory and symmetry exploitation. 

I'm looking for some help with some visualization tools I require. Specifically I am interested in rendering (one or multiple) "ghost"/"visual-only"/"decorative" version of a robot, such as what is done in Rviz. 
  
[image]

While this can be used for visualizing motion planning, I am interested in transfering my PhD project repository [MorphoSymm](https://github.com/Danfoa/MorphoSymm) to Mujoco, and need this feature to dynamically visualize the symmetric configuration of a robot, but simulate only one of the robots/symmetric-state-configurations. That is in the example below, to simulate one robot, but visualize all its symmetric configurations. 

[image]

I had an idea to do this on the python API, but I cannot seem to get access to the meshes of the original mjModel. 

```python
[...]
mjModel: MjModel = mujoco.MjModel.from_xml_path(str(model_file_path))
mjData: MjData = mujoco.MjData(self.mjModel)
# Define optional ghost robot data. 
ghost_mjData: MjData = mujoco.MjData(self.mjModel)
[...]
# Use forward kinematics to populate the ghost robot data 
ghost_mjData.qpos, ghost_mjData.qvel = qpos, qvel
mujoco.mj_forward(self.mjModel, self.ghost_mjData)
# Generate decorative geoms that I can update at any point in time.
# Create only once and then update 
_geom_ids = render_ghost_robot(viewer, mjModel, ghost_mjData, alpha=0.5,ghost_geom_ids=_geom_ids)
```

Where render function is supposed to create decorative geometries that are copies of the original robot's visual geometries. 

```python
def render_ghost_robot(viewer, mjModel, mjData, alpha=0.5, ghost_geom_ids=None):
    """Render a ghost robot in the MuJoCo viewer."""
    if ghost_geom_ids is None:
        ghost_geom_ids = {}

    # Iterate through all geometries in the model
    for model_geom_id in range(mjModel.ngeom):
        geom_name = mujoco.mj_id2name(mjModel, mujoco.mjtObj.mjOBJ_GEOM, model_geom_id)
        body_id = mjModel.geom_bodyid[model_geom_id]
        body_name = mujoco.mj_id2name(mjModel, mujoco.mjtObj.mjOBJ_BODY, body_id)
        geom_rgba = mjModel.geom_rgba[model_geom_id]

        # Rules to ignore some geometries we are not interested in rendering for visualization
        # Skip collision geometries (alpha == 0)
        if geom_name in ['floor', 'plane', 'world', 'ground']: continue
        if geom_rgba[3] == 0:
            continue

        # Create a name for the geometry (use body name if geom_name is None)
        ghost_geom_name = f"{model_geom_id}:{geom_name if geom_name is not None else body_name}"

        if ghost_geom_name not in ghost_geom_ids:
            # Create a new geometry if it doesn't exist
            viewer.user_scn.ngeom += 1
            model_geom_id = viewer.user_scn.ngeom - 1
            ghost_geom_ids[ghost_geom_name] = model_geom_id

        # Update the geometry with the current position and orientation
        ghost_geom = viewer.user_scn.geoms[ghost_geom_ids[ghost_geom_name]]
        pos = mjData.geom_xpos[model_geom_id]
        mat = mjData.geom_xmat[model_geom_id].reshape(3, 3)
        size = mjModel.geom_size[model_geom_id]

        # Define the color with transparency
        color = geom_rgba.copy()
        color[3] = alpha

        geom_type = mjModel.geom_type[model_geom_id]
        mujoco.mjv_initGeom(
            ghost_geom,
            type=geom_type,
            size=size,
            pos=pos,
            mat=mat.flatten(),
            rgba=color
            )

        # Set additional properties. TODO: Here I should be able to set texture, and mesh as in original geeom.
        ghost_geom.dataid = mjModel.geom_dataid[model_geom_id]
        ghost_geom.objtype = mujoco.mjtObj.mjOBJ_GEOM
        ghost_geom.objid = -1 # For decoration
        ghost_geom.category = mujoco.mjtCatBit.mjCAT_DECOR

        print(f"Rendering ghost robot: {ghost_geom_name} at position {pos} with color {color}")

    return ghost_geom_ids
```

However, the loaded mesh is not the correct one and most of the relevant geometry attributes are not accesible through the mjModel, nor any mujoco function. 

Is there a way to get the Geom instance from the mjModel, such that I can copy the geom attributes? how else should I ensure the appropiate mesh, and geom parameters are copied?

yuvaltassa, 2024-06-26:
The general approach seems correct, but I suspect you have some wrong assumptions somewhere, but it's a bit hard to see what they are. This line seems suspicious:

> model_geom_id = viewer.user_scn.ngeom - 1 

as it seems to imply that there is a relationship between the index of an `mjvGeom` in the scene and the geom id in `mjModel`.

> Is there a way to get the Geom instance from the mjModel, such that I can copy the geom attributes? how else should I ensure the appropiate mesh, and geom parameters are copied?

Why are you not just copying the values from the original `mjvGeom` that you are copying?

Try asking very specific questions about the semantics of `mjModel` geoms and the `mjvGeoms` in `mjvScene` and we should hopefully make progress from there.

Danfoa, 2024-06-27:
> The general approach seems correct, but I suspect you have some wrong assumptions somewhere, but it's a bit hard to see what they are. This line seems suspicious:
>
>
> as it seems to imply that there [...]

Indeed this is a typo, sorry.

> > Is there a way to get the Geom instance from the mjModel, such that I can copy the geom attributes? how else should I ensure the appropiate mesh, and geom parameters are copied?
> 
> Why are you not just copying the values from the original `mjvGeom` that you are copying?

Mainly because I dont know how to obtain the `mjvGeom` instance in the mujoco API (I am a new mujoco user), and I cannot seem to find the appropiate documentation entry for this. What I have done thus far is to access the geom attributes through the mjModel.

> Try asking very specific questions about the semantics of `mjModel` geoms and the `mjvGeoms` in `mjvScene` and we should hopefully make progress from there.

Say I have the `geom id` and `geom name` in the mjModel, how can I get the `mjvGeom` instance, holding the mesh (data or path), texture, material, etc.?

yuvaltassa, 2024-06-27:
Geoms in mjModel (`mjOBJ_GEOM`s) and `mjvGeoms` are different. The latter is anything in your scene. If you are visualizing contact forces, each arrow is an `mjvGeom` but has no correspondence to an `mjModel` geom. The mjModel ids live in `mjvGeom.objid`, if `mjvGeom.objtype == mjOBJ_GEOM`

closed 2024-06-27 by yuvaltassa

Danfoa, 2024-06-28:
Hi @yuvaltassa,

I understand the difference between mjModel geoms and the scene `mjvGeoms`. My main problem is that I want to create a `mjvGeom` that copies the the "visual" geometries of the robot's model, and I do not know how to do this copying. 

Specifically, having identified the mjModel geometry to copy, I do not know how to instanciate a `mjvGeom` copy of this original geometry, because I do not know how to set the following `mjvGeom` attributes:

```
struct mjvGeom_ {                 // abstract geom
 [...] 
  int      dataid;                // mesh, hfield or plane id; -1: none
  int      objtype;               // mujoco object type; mjOBJ_UNKNOWN for decor
  int      objid;                 // mujoco object id; -1 for decor
  int      texid;                 // texture id; -1: no texture
  int      texuniform;            // uniform cube mapping
  int      texcoord;              // mesh or flex geom has texture coordinates
 [...]
};
typedef struct mjvGeom_ mjvGeom;
```
Say the mjModel geom I want to copy is a mesh type geometry (e.g., the base of my quadruped robot), how can I obtain the base's mesh `dataid`, `objid`, `textid`, ... ?. I assume these values exist somewhere within the viewer, model, or other mujoco class, considering that the mesh was loaded on the mjModel. 

It is unclear to me how to access/list/use these values since from the `mjModel` instance, for a mesh type geometry, the user only has access to:

```
struct mjModel_ {
  [...]
  // meshes
  int*      mesh_vertadr;         // first vertex address                     (nmesh x 1)
  int*      mesh_vertnum;         // number of vertices                       (nmesh x 1)
  int*      mesh_faceadr;         // first face address                       (nmesh x 1)
  int*      mesh_facenum;         // number of faces                          (nmesh x 1)
  int*      mesh_bvhadr;          // address of bvh root                      (nmesh x 1)
  int*      mesh_bvhnum;          // number of bvh                            (nmesh x 1)
  int*      mesh_normaladr;       // first normal address                     (nmesh x 1)
  int*      mesh_normalnum;       // number of normals                        (nmesh x 1)
  int*      mesh_texcoordadr;     // texcoord data address; -1: no texcoord   (nmesh x 1)
  int*      mesh_texcoordnum;     // number of texcoord                       (nmesh x 1)
  int*      mesh_graphadr;        // graph data address; -1: no graph         (nmesh x 1)
  float*    mesh_vert;            // vertex positions for all meshes          (nmeshvert x 3)
  float*    mesh_normal;          // normals for all meshes                   (nmeshnormal x 3)
  float*    mesh_texcoord;        // vertex texcoords for all meshes          (nmeshtexcoord x 2)
  int*      mesh_face;            // vertex face data                         (nmeshface x 3)
  int*      mesh_facenormal;      // normal face data                         (nmeshface x 3)
  int*      mesh_facetexcoord;    // texture face data                        (nmeshface x 3)
  int*      mesh_graph;           // convex graph data                        (nmeshgraph x 1)
  mjtNum*   mesh_scale;           // scaling applied to asset vertices        (nmesh x 3)
  mjtNum*   mesh_pos;             // translation applied to asset vertices    (nmesh x 3)
  mjtNum*   mesh_quat;            // rotation applied to asset vertices       (nmesh x 4)
  int*      mesh_pathadr;         // address of asset path for mesh; -1: none (nmesh x 1)
  [...]
```
That is, updating the proposed render function, I need assistance with the fields marked by ??

```python
def render_ghost_robot(viewer, mjModel, mjData, alpha=0.5, ghost_geom_scn_ids=None):
    """Render a ghost robot in the MuJoCo viewer."""
    if ghost_geom_scn_ids is None:
        ghost_geom_scn_ids = {}

    # Iterate through all geometries in the model
    for model_geom_id in range(mjModel.ngeom):
        geom_name = mujoco.mj_id2name(mjModel, mujoco.mjtObj.mjOBJ_GEOM, model_geom_id)
        body_id = mjModel.geom_bodyid[model_geom_id]
        body_name = mujoco.mj_id2name(mjModel, mujoco.mjtObj.mjOBJ_BODY, body_id)
        geom_rgba = mjModel.geom_rgba[model_geom_id]

        # Rules to ignore some geometries we are not interested in rendering for visualization
        # Skip collision geometries (alpha == 0)
        if geom_name in ['floor', 'plane', 'world', 'ground']: continue
        if geom_rgba[3] == 0:
[... 10 lines cut ...]
        # mjvGeom structure which will copy the mjModel geom properties
        ghost_geom = viewer.user_scn.geoms[ghost_geom_scn_ids[ghost_geom_name]]
        # Define the color with transparency
        color = geom_rgba.copy()
        color[3] = alpha
        # Update the geometry with the current position and orientation
        mujoco.mjv_initGeom(
            ghost_geom,
            type=mjModel.geom_type[model_geom_id],
            size=mjModel.geom_size[model_geom_id],
            pos=mjData.geom_xpos[model_geom_id],
            mat=mjData.geom_xmat[model_geom_id],
            rgba=color,
            category=mujoco.mjtCatBit.mjCAT_DECOR,   # This Geom should be treated as a decoration 
            dataid=None,   # ?? mesh, hfield or plane id; -1: none
            objtype=None,  # ?? mujoco object type; mjOBJ_UNKNOWN for decor
            objid=None,    # ?? object id within the object type
            texid=None,    # ?? texture id; -1: no texture
            textuniform=None,  # ??  uniform cube mapping
            texcoord=None,  # ?? texture coordinates
            segid=-1,
            )

    return ghost_geom_scn_ids
```

yuvaltassa, 2024-06-28:
I'm confused by your confusion 🙂 

First, the `mjModel` id's are simply the indices in `mjModel`. You see on the right of the `mjModel` docstrings you have e.g. `(nmesh x 1)`? That means that the mesh id's are in the range `[0 ... nmesh-1]`. Makes sense?

Second, since you are basically copying `mjvGeom`s, why can't you just copy the values?
This below is very inefficient, but should give you an idea:
```py
# given model_geom_id
for i in range(viewer.user_scn.ngeom)
  mjv_geom = viewer.user_scn.geoms[i]
  if mjv_geom.objtype == mujoco.mjtOBJ.mjOBJ_GEOM:
    # this visual geom corresponds to a model geom, not a decorative thing
    if mjv_geom.objid == model_geom_id:
      # this is the mjv_geom I was looking for
      # I can copy it into a new translucent geom and reposition it or whatever
```

Hope this helps!

yuvaltassa, 2024-06-28:
Relatedly, [here](https://github.com/google-deepmind/mujoco/blob/0e19722c616daf465fbaf94c231a20d381b487c6/python/mujoco/renderer.py#L217) is the code in the renderer that converts the `mjvScene` geom id's to `mjModel` id's and types (`mjtObj`), if you want to see something less inefficient 🙂

Danfoa, 2024-07-01:
Hi @yuvaltassa 

Thank you for your patience and kind explanations. 

I had tried in the past to access the robot geometries using something similar as you suggested `mjv_geom = 
viewer.user_scn.geoms[i]`. However, after creating the scene and loading my robot, I am unable to do this since I get `self.viewer.user_scn.ngeom=0`. 

However, if I add some decorative geometries, these geoms do increase the ngeom count in the scene. Therefore originally I thought the `user_scn` geoms only accounted for the decorative ones. Hence my original question **how do I get the instances `mjv_geom` of a robot visual geometries?.

This is the code I use to create the scene for the first time (in passive mode):
```python
[...]
    def render(self, mode='human', ghost_configs=None, tint_robot=False):
        [...]
        if self.viewer is None:
            self.viewer = mujoco.viewer.launch_passive(
                self.mjModel, self.mjData, show_left_ui=False, show_right_ui=False,
                key_callback=lambda x: self._key_callback(x)
                )

            mujoco.mjv_defaultFreeCamera(self.mjModel, self.viewer.cam)
            self.viewer.user_scn.flags[mujoco.mjtRndFlag.mjRND_SHADOW] = 0
            
            self.viewer.sync()
            ngeoms = self.viewer.user_scn.ngeom
            print(f"Number of geometries: {ngeoms}")  # This is always 0.

        # Define/Update markers for visualization of desired and current base velocity
        self._geom_ids['ref_dr_B_vec'] = render_vector(
            self.viewer, vector=ref_base_lin_vel_B, pos=ref_vec_pos, scale=ref_vec_scale, color=ref_vel_vec_color,
            geom_id=self._geom_ids.get('ref_dr_B_vec', -1)
            )
        self._geom_ids['dr_B_vec'] = render_vector(
            self.viewer, vector=dr_B, pos=vec_pos, scale=vec_scale, color=vel_vec_color,
            geom_id=self._geom_ids.get('dr_B_vec', -1)
            )

        ngeoms = self.viewer.user_scn.ngeom
        print(f"Number of geometries: {ngeoms}")     # Now there are two (the two vectors)
```
And this is a screenshot of the scene 
[image]

yuvaltassa, 2024-07-01:
Ah hold on. Sorry, I just realized that you're using the passive viewer. From the docs [here](https://mujoco.readthedocs.io/en/latest/python.html#passive-viewer): "This is separate from the mjvScene that the viewer uses internally to render the final scene, and is entirely under the user’s control."

So you don't actually have access to the internal scene, only the `user_scn`, which is empty. @saran-t can correct me if I'm wrong, but what I think you want to do is:
1. Create your own, local, mjvScene for the full model by doing by `my_scene = mujoco.MjvScene(model, MAX_GEOM)` 
2. Call `mjv_updateScene` to populate it after every step.
3. Copy/modify whatever you need from `my_scn` into `user_scn`.

saran-t, 2024-07-01:
What @yuvaltassa suggested should work, but better still you should be able to just `mjv_updateScene` directly into `viewer.user_scn`.

yuvaltassa, 2024-07-01:
I think the OP wants only _some_ of the geoms from the main scene, while `mjv_updateScene` would basically duplicate everything, but ya, maybe that could work for them.

Danfoa, 2024-07-02:
Thank you very much @yuvaltassa and @saran-t for the prompt, kind, and comprehensive support. 

Attached the final version of the code, which is working as expected and enable users to introduce one (or multiple ghost/decorative robots). 

Finding the geometries can be run only in the first call of the function if desired, such that consecutive function calls update the position/orientation of the geometries.
 
```python
mj_model = ... 
mj_data = ...  # Used to store the data of the robot/model to simulate
ghost_mj_data = mujoco.MjData(mjModel)  # Used to store the config of ghost/decorative robots

[...]
#  Use FK to update the configuration of the ghost robot
ghost_geoms = {}
mujoco.mj_forward(mj_model, ghost_mj_ata)
ghost_geoms |= render_ghost_robot(viewer,
                                  mj_model,
                                  ghost_mj_ata,
                                  alpha=0.5,
                                  ghost_geoms=ghost_geoms)
```

```python
def render_ghost_robot(viewer: Handle, mj_model: MjModel, mj_data: MjData, alpha=0.5, ghost_geoms: dict | None = None):
    """ Render a ghost robot in the viewer with transparency.
    
    :param viewer: Mujoco Handle to the viewer. Assumed to be in passive mode.
    :param mj_model: Mujoco MjModel containing the Geoms of the robot.
    :param mj_data: Mujoco MjData containing the position (xpos) and orientation (xmat) of the Geoms.
    :param alpha: The transparency of the ghost robot. 0.0 is fully transparent, 1.0 is fully opaque.
    :param ghost_geoms: A dictionary with keys as the decorative geometry ids (idx in `viewer.user_scn.geoms`) and 
        values as the corresponding `MjvGeom` from visual Geometries of the MjModel. If None, the function will
        automatically create the ghost_geoms and return this dictionary so recurrent calls avoid re-creating the
        ghost_geoms. 
    :return: A dictionary with keys as the decorative geometry ids (idx in `viewer.user_scn.geoms`) and values as the 
        corresponding `MjvGeom` from visual Geometries of the MjModel.
    """

    if ghost_geoms is None or len(ghost_geoms) == 0:
        # Hacky way to get the robot Geometries from a "passive" viewer, as we cant get these from the viewer.user_scn
        # See: https://github.com/google-deepmind/mujoco/issues/1757
        scene = mujoco.MjvScene(mj_model, 200)
        mujoco.mjv_updateScene(mj_model, mj_data,  # Create mock scene to get the Geoms
                               mujoco.MjvOption(),
                               None, mujoco.MjvCamera(),
                               mujoco.mjtCatBit.mjCAT_ALL, scene)
        visible_geoms = [g for g in scene.geoms[:scene.ngeom] if g.segid != -1]  # All non-decorative geoms in scene
        visible_objid = np.array([g.objid for g in visible_geoms], np.int32)  # ObjectID = model_geom_id
        visible_objtype = np.array([g.objtype for g in visible_geoms], np.int32)

        ghost_geoms = {}
        for geom, model_geom_id, objtype in zip(visible_geoms, visible_objid, visible_objtype):
            geom_name = mujoco.mj_id2name(mj_model, mujoco.mjtObj.mjOBJ_GEOM, model_geom_id)
            body_id = mj_model.geom_bodyid[model_geom_id]
            body_name = mujoco.mj_id2name(mj_model, mujoco.mjtObj.mjOBJ_BODY, body_id)
            geom_rgba = mj_model.geom_rgba[model_geom_id]

            # Rule set of geoms to ignore. We dont want collision geoms or the floor ______________________________
            ignored_names = ['floor', 'plane', 'world', 'ground']
            if geom_name in ignored_names or body_name in ignored_names or geom_rgba[3] == 0:
                continue
            # ________________________________________________________________________________________________________

            viewer.user_scn.ngeom += 1
            dec_geom_scn_id = viewer.user_scn.ngeom - 1
            ghost_geoms[dec_geom_scn_id] = geom  # Store a map from `viewer.user_scn.geom_id` and model's `MjvGeom`

    for dec_geom_scn_id, geom in ghost_geoms.items():
        geom_model_id = geom.objid
        # Update the color of the geometry with transparency
        geom.rgba[3] = alpha
        # Copy the Body Geom to the decorative geom in viewer.user_scn
        dec_geom = viewer.user_scn.geoms[dec_geom_scn_id]
        mujoco.mjv_initGeom(
            dec_geom, type=geom.type, rgba=geom.rgba, size=geom.size,
            # Use the position and orientation from the mjData
            pos=mj_data.geom_xpos[geom_model_id], mat=mj_data.geom_xmat[geom_model_id].reshape(9),
            )
        # Ensure ghost decorative geometries are ignored in segmentation, collisions, etc.
        dec_geom.category = mujoco.mjtCatBit.mjCAT_DECOR
        dec_geom.segid = -1
        dec_geom.objid = -1
        dec_geom.reflectance = 0.0
        # Copy the mesh and texture attributes
        dec_geom.dataid = geom.dataid
        dec_geom.texid = geom.texid
        dec_geom.texcoord = geom.texcoord
        dec_geom.texrepeat = geom.texrepeat
        dec_geom.texuniform = geom.texuniform
        dec_geom.emission = geom.emission
        dec_geom.specular = geom.specular
        dec_geom.shininess = geom.shininess

    return ghost_geoms
``` 
The result looks as expected: 

[image]

yuvaltassa, 2024-09-07:
So I looked into this again, it is actually easier. Unlike mjv_updateScene which clears the scene, [mjv_addGeoms](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mjv-addgeoms) does not. So after the first call to update scene you can change the state and call mjv_addGeoms, just make sure to disable `mjVIS_STATIC` so you don't render the static geoms multiple times. I think I'll make a new cell in the tutorial colab.

commit 8fa7fe7f (2024-09-10): Add a tutorial section on multiple frames in the same scene. Fixes #1757, #1356.

PiperOrigin-RevId: 672902542
Change-Id: Ie9fae2151354206b1418c129e2bd2ab016695ec3