google-deepmind/mujoco #1760: Object with <flexcomp type="ellipsoid" ...> does not move

jviereck, 2024-06-26:
I want to model a deformable finger tip of my robot hand. I managed to get a deformable cube fall on a plain as defined in this xml:

```xml
<mujoco>
  <option timestep=".001"/>
  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
  </extension>
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
    <geom type="plane" size="1 1 0.1" pos="0 0 0.0" rgba=".9 0 0 1"/>
    <flexcomp type="grid" count="4 4 4" spacing=".1 .1 .1" pos="0 0 0.3" radius=".0" rgba="0 .7 .7 1" name="softbody" dim="3" mass="1">
      <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
      <edge damping="1"/>
      <plugin plugin="mujoco.elasticity.solid">
        <config key="poisson" value="0.2"/>
        <config key="young" value="5e4"/>
      </plugin>
    </flexcomp>
  </worldbody>
</mujoco>
```

However, when I try to change the `flexcomp`'s `type` to `ellipsoid`, I see the ellipsoid in the viewer but the ellipsoid is not falling down. Any idea what is going on?

Here is the xml I try to use:

```xml
<mujoco>
  <option timestep=".001"/>
  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
  </extension>
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
    <geom type="plane" size="1 1 0.1" pos="0 0 0.0" rgba=".9 0 0 1"/>
    <flexcomp type="ellipsoid" count="4 4 4" spacing=".1 .1 .1" pos="0 0 0.3" radius=".0" rgba="0 .7 .7 1" name="softbody" dim="3" mass="1">
      <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
      <edge damping="1"/>
      <plugin plugin="mujoco.elasticity.solid">
        <config key="poisson" value="0.2"/>
        <config key="young" value="5e4"/>
      </plugin>
    </flexcomp>
  </worldbody>
</mujoco>
```

Is there any other change I have to do to get this to work?

quagla, 2024-07-12:
Try putting the flexcomp in a body with a freejoint. Radial flexcomp types are pinned to the center.

closed 2024-07-12 by quagla

jviereck, 2024-09-10:
Thanks quagla! Here is the working example for reference:

```
<mujoco>
  <option timestep=".001"/>
  <extension>
    <plugin plugin="mujoco.elasticity.solid"/>
  </extension>
  <worldbody>
    <light diffuse=".5 .5 .5" pos="0 0 3" dir="0 0 -1"/>
    <geom type="plane" size="1 1 0.1" pos="0 0 0.0" rgba=".9 0 0 1"/>
    <body pos="0 0 0.025" name="soft_parent">
        <flexcomp type="ellipsoid" count="4 4 4" spacing=".1 .1 .1" pos="0 0 0.3" radius=".0" rgba="0 .7 .7 1" name="softbody" dim="3" mass="1">
          <contact condim="3" solref="0.01 1" solimp=".95 .99 .0001" selfcollide="none"/>
          <edge damping="1"/>
          <plugin plugin="mujoco.elasticity.solid">
            <config key="poisson" value="0.2"/>
            <config key="young" value="5e3"/>
          </plugin>
        </flexcomp>
        <inertial mass="1.0" pos="0 0 0" diaginertia="0.001 0.001 0.001"></inertial>
        <joint type="free"></joint>
    </body>
  </worldbody>
</mujoco>
```