google-deepmind/mujoco #1761: Toolbar/buttons disappear after starting mujoco

owenchen0408, 2024-06-27:
Hi everyone, I am new to Mujoco. I am trying to run some example models on Mujoco. However, the tool bar disappears after 1 second of opening the GUI. Im using windows and ver 3.1.6
Here is a image if the GUI after the buttons/tool bar disappeared. I think this has to do with a zoom of the GUI. how can I solve this or is this abug? I didn't do any action before the tool bar or buttons disapear
[image]

Thanks!

yuvaltassa, 2024-06-27:
What GPU do you have?

owenchen0408, 2024-06-27:
AMD RX 6800M

yuvaltassa, 2024-06-27:
duplicate of #639, there is an ongoing issue with AMD GPUs

closed 2024-06-27 by yuvaltassa