google-deepmind/mujoco #1767: Error converting URDF file in Mojoco

Akanksha0401, 2024-06-29:
Dear Colleagues,

I was having trouble in converting the URDF file of Single Leg Hopper in mujoco , I tried adding
 ```
<mujoco> 
	 <compiler meshdir="C:\Users\Akanksha Mishra\Dropbox\My PC (DESKTOP-FFNO6EU)\Downloads\mujoco-3.1.6-windows-x86_64\model\meshes" balanceinertia="true"  discardvisual="false" />
    <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
    </mujoco>
```

but when it gets open in Mujoco, the object falls immediately.

My XML file looks like this-
 ```
<?xml version="1.0" ?>
<robot name="svanm2">
<mujoco> 
	 <compiler meshdir="C:\Users\Akanksha Mishra\Dropbox\My PC (DESKTOP-FFNO6EU)\Downloads\mujoco-3.1.6-windows-x86_64\model\meshes" balanceinertia="true"  discardvisual="false" />
    <worldbody>
    <light pos=".4 -.4 .3" dir="-2 2 -1.5" diffuse=".6 .6 .6"/>
    <light pos="-.2 -.4 .3" dir="1 2 -1.5" diffuse=".6 .6 .6"/>
    <geom type="plane" size=".5 .5 .01"/>
    <body name="ball" pos="0 0 0.05">
      <freejoint/>
      <geom size=".03"/>
    </body>
    <body name="hammer" pos="-.05 0 0.18">
      <joint axis="0 1 0"/>
      <geom type="capsule" size=".01" fromto="0 0 0 -.15 0 0"/>
      <geom type="box" size=".02 .02 .03" pos="-.15 0 0"/>
    </body>
  </worldbody>
    </mujoco>
  <link name="world"/>
  <joint name="fixed" type="fixed">
```
Please guide , what could be a possible solution.

yuvaltassa, 2024-06-30:
There is not enough information in this issue for us to help you.
From a cursory look it seems like "hammer" is supposed to be a child of "ball"?

kevinzakka, 2024-08-15:
Marking as closed due to inactivity.

closed 2024-08-15 by kevinzakka