google-deepmind/mujoco #1768: Why the contact point breaks when an object slides?

2-woo-10, 2024-06-30:
Hello.

I'm a student and I'm going to use Mujoko to get sensor information about situations where objects slide through Mujoko.

Currently, I want to get sensor data on situations where I grab the edge of an object and slide while holding it up.

When enough grip force was not applied to lift the object, the object slipped.

At this time, the contact force and contact point are disconnected when the gripper moves with the object, may I know the reason for this?

The image below is a graph that visualizes the contact force, contact point, and the collected data when I test it.

[image]
[image]
[image]
[image]
[image]

yuvaltassa, 2024-06-30:
Is this with pyramidal or elliptic cones?

2-woo-10, 2024-07-01:
> Is this with pyramidal or elliptic cones?

I am using elliptic cones!!!