google-deepmind/mujoco #1771: How to apply a force on a body ?

AnxSnip, 2024-07-01:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating the stability of a bunch of boxes in a truck cargo. I load models, and I need to know how many will move/ fall.

I'm looking for some help with applying force on a body, here I want to move the cargo simulating the truck moving, but I have some issues on how to do it. I have checked this [issue](https://github.com/google-deepmind/mujoco/discussions/641) but it seems outdated as there is no example from the links. The documentation for xfrc_applied is not very clear and I want to directly be put it in the xml. I ve seen people using actuator/motor, but It does not move.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>
<details>
  <summary>minimal XML</summary>
[container_1.txt](https://github.com/user-attachments/files/16051228/container_1.txt)
</details>

Another question: is it possible to simulate the crushing of an item, like a box is too heavy for boxes at its bottom to handle, and to detect if the items will be crushed, even in cases when  a box is supported by 2 or more others boxes.

Balint-H, 2024-07-02:
Heya! It's pretty fun seeing all those boxes tumble in your example when shaking the container.

If you can make the assumption of the truck's mass/inertia being much larger than the cargo and that the truck has negligible angular acceleration, I think the easiest approach would be to move everything to the truck's reference frame, remove the container's freejoint, and add the -1 times of the truck's intended acceleration to the gravity instead! 

If you can't make either of those assumptions, you can indeed actuate the container. In your case the following modifications need to be made:

1. Change the `condim` of the ground to 1, and its `priority` to 2. This lets the container to smoothly slide around the ground plane. (Alternatively, you change the freejoint to two slide (x and y) and 1 hinge joints (z) for planar movement, then you don't need the ground in the first place)

2. Check out the [documentation of the site attribute](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-general-site). You need to additionally specify the `gear` attribute to define which direction relative to the site frame the forces are applied in. Also check out `refsite` in case you want it in a different frame than the container's (e.g. world frame). You can either set the gearing to be a high value (e.g. "1000 0 0 0 0 0") or the ctrlrange to be greater, as you'll need quite a bit of force to move all these boxes.

For your second question: Check out `touch` sensors, you can read how much force is applied to a body. In your script for your environment you can then threshold this sensor reading, and then either declare the run a failure, or, if you want to keep going but "destroy" the box you could teleport it away by setting its `qpos` (This is one example, you can represent it in other ways as well).

AnxSnip, 2024-07-04:
Hi @Balint-H 
Thank you for your quick response, I wanted to compare the two reference before deciding on which one to choose. And I think the simplest will be the gravity. But is it possible to increase the gravity during simulation ? I remember the value is `model.opt.gravity` and that is not included in the data.
For the touch sensors, I tried to use them but the values don't seems constant, should I place the site to include the whole box? or just the top part of the box?.

Balint-H, 2024-07-04:
Some fields in `model` are editable during runtime without scene recompilation, including gravity! As far as the touch sensors go, if any contact with a box is valid, then the site should cover the whole box.

AnxSnip, 2024-07-04:
Thank you
When I make the site cover the whole box, it detect its own weight. 
For the acceleration I am currently only testing for linear acceleration but it might be great to see for turns as well.

Balint-H, 2024-07-04:
For turns you can get a reasonably good approx with sideways components to model centrifugal "forces" (this approach does miss the period of torque applied to the boxes when a turn is initiated and stopped, but is appropriate when the turn arc is maintained). 

When modelling when an object is crushed I think including its own weight is fair (imagine the truck accelerating so much that a single box could get crushed under its own weight). You can subtract the box's weight from the sensor reading if you are only interested in the excess forces with interactions with other items in the truck.

closed 2024-08-15 by kevinzakka