google-deepmind/mujoco #1775: Title: Block on Rotating Plate Changes Position but Not Angle

z-yf17, 2024-07-02:
Content:

Hi,

I am facing an issue with my MuJoCo simulation and I would appreciate some help.

I have a rotating plate, and when I place a block on it, the block's position changes as the plate rotates, but its angle does not. Ideally, the block should change its angle along with its position as it rotates on the plate.

Why is this happening, and how can I adjust the setup so that the block's angle changes appropriately with the rotation of the plate?

Thank you very much for any assistance!

![Uploading 8fabb63a506d7
[image]
db7b707ba26574d783.png…]()
[image]
[image]

Balint-H, 2024-07-02:
Can you share a minimum working example XML of only your spinning plate and cube where this effect is reproduced?

z-yf17, 2024-07-03:
> Can you share a minimum working example XML of only your spinning plate and cube where this effect is reproduced?

Thanks for your reply.

<body name="rotating_plate" pos="0.8 0 0.5">
    <geom type="cylinder" size="0.3 0.02" rgba="0.3 0.3 0.3 1" friction="2 1 0.005"/>
    <joint name="rotation" type="hinge" axis="0 0 1" pos="0 0 0" limited="false"/>
</body>

<body name="object" pos="0.7 0 0.65">
    <joint name="box" type="free" frictionloss="0.01" />
    <inertial pos="0 0 0" mass="0.05" diaginertia="0.002 0.002 0.002" />
    <geom mass="0.003" solimp="1 1 0.01" solref="0.01 1" friction="1 0.1 0.005" pos="0 0 0" size="0.02 0.02 0.02" type="box" name="box" rgba="0.9 0. 0. 1" />
    </body>

<actuator>
    <velocity joint="rotation" kv="100" ctrlrange="-1 1" />
</actuator>

This is the code related to the problem.

z-yf17, 2024-07-03:
[image]
[image]

Here is the code. Thanks.

Balint-H, 2024-07-03:
Its caused by the large diagonal inertia you set (by a couple orders of magnitude larger than usual for a cube that size) and the frictionloss you specified in the joint (which resists all movement)

Unless you have a specific reason to use inertial (e.g. measurements from CAD software or real life experiments) I recommend using the default inertia calculations and removing the inertial element.