google-deepmind/mujoco #1776: about dofs in mujoco

0noendlearn0, 2024-07-03:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating robot to grab cylinder.

I'm looking for some help with dofs use. when the cylinder's dofs is setted to 6, the cylinder slides on the plane.
Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<cylinder>xml definition:
 <body name="item3" pos="1.738 2.885 0.6829">
       <body name="item3_point" />
      <joint armature="0" axis="1 0 0" damping="0" limited="true" name="item3_joint_x" pos="0 0 0" range="-5. 5." stiffness="0" 
      type="slide" />
      <joint armature="0" axis="0 1 0" damping="0" limited="true" name="item3_joint_y" pos="0 0 0" range="-5. 5." stiffness="0" type="slide"/>
      <joint armature="0" axis="0 0 1" damping="0" limited="true" name="item3_joint_z" pos="0 0 0" range="-2. 2." stiffness="0" type="slide"/>
      <joint armature="0" damping="0" limited="true" pos="0 0 0" stiffness="0" type="ball" />
      <geom name="item3_geom" type="cylinder" size="0.032 0.015" density="50" friction="2 2 2" material="off_white" solimp="1 1 0 0.5 2"/>
    </body>
the plane definition:
<body name="shelf" pos="0 3 -0.5" quat="0.5 0.5 -0.5 -0.5">
      <geom mesh="shelf" material="blue" class="object_visual"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 1.1373 -1.5728" quat="0.707 0 0.707 0" solimp="1 1 0 0.5 2" rgba="1 1 1 0"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 0.43 -1.5728" quat="0.707 0 0.707 0" solimp="1 1 0 0.5 2" rgba="1 1 1 0"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -0.78279" quat="0.5 0.5 0.5 -0.5" solimp="1 1 0 0.5 2" rgba="1 1 1 0"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -2.3628" quat="0.5 0.5 0.5 -0.5" solimp="1 1 0 0.5 2" rgba="1 1 1 0"/>
      <geom mesh="item_basket" material="off_white" class="object_visual" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0 0.77" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0.77" conaffinity="0" contype="0" group="1" />
      <inertial pos="0 0 0" mass="1000" diaginertia="0 0 0" />
  </body>
```

</details>

Here is a screenshot / video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/162142936/f6b6ed54-2f35-4782-b195-a29589e56e53

Balint-H, 2024-07-03:
My recommendations:

- Reset `solimp`s and friction on geoms.
- Switch to `freejoint` on the object.
- Bump `condim` to 6 (to oppose rolling)
- Significantly reduce the timeconstant on `solref`. (Perhaps the most effective)
- Optionally enable [MultiCCD](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-flag-multiccd)

``` xml
<mujoco>
  <visual>
    <map force="0.1" zfar="30"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global offwidth="2560" offheight="1440" elevation="-20" azimuth="120"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1=".3 .5 .7" rgb2="0 0 0" width="32" height="512"/>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1=".1 .2 .3" rgb2=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
  </asset>

  <option><flag multiccd="enable"/></option>
  
  <worldbody>
    <geom name="floor" size="0 0 .05" type="plane" material="grid" condim="3"/>
    <light name="spotlight" mode="targetbodycom" target="item3" diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 -6 4" cutoff="30"/>
   
     <body name="item3" pos="1.738 2.885 0.6829">
      <freejoint/>
      <geom name="item3_geom" type="cylinder" size="0.032 0.015" density="50" solref="0.000002 1" condim="6"/>
    </body>
<body name="shelf" pos="0 3 -0.5" quat="0.5 0.5 -0.5 -0.5">
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 1.1373 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 1"/>
  </body>
  </worldbody>
</mujoco>
```

0noendlearn0, 2024-07-04:
Thanks your reply. this some part according to your suggestions：
   I <default class="object_visual">
      <geom type="mesh" conaffinity="0" contype="0" group="1"/>
    </default>
  <default>
    <geom contype='1' conaffinity='1' condim='4' margin="1e-3" solref=".000001 1" solimp=".99 .99 .01" />
  </default>
<body name="shelf" pos="0 3 -0.5" quat="0.5 0.5 -0.5 -0.5">
      <geom mesh="shelf" material="blue" class="object_visual"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 1.1373 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 0.43 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -0.78279" quat="0.5 0.5 0.5 -0.5" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -2.3628" quat="0.5 0.5 0.5 -0.5" rgba="1 1 1 0" condim="3"/>
      <geom mesh="item_basket" material="off_white" class="object_visual" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0 0.77" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0.77" conaffinity="0" contype="0" group="1" />
      <inertial pos="0 0 0" mass="1000" diaginertia="0 0 0" />
 </body>
 <body name="item3" pos="1.738 2.885 0.6829">
      <body name="item3_point" />
      <freejoint/>
      <geom name="item3_geom" type="cylinder" size="0.032 0.015" density="50" material="off_white" solref="0.000002 1" condim="6"/>
    </body>

the item3 is placed in shelf. I adjust the shelf's box attribute but it works is not good .
Here is a screenshot / video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/162142936/a62c5d23-cdc9-4d23-8dc1-e6f3fb8ec822

add  <option><flag multiccd="enable"/></option>
https://github.com/google-deepmind/mujoco/assets/162142936/dd0b5e49-ded5-47e7-8393-2de77ad87849

I'm looking forward to your reply to solve my difficults.

0noendlearn0, 2024-07-04:
Thanks your reply. this some part according to your suggestions：
minimal xml:
   I <default class="object_visual">
      <geom type="mesh" conaffinity="0" contype="0" group="1"/>
    </default>
  <default>
    <geom contype='1' conaffinity='1' condim='4' margin="1e-3" solref=".000001 1" solimp=".99 .99 .01" />
  </default>
<body name="shelf" pos="0 3 -0.5" quat="0.5 0.5 -0.5 -0.5">
      <geom mesh="shelf" material="blue" class="object_visual"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 1.1373 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 0.43 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -0.78279" quat="0.5 0.5 0.5 -0.5" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -2.3628" quat="0.5 0.5 0.5 -0.5" rgba="1 1 1 0" condim="3"/>
      <geom mesh="item_basket" material="off_white" class="object_visual" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0 0.77" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0.77" conaffinity="0" contype="0" group="1" />
      <inertial pos="0 0 0" mass="1000" diaginertia="0 0 0" />
 </body>
 <body name="item3" pos="1.738 2.885 0.6829">
      <body name="item3_point" />
      <freejoint/>
      <geom name="item3_geom" type="cylinder" size="0.032 0.015" density="50" material="off_white" solref="0.000002 1" condim="6"/>
    </body>

the item3 is placed in shelf. I adjust the shelf's box attribute but it works is not good .
Here is a screenshot / video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/162142936/a62c5d23-cdc9-4d23-8dc1-e6f3fb8ec822

add  <option><flag multiccd="enable"/></option>
https://github.com/google-deepmind/mujoco/assets/162142936/dd0b5e49-ded5-47e7-8393-2de77ad87849

I'm looking forward to your reply to solve my difficults.

Balint-H, 2024-07-04:
The XML is not visible, could you share it?

0noendlearn0, 2024-07-05:
Sorry,the details of this part of the xml are as follows:
<details>
  <summary>minimal XML</summary>

```XML
 <default class="object_visual">
      <geom type="mesh" conaffinity="0" contype="0" group="1"/>
    </default>
  <default>
    <geom contype='1' conaffinity='1' condim='4' margin="1e-3" solref=".000001 1" solimp=".99 .99 .01" />
  </default>
<body name="shelf" pos="0 3 -0.5" quat="0.5 0.5 -0.5 -0.5">
      <geom mesh="shelf" material="blue" class="object_visual"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 1.1373 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.75 0.03 0.28" pos="-0.035 0.43 -1.5728" quat="0.707 0 0.707 0" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -0.78279" quat="0.5 0.5 0.5 -0.5" rgba="1 1 1 0" condim="3"/>
      <geom type="box" size="0.04 0.28 1" pos="-0.035 1 -2.3628" quat="0.5 0.5 0.5 -0.5" rgba="1 1 1 0" condim="3"/>
      <geom mesh="item_basket" material="off_white" class="object_visual" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0 0.77" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0" conaffinity="0" contype="0" group="1" />
      <geom mesh="item_basket" material="off_white" class="object_visual" pos="0 0.7073 0.77" conaffinity="0" contype="0" group="1" />
      <inertial pos="0 0 0" mass="1000" diaginertia="0 0 0" />
 </body>
 <body name="item3" pos="1.738 2.885 0.6829">
      <body name="item3_point" />
      <freejoint/>
      <geom name="item3_geom" type="cylinder" size="0.032 0.015" density="50" material="off_white" solref="0.000002 1" condim="6"/>
    </body>
```
</details>

the item3 is placed in shelf. I adjust the shelf's box attribute but it works is not good .
Here is a  video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/162142936/a62c5d23-cdc9-4d23-8dc1-e6f3fb8ec822

add flag multiccd="enable"，the video:
https://github.com/google-deepmind/mujoco/assets/162142936/dd0b5e49-ded5-47e7-8393-2de77ad87849

I'm looking forward to your reply to solve my difficults.

Balint-H, 2024-07-08:
Set the `priority` of the item object's geom to a higher value, to ensure its solver parameters are used. Make sure timestep is not too high (I have stable contacts at 0.002 for example).

closed 2024-07-31 by quagla