google-deepmind/mujoco #1778: Rendering with OpenGL fails with "ARB_framebuffer_object required"

SkytAsul, 2024-07-03:
Hi,

I'm a student in research and I'm trying to use MuJoCo for simulating soft body touch and rendering haptic feedback, all of this in Virtual Reality.

I'm looking for some help with the render pipeline using OpenGL.

To integrate the virtual headsets in MuJoCo, I am trying to use the pyopenxr library. It uses glfw under the hood.
What is actively driving me nuts is this error when instantiating the MjrContext:
```
    context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_100)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
mujoco.FatalError: OpenGL ARB_framebuffer_object required
```

Here is the related code:
```py
if not glfw.init():
    raise RuntimeError("GLFW initialization failed")
glfw.window_hint(glfw.CONTEXT_VERSION_MAJOR, 4)
glfw.window_hint(glfw.CONTEXT_VERSION_MINOR, 5)
window = glfw.create_window(1000, 600, "gl_example", None, None)
if window is None:
    raise RuntimeError("Failed to create GLFW window")
glfw.make_context_current(window)

context = mujoco.MjrContext(model, mujoco.mjtFontScale.mjFONTSCALE_100)
```

What is really weird is that I tried this piece of code to check if the context indeed misses the `GL_ARB_framebuffer_object` extension:
```py
print("GL_ARB_framebuffer_object supported:", glfw.extension_supported("GL_ARB_framebuffer_object"))
# ...prints "1"
```
... and no, the context _has_ the extension. I do not understand why MuJoCo complains.

What is even weirder is that if I remove the two glfw window hints about the context version, the error disappears and everything works fine. The problem is I need those hints for OpenXR. What should I do?

yuvaltassa, 2024-07-03:
Perhaps @nishbo has an idea?

Relatedly, look at #1452

SkytAsul, 2024-07-03:
I did, but it's not using OpenXR. Moreover, the glfw window hints that cause the trouble are not set in this project.
I made some tests and found out some interesting results:
If I set the context version to 2.0, 2.1, 3.0 and 3.1 it works (but will not work with OpenXR).
If I want to set the context version to anything higher than 3.1, I have to change the `OPENGL_PROFILE` to `OPENGL_COMPAT_PROFILE`. Unfortunately, this is not possible with OpenXR because the OpenGL profile used is hardcoded.
Is it possible to investigate why the core profile does not work?

nishbo, 2024-07-03:
Interesting, I am not sure why this would happen, although I also have somewhat similar problems with using the wrong context when initializing opengl for openxr. I do not have a solution at the moment.

SkytAsul, 2024-07-08:
It's been a pain, but I eventually succeeded in rendering MuJoCo to a VR headset using the python OpenXR wrapper pyopenxr.  
Here is the code: https://gist.github.com/SkytAsul/b1a48a31c4f86b65d72bc8edcb122d3f

The OpenGL version/profile bug I reported above prevents the use of an OpenGL debugger such as RenderDoc, which requires at least OpenGL 3.2 and the Core Profile...

nishbo, 2024-07-09:
Thanks for sharing the code! I will look into what can be done about the MuJoCo OpenGL