google-deepmind/mujoco #1783: Excessive Duplicate Vertices in Imported Mesh in MuJoCo

binbinbai, 2024-07-04:
Hello,

I encountered an issue when importing a mesh with 48,164 vertices into MuJoCo. When I accessed the vertices through sim.model.mesh_vert, the number of vertices reported was 193,569, which is significantly larger than the actual number of vertices. Upon filtering out the duplicate coordinates from the 193,569 vertices, I found the correct number of vertices to be 48,164. This indicates that there are a large number of duplicate vertices in the MuJoCo representation of the mesh.

Could you please explain the logic behind these duplicate vertices? Additionally, I would like to restore the vertex count to 48,164 while ensuring the order is consistent with the original mesh indices. How can I achieve this?

yuvaltassa, 2024-07-04:
Interesting! Can you please provide the mesh so we can reproduce?

binbinbai, 2024-07-05:
Sure，here is the stl file
[stamp.zip](https://github.com/user-attachments/files/16103968/stamp.zip)

When opening the STL file in MeshLab, the number of vertices is 48,164.However, when importing the same file into MuJoCo, the face count is correct, but the vertex count is significantly higher.
[image]

[image]

closed 2024-07-09 by binbinbai

yuvaltassa, 2024-07-09:
I didn't get around to looking at this yet, any reason why you closed it?