google-deepmind/mujoco #1786: Unity Plugin DLLNotFoundException: mujoco assembly:<unknown assembly>

dhruvmsheth, 2024-07-06:
Hello,

I'm a student trying to use the Unity Plugin for Mujoco to import a mujoco simulation into unity and use it with ml-agents. For this, I'm planning to use the ml-agents llapi to interact with the mujoco environment in unity. However, when I build the simulation in unity and then try to run it using the llapi in ml-agents, I get a DLLNotFoundException. I assume it's because my dll was initially located at Users folder and unity doesn't directly include it during build. I assume this is somewhat similar to the issue given here: https://github.com/google-deepmind/mujoco/issues/1637#issuecomment-2111379171
I have the mujoco.dll file in Assets folder as well and tried creating a Plugin folder within Assets and moving it there, but none of the options work after building. For context, the Mujoco version I'm using is 3.1.6 and Unity version is 2022.3.32f1. I saw that @Balint-H had some workaround for this. Would you know how to fix the issue here? 

[image]

Balint-H, 2024-07-06:
Hello, since you are using WSL, you need the Linux build of MuJoCo, and the .so file in it. Before you move the library file to your project, make sure the file extension is only ".so" (if I remember well, it is appended by the version number e.g. libmujoco.so.313) and truncate the file name if needed.

You can have the .dll in your project as well, which would allow you to make builds for windows with cross-compilation. But you need the .so file to run the plugin in the editor, or make builds for Linux.

dhruvmsheth, 2024-07-06:
Thanks a lot! I completely missed that. That solved my issue!

closed 2024-07-06 by dhruvmsheth