google-deepmind/mujoco #1787: mjx-accelerated gymnasium-like environment primitives (env.reset(), env.step()) without brax

7oponaut, 2024-07-06:
Hello,

I am currently working on an RL project that involves training models from mujoco_menagerie to perform certain tasks.

I already have my own working code for the whole process, which uses pytorch. I would like to accelerate the mujoco simulator using mjx, so I am planning on porting my code to jax.

I want to implement a gymnasium-like environment interface such that the batch of RL environments can be accelerated using my GPU.

I am aware of the tutorial [notebook](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb) that covers mjx acceleration. This notebook uses brax for interacting with the simulator and brax's own ppo implementation. As I have said, I have my own implementation which I want to stick with. I have been browsing the mujoco and brax githubs to figure out how to implement the primitives I need for an accelerated training loop.

I need a way to
1) reset a batch of mujoco simulations with the option to control which instances to reset (with e.g. masking)
2) randomly modify the initial state configuration per simulator instance to diversify training data
3) convey batched agent actions to the simulator instances

I would like to perform these actions on the accelerated mjx objects.

The mjx tutorial uses brax, which abstracts away implementation details. I would like your assistance with figuring out how to implement the primitives I need without brax, since I would like to understand how to work with mujoco/mjx/jax directly, and I want flexibility in how my code is structured.

1)
For simulator resets, I suspect that it might be enough to reset `mjx_data.qpos` and `mjx_data.qvel` to their initial values, but `mujoco.mj_resetData()` does a bunch of other things based on the code in [src/engine/engine_io.c](https://github.com/google-deepmind/mujoco/blob/f57dc90669208914f97188acc699a1fb251f8079/src/engine/engine_io.c#L1566)

How should I reset `mjx_data`?

2)
Afaik modifying the freejoint parameters in `mjx_data.qpos` after reset should be sufficient to randomize the robot configuration.

Another issue: I use hfields to randomize terrain for each environment instance. In my current implementation I have a unique `mj_model` for each environment instance, and I can modify the `mj_model.hfield_data` of each environment to a unique value (of identical shape) during reset.

However, in the mjx tutorial, `mjx.step()` takes as input a single `mjx_model` and a batch of `mjx_data`. Since `mjx_model` contains the `hfield_data` array, there seems to be no way to set a different hfield for each environment instance. Is there a way around this?

3)
Is setting the `mjx_data.ctrl` array the canonical way of sending control signals when using mjx?

----

It would be nice to have a brax-free mjx tutorial that clarifies these questions. If this already exists, please do refer me to it.

7oponaut, 2024-07-06:
I did more research, reading the mujoco docs and the brax source code.

The relevant bits seem to be at [brax/mjx/pipeline.py](https://github.com/google/brax/blob/c89a3ad7900648613ead04dedacc91e923e10a68/brax/mjx/pipeline.py). The class `System` is subclassed from `mjx.Model` and the class `State` is subclassed from `mjx.Data`. Looks like setting `mjx_data.ctrl` is the way to [set the actuators](https://github.com/google/brax/blob/c89a3ad7900648613ead04dedacc91e923e10a68/brax/mjx/pipeline.py#L99) like I mentioned.

I came up with some code that I think is roughly correct. I am using jax for the first time so please excuse my French. In short, I batch both `mjx_model` and `mjx_data` and I use `jax.vmap()` to vectorize everything.

The version I am posting here is set up for testing on [mujoco_menagerie/google_barkour_vb/scene_hfield_mjx.xml](https://github.com/google-deepmind/mujoco_menagerie/blob/89f610538ec6c424e620d393b90dffeebd8cedbe/google_barkour_vb/scene_hfield_mjx.xml) to make it reproducible, although I am working with a slightly different setup. You can save the code in a new file in the mujoco_menagerie repo folder and run it from there.

`jax.jit()` seems to be crucial for acceptable runtimes: `mjx.step()` is very slow without it, even the subsequent runs.

I am measuring ~1 ms runtimes on the second run for `reset()` and `step()` on my machine (RTX 4090).

```
import time

import jax
import jax.numpy as jnp
import jax.random as jr
import mujoco
import mujoco.mjx as mjx

def create_hfield(rng, shape) -> jnp.ndarray:
    # dummy implementation
    return jr.uniform(rng, shape, minval=0.0, maxval=1.0)

def create(xml_path: str, batch_size: int):
    mj_model = mujoco.MjModel.from_xml_path(xml_path)
    mjx_model = mjx.put_model(mj_model)
    mjx_model_batch = jax.tree.map(
        lambda x: x[None].repeat(batch_size, axis=0), mjx_model
    )
    mjx_data_batch = jax.vmap(mjx.make_data)(mjx_model_batch)
    return mjx_model, mjx_model_batch, mjx_data_batch

def reset_(key, mjx_model_batch, mjx_data_batch, reset_mask: jnp.ndarray | None):
    key, subkey = jr.split(key)
    subkey = jr.split(subkey, reset_mask.shape[0])
    mjx_model_batch = jax.vmap(
        lambda key, reset, mjx_model: (
            jax.lax.cond(
                reset,
                lambda _: mjx_model.replace(
                    hfield_data=create_hfield(key, mjx_model.hfield_data.shape)
                ),
                lambda _: mjx_model,
                operand=None,
            )
        )
    )(subkey, reset_mask, mjx_model_batch)
    del subkey

    mjx_data_batch = jax.vmap(
        lambda reset, mjx_model, mjx_data: (
            jax.lax.cond(
                reset,
                lambda _: mjx.make_data(mjx_model),
                lambda _: mjx_data,
                operand=None,
            )
        )
    )(reset_mask, mjx_model_batch, mjx_data_batch)

    return key, mjx_model_batch, mjx_data_batch

def step_(mjx_model_batch, mjx_data_batch, action):  # TODO add ctrl = act
    mjx_data_batch = mjx_data_batch.replace(ctrl=action)
    mjx_data_batch = jax.vmap(mjx.step)(mjx_model_batch, mjx_data_batch)
    return mjx_data_batch

def main() -> None:
    key = jr.key(42)

    batch_size = 4096

    reset = jax.jit(reset_)
    step = jax.jit(step_)

    for idx in range(2):
        print(f"run {idx + 1}")

        t = time.perf_counter()
        mjx_model, mjx_model_batch, mjx_data_batch = create(
            "google_barkour_vb/scene_hfield_mjx.xml", batch_size
        )
        print("create", time.perf_counter() - t)

        t = time.perf_counter()
        key, mjx_model_batch, mjx_data_batch = reset(
            key,
            mjx_model_batch,
            mjx_data_batch,
            jnp.ones((batch_size,), dtype=jnp.bool),
        )
        print("reset all", time.perf_counter() - t)

        ctrl_range = mjx_model.actuator_ctrlrange
        key, subkey = jr.split(key)
        action = jr.uniform(
            subkey,
            (batch_size, ctrl_range.shape[0]),
            minval=ctrl_range[:, 0],
            maxval=ctrl_range[:, 1],
        )
        del subkey

        t = time.perf_counter()
        mjx_data_batch = step(mjx_model_batch, mjx_data_batch, action)
        print("step", time.perf_counter() - t)

        reset_mask = jnp.zeros((batch_size,), dtype=jnp.bool)
        reset_mask = reset_mask.at[: batch_size // 2].set(True)

        t = time.perf_counter()
        key, mjx_model_batch, mjx_data_batch = reset(
            key, mjx_model_batch, mjx_data_batch, reset_mask
        )
        print("reset some", time.perf_counter() - t)

if __name__ == "__main__":
    main()
```

Output:
```
run 1
create 2.358741555995948
reset all 0.4552512099980959
step 5.95394677999866
reset some 0.001281887998629827
run 2
create 0.10158524000144098
reset all 0.0010806529971887358
step 0.0009590289992047474
reset some 0.0007585439961985685
```

7oponaut, 2024-07-08:
I iterated on this and came up with a solution that's more viable I think. The code has gotten a bit long to keep posting it in github comments but I will put it here for completeness.

In this version, I define `init_mjx_single()` and `step_mjx_single()` that run on single `mjx.Model` and `mjx.Data` instances. This seems more in line with jax examples I have seen. I can use these functions to take care of environment-specific business:
```
def _get_obs(mjx_data: mjx.Data) -> Array:
    return jnp.concatenate(
        [
            mjx_data.qpos,
            mjx_data.qvel,
            mjx_data.cinert.ravel(),
            mjx_data.cvel.ravel(),
        ]
    )

def init_mjx_single(key: Array, mjx_model: mjx.Model):
    hfield_data = jr.uniform(key, mjx_model.hfield_data.shape, minval=0.0, maxval=1.0)
    mjx_model = mjx_model.replace(hfield_data=hfield_data)

    mjx_data = mjx.make_data(mjx_model)
    mjx_data = mjx.forward(mjx_model, mjx_data)
    obs = _get_obs(mjx_data)

    return mjx_model, mjx_data, obs

def step_mjx_single(
    mjx_model: mjx.Model, mjx_data: mjx.Data, action: Array
) -> mjx.Data:
    action = action.clip(min=-1.0, max=1.0)
    ctrl_range = mjx_model.actuator_ctrlrange
    ctrl_mid = ctrl_range.mean(axis=1)
    ctrl_half = ctrl_range[:, 1] - ctrl_mid
    action = ctrl_mid + ctrl_half * action

    mjx_data = mjx_data.replace(ctrl=action)
    mjx_data = mjx.step(mjx_model, mjx_data)

    obs = _get_obs(mjx_data)
    reward = 0.0
    terminated = False
    truncated = False

    return mjx_data, obs, reward, terminated, truncated
```

There are three more functions.

One that creates `mjx.Model` and `mjx.Data` batched data structures:
```
def create(xml_path: str, batch_size: int):
    mj_model = mujoco.MjModel.from_xml_path(xml_path)
    mjx_model = mjx.put_model(mj_model)
    mjx_model_batch = jax.tree.map(
        lambda x: x[None].repeat(batch_size, axis=0), mjx_model
    )
    mjx_data_batch = jax.vmap(mjx.make_data)(mjx_model_batch)
    return mjx_model_batch, mjx_data_batch
```

A batched `reset()` function constructed from `init_mjx_single()`. No need for masked resets here after all: I take care of that in the `step()` function.
```
def make_reset(init_mjx):
    def reset(key: Array, mjx_model: mjx.Model):
        batch_size = mjx_model.hfield_data.shape[0]
        key = jr.split(key, batch_size)
        mjx_model, mjx_data, obs = jax.vmap(init_mjx)(key, mjx_model)
        return mjx_model, mjx_data, obs

    return reset
```

[... 99 lines cut ...]
        t = time.perf_counter()
        mjx_model_batch, mjx_data_batch, obs, reward, terminated, truncated = step(
            subkey, mjx_model_batch, mjx_data_batch, terminated, truncated, action
        )
        print("step/reset", time.perf_counter() - t)
        del subkey

if __name__ == "__main__":
    main()
```

Output:

```
run 1
create 2.308881144002953
reset 5.953607456001919
step 11.88598394300061
step/reset 0.001969937002286315
run 2
create 0.09895871399930911
reset 0.0023224000033224
step 0.0016138470018631779
step/reset 0.0014887260040268302
```

JeyRunner, 2024-07-09:
Looks interesting :) 
Did you compare the step performance of the step function with and without the `jax.lax.cond(` for `init_mjx_` ?
I am not sure if cond will ensure that just the correct branch is executed (although just the correct value will be used as a result of the expression). This may cause some overhead of executing the `init_mjx_`  at every step.

Although in the xla documentation this described as just executing the correct branch (https://openxla.org/xla/operation_semantics#conditional).
Nevertheless it would be interesting to test this, since in brax the environment is not really reset in the step function but a cached first state is applied (for performance reasons I guess).

erikfrey, 2024-07-10:
@JeyRunner - you can refresh the pool of reset states at some regular interval without a performance hit, see the param [num_resets_per_eval](https://github.com/google/brax/blob/main/brax/training/agents/ppo/train.py#L92) in brax's PPO trainer for an example of how to do this.

@7oponaut I'm glad you figured this out!  Since it seems the issue is resolved, I'm going to migrate this thread to a discussion for anyone else that's interested in what you figured out here.  Feel free to open a new issue if you hit any more blockers.

closed 2024-07-10 by erikfrey