google-deepmind/mujoco #1790: Implementing a Velcro-sticking arrangement using the adhesion actuator.

AkashVardhan7, 2024-07-07:
Hello, I am trying to develop a simulation of 3 link shape changing robots using Mujoco. In the experiments the robots have Velcro patches on their arms which bind in a certain configuration only i.e. M-F and not vice versa. I have implemented this in my model using patches belonging to specific groups 1 and 2 as a proxy for the M and F parts of the Velcro and colored them B and G. However since these patches are children of the arms, when I try to declare the adhesion actuator later on in my code I am having to use the parent body which are the arms for the adhesion actuator. I try to manually enforce this constraint by implementing an adhesion function which  assigns an adhesion force to a pair of bodies coming in contact if they belong to groups 1 and 2 respectively, however since the whole robot arm is declared as the adhesion body it causes the arms to stick in unnatural ways. I am attaching my simulation file and the video I generated when  implementing this algorithm.

https://github.com/google-deepmind/mujoco/assets/48886152/6c532332-63a6-4dfb-bcf2-e47f3490fad9
Here is my code.. Couldn't attach the file hence pasting it out.

import mujoco as mj
from mujoco.glfw import glfw
from time import sleep
import numpy as np
import random
import matplotlib.pyplot as plt

# Base XML configuration for a single robot with planes on the arms
modified_base_robot_xml = '''
    
    <body name="{robot_name}_central_link" pos="{x_pos} {y_pos} 0" quat="{quat_w} {quat_x} {quat_y} {quat_z}">
        <joint name="{robot_name}_free_joint" type="free"/>
        <geom type="box" size="0.04 0.01 0.01" mass="0.1" rgba="1 0 0 1"/>
        
        <body name="{robot_name}_arm_1" pos="-0.07 0 0.0">
            <joint name="{robot_name}_hinge1" type="hinge" axis="0 1 0" pos="0.03 0 0.0" range="0 1.5708" damping="0.25" frictionloss="0.1" stiffness="0.25"/>
            <geom type="box" size="0.03 0.01 0.0025" mass="0.05" rgba="0 0 0 1" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad1_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad2_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad3_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 -0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad4_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 -0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
        </body>

         
        <body name="{robot_name}_arm_2" pos="0.07 0 0.0">
            <joint name="{robot_name}_hinge2" type="hinge" axis="0 -1 0" pos="-0.03 0 0.0" range="0 1.5708" damping="0.25" frictionloss="0.1" stiffness="0.25"/>
            <geom type="box" size="0.03 0.01 0.0025" mass="0.05" rgba="0 0 0 1" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad1_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad2_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad3_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 -0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad4_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 -0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
        </body>
    </body>
'''

simend = 600  # Simulation time (15 seconds)
relaxation_time = 1  # Relaxation time (2 seconds)
print_camera_config = 0  # Set to 1 to print camera config

# Initialize simulation parameters
num_robots = 10
x_range = (-0.15, 0.15)
y_range = (-0.15, 0.15)
robot_positions = []
gain = 8
ctrlrange = [0, 1]

# For callback functions
button_left = False
button_middle = False
button_right = False
lastx = 0
lasty = 0

def init_controller(model, data):
    # Initialize the controller here. This function is called once, at the beginning
    pass

def apply_adhesion(model, data, gain, ctrlrange):
    print(f"data.xfrc_applied.shape: {data.xfrc_applied.shape}")

[... 219 lines cut ...]

    # Get framebuffer viewport
    viewport_width, viewport_height = glfw.get_framebuffer_size(window)
    viewport = mj.MjrRect(0, 0, viewport_width, viewport_height)

    # Print camera configuration (help to initialize the view)
    if print_camera_config == 1:
        print('cam.azimuth =', cam.azimuth, ';', 'cam.elevation =', cam.elevation, ';', 'cam.distance =', cam.distance)
        print('cam.lookat = np.array([', cam.lookat[0], ',', cam.lookat[1], ',', cam.lookat[2], '])')

    # Update scene and render
    mj.mjv_updateScene(model, data, opt, None, cam, mj.mjtCatBit.mjCAT_ALL, scene)
    mj.mjr_render(viewport, scene, context)

    # Swap OpenGL buffers (blocking call due to v-sync)
    glfw.swap_buffers(window)

    # Process pending GUI events, call GLFW callbacks
    glfw.poll_events()

# Plot the robot positions after the simulation ends
plot_robot_positions()    

# Terminate GLFW
glfw.terminate()

AkashVardhan7, 2024-07-07:
This didn't get pasted properly....
 modified_base_robot_xml = '''
    
    <body name="{robot_name}_central_link" pos="{x_pos} {y_pos} 0" quat="{quat_w} {quat_x} {quat_y} {quat_z}">
        <joint name="{robot_name}_free_joint" type="free"/>
        <geom type="box" size="0.04 0.01 0.01" mass="0.1" rgba="1 0 0 1"/>
        
        <body name="{robot_name}_arm_1" pos="-0.07 0 0.0">
            <joint name="{robot_name}_hinge1" type="hinge" axis="0 1 0" pos="0.03 0 0.0" range="0 1.5708" damping="0.25" frictionloss="0.1" stiffness="0.25"/>
            <geom type="box" size="0.03 0.01 0.0025" mass="0.05" rgba="0 0 0 1" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad1_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad2_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad3_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 -0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad4_arm1" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 -0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
        </body>

         
        <body name="{robot_name}_arm_2" pos="0.07 0 0.0">
            <joint name="{robot_name}_hinge2" type="hinge" axis="0 -1 0" pos="-0.03 0 0.0" range="0 1.5708" damping="0.25" frictionloss="0.1" stiffness="0.25"/>
            <geom type="box" size="0.03 0.01 0.0025" mass="0.05" rgba="0 0 0 1" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad1_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad2_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            
            <geom name="{robot_name}_pad3_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 0.004 -0.0025" rgba="0.2 0.8 0.2 1.0" group="2" friction="1.0" margin="0.0001"/>
            <geom name="{robot_name}_pad4_arm2" type="box" size="0.02 0.002 0.00001" pos="0.0 -0.004 -0.0025" rgba="0.0 0.0 0.5 1.0" group="1" friction="1.0" margin="0.0001"/>
        </body>
    </body>
'''

AkashVardhan7, 2024-07-07:
[image]

yuvaltassa, 2024-07-08:
I didn't quite follow the question, and the video was not helpful, but I think there is something you might be missing. Yes, adhesion applies to all geoms in a body, but you can wrap some subset with a _static body_ (no joints).

So instead of
```xml
<body>
  <geom name="1"/>
  <geom name="2"/>
</body>
```
where I only want geom `2` to be adhesive but I get unwanted adhesion from geom `1`, I can do
```xml
<body>
  <geom name="1"/>
  <body name="static_child">
    <geom name="2"/>
  </body>
</body>
```
and connect the adhesion actuator to the `static_child` body.

AkashVardhan7, 2024-07-08:
Hi, thank you for your reply. I realize my question could have been clearer and more precise. I want to implement adhesion between 2 specific groups of objects. These objects will be children to the robot arms (in my case) and I assigned them groups 1 and 2. I can follow your advice and make them static bodies and then connect the adhesion actuator to them in part where I generate the combined xml's for the robots. How would I be able to ensure that the adhesion between them is directional, which is to say adhesion happens only when a static body belonging to group 1 comes in contact with a static body belonging to group 2? Currently I have a function in my code which does this, where I check for the group identity of a pair of colliding bodies and assign them a force based on the specified gain and ctrl-range. I am not sure if this is the right way.

yuvaltassa, 2024-07-14:
In the current design, if body 1 has an adhesion actuator then if body 2 can contact body 1, adhesion force can be generated between them. So given the current design you need a workaround similar to what you are doing.

Basically what you describe is a feature request. I doubt it will happen since this is a pretty rare use case.

However, it is very very easy to just modify the related code and rebuild from source, if you don't mind a bit of hacking. See this block [here](https://github.com/google-deepmind/mujoco/blob/3d28a1d8c7f212e5fe2d4d7e610b961231586d87/src/engine/engine_core_smooth.c#L1180)?

```c
// irrelevant contact, continue
if (b1 != id && b2 != id) {
  continue;
}
```

just add similar thing just after like 

```c
// geom ids
int g1 = con->geom[0];
int g2 = con->geom[1];

// contact not between "velcro" geoms, continue
if ((m->geom_group[g1] != 1 && m->geom_group[g2] != 2) || 
     m->geom_group[g2] != 1 && m->geom_group[g1] != 2)) {
  continue;
}
```
(or whatever other condition you want)

closed 2024-07-14 by yuvaltassa

AkashVardhan7, 2024-07-15:
Thanks for your help on this. I can try out the solution you suggested and post in the forum to let you guys know how it went. Btw, I work in the same lab as Tingnan Zhang who is a part of your team .