google-deepmind/mujoco #1794: Problem of touch_grid plug-in when collision happens between capsule-geom and the plane-geom

Josh00-Lu, 2024-07-09:
Hi MuJoCo team,

I'm a student and currently trying to use MuJoCo for research. I'm looking for some help with the `touch_grid` plug-in.

Here is the core-part of my xml model:

```xml
<mujoco model="hopper">

  <extension>
    <plugin plugin="mujoco.sensor.touch_grid"/>
  </extension>

  <asset>
    <texture name="skybox" type="skybox" builtin="gradient" rgb1=".2 .3 .4" rgb2="0 0 0" width="800" height="800" mark="random" markrgb="1 1 1"/>
    <texture name="grid" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="300" height="300" mark="edge" markrgb=".2 .3 .4"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
    <material name="self" rgba=".7 .5 .3 1"/>
  </asset>
  <visual>
    <headlight ambient=".4 .4 .4" diffuse=".8 .8 .8" specular="0.1 0.1 0.1"/>
    <map znear=".001"/>
    <quality shadowsize="16384"/>
  </visual>
  <compiler angle="degree" coordinate="local" inertiafromgeom="true"/>
  <default>
    <joint armature="1" damping="1" limited="true"/>
    <geom conaffinity="0" condim="1" margin="0.001" solimp=".8 .8 .01" solref=".02 1" rgba=".7 .5 .3 0.1"/>
    <motor ctrllimited="true" ctrlrange="-.4 .4"/>
  </default>
  <option integrator="RK4" timestep="0.002"/>
  <worldbody>
    <light cutoff="100" diffuse="1 1 1" dir="0 -0.7 -1.3" directional="true" exponent="1" pos="0 70 130" specular=".1 .1 .1"/>
    <geom conaffinity="1" condim="3" name="floor" pos="0 0 0" rgba="1 1 1 1" size="200 200 .125" type="plane" material="grid"/>
    <body name="0" pos="0 0 1.25">
      <camera name="track" mode="trackcom" pos="0 -3 1" xyaxes="1 0 0 0 0 1"/>
      <joint armature="0" axis="1 0 0" damping="0" limited="false" name="rootx" pos="0 0 -1.25" stiffness="0" type="slide"/>
      <joint armature="0" axis="0 0 1" damping="0" limited="false" name="rootz" pos="0 0 -1.25" ref="1.25" stiffness="0" type="slide"/>
      <joint armature="0" axis="0 1 0" damping="0" limited="false" name="0_joint" pos="0 0 0" stiffness="0" type="hinge"/>
      <geom friction="0.9" fromto="-0 0 0.2 0 0 -0.2" size="0.05" type="capsule"/>
      <body name="1" pos="0 0 -0.2">
        <joint axis="0 -1 0" name="1_joint" pos="0 0 0" range="-360 360" type="hinge"/>
        <geom friction="2.0" fromto="0 0 0 0 0 -0.45" size="0.05" type="capsule"/>
        <site name="1_touchgrid" pos="0 0 -0.45" zaxis="0 0 0.45" />
        <site name="1_torque" pos="0 0 -0.45" zaxis="0 0 0.45" />
        <site name="1_force" pos="0 0 -0.45" zaxis="0 0 0.45" />
      </body>
    </body>
  </worldbody>
  <actuator>
    <motor ctrllimited="true" ctrlrange="-1.0 1.0" gear="200" joint="1_joint" name="1_joint"/>
  </actuator>
  <sensor>
    <plugin name="1_touchgrid" plugin="mujoco.sensor.touch_grid" objtype="site" objname="1_touchgrid">
      <config key="size" value="5 5"/>
      <config key="fov" value="180 90"/>
      <config key="gamma" value="0"/>
      <config key="nchannel" value="1"/>
    </plugin>
    <torque name="1_torque" site="1_torque" />
    <force name="1_force" site="1_force" />
  </sensor>
</mujoco>
```

[image]

As shown in the picture above, I am running a rigid-robot for reinforcement learning in MuJoCo. I found that when a high-speed moving `capsule-geom` running towards the `plane-geom`, penetration issue is **heavy**. Previously, this issue wasn’t a major concern, but now I want to add a touch_grid sensor to the end of the capsule to simulate touch.

[image]

However, as the image illustrates, the overlap causes **incorrect** force direction perception (the force direction is completely reversed).

[image]

While using contact-force visualization, I get accurate and wanted results!

I’m curious about how to achieve these results with the `touch_grid` plug-in. Alternatively, if not using touch_grid, what other viable solutions exist? I look forward to your suggestions.

It’s worth noting that I don’t think placing the site at the center or starting point of the geom (in order to make it "high enough") is a good idea, as this would result in an abnormal touch grid distribution.

yuvaltassa, 2024-07-09:
The direction of the touch grid visualization is not related to the direction of the force, only to the orientation of the touch-grid viewport.

I think this is pretty clear if you look at the associated video: https://www.youtube.com/watch?v=0LOJ3WMnqeA

closed 2024-07-09 by yuvaltassa

Josh00-Lu, 2024-07-10:
> The direction of the touch grid visualization is not related to the direction of the force, only to the orientation of the touch-grid viewport.
> 
> I think this is pretty clear if you look at the associated video: https://www.youtube.com/watch?v=0LOJ3WMnqeA

Thank you very much for your prompt answer. However, the current activated grid unit the upper one. I want to leverage the position of the activate grid unit to simulate the touch point, are there any methods?