google-deepmind/mujoco #1795: False force sensor reading in measuring spatial tendon

NuCapybara, 2024-07-09:
Hi MuJoCo contributors,

I'm a student who has been using MuJoCo for a month, and I'm trying to use it to simulate exoskeleton-human interaction. Thank you so much for answering my question.

I'm looking for some help with the spatial tendon force readings. In the model, I use two spatial tendons acting like spring-dampers, connecting the exoskeleton cuff to the human leg (knee part, represented as a pink cylinder). I wish to measure the force in each spring by using a tendonlimitfrc sensor. However, the force fluctuates rapidly between 1N and 4600N even in a resting scene (no external force/actuation), which is unstable and completely off. I think one of the causes might be the instability of the tendon, but I'm not sure why.

Could you please help identify the issues:

Why does the spatial tendon keep swinging and remain unstable? Is that the reason why the force sensors are off as well?
Am I supposed to use a limited spatial tendon to simulate the spring-damper system, so it acts like a spring when the length is larger than the limit, generating force? It seems that with the limited spatial tendon, when I change the stiffness to 0, the force is still around 5000N.
Did I use the correct force sensor as tendonlimitfrc?
Thank you for your assistance.

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="x2">
  <option gravity="0 0 -9.81" />
  <option integrator="RK4">
    <flag energy="enable"/>
  </option>
  <compiler angle="radian" />
  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.1 0.1 0.1" specular="0.9 0.9 0.9"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="140" elevation="-20"/>
  </visual>
  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>
  <actuator>
    <motor name="left_hip_joint_motor" joint="left_hip_joint" ctrlrange="-1000 1000" ctrllimited="true" />
    <motor name="left_knee_joint_motor" joint="left_knee_joint" ctrlrange="-1000 1000" ctrllimited="true" />
  </actuator>
  <asset>
    <mesh name="back_pack" file="back_pack.stl" scale="0.001 0.001 0.001" />
    <mesh name="hip_sa" file="hip_sa.stl" scale="0.001 0.001 0.001" />
    <mesh name="thigh-l_s" file="thigh-l_s.stl" scale="0.001 0.001 0.001" />
    <mesh name="knee-l" file="knee-l.stl" scale="0.001 0.001 0.001" />
    <mesh name="shank-l" file="shank-l.stl" scale="0.001 0.001 0.001" />
    <mesh name="ankle" file="ankle.stl" scale="0.001 0.001 0.001" />
    <mesh name="foot-l" file="foot-l.stl" scale="0.001 0.001 0.001" />
    <mesh name="hip-r" file="hip-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="thigh-r" file="thigh-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="knee-r" file="knee-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="shank-r" file="shank-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="foot-r" file="foot-r.stl" scale="0.001 0.001 0.001" />
  </asset>
  <worldbody>
    <geom name="floor" pos="0 0 -0.5" size="0 0 0.05" type="plane" material="groundplane"/>
    <body name="backpack" pos="0 0 1.05" quat="0.5 0.5 0.5 0.5">
      <inertial pos="0.00139519 -0.0708429 0.0143736" quat="0.155308 0.69016 0.160036 0.688438" mass="8.063" diaginertia="0.275688 0.187219 0.144619" />
      <joint name="world_to_backpack" pos="0 0 0" axis="-1 0 0" range="-1.5708 1.5708" actuatorfrcrange="-40 40" damping="1" frictionloss="2" />
      <geom type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="back_pack" />
      <geom pos="0.191605 -0.211166 0.13831" quat="0 0 0.707107 0.707107" type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="hip_sa" />
      <geom pos="-0.191605 -0.211166 0.13831" quat="0.707107 -0.707107 0 0" type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="hip-r" />
      <body name="left_upper_thigh" pos="0.182275 -0.211166 0.13831" quat="-1 0 0 0">
        <inertial pos="-0.0044319 -0.254625 0.000368545" quat="0.713939 0.671779 -0.127372 0.150933" mass="2.1844" diaginertia="0.0398119 0.0387337 0.00242996" />
        <joint name="left_hip_joint" type="hinge" pos="0 0 0" axis="-1 0 0" range="-0.698132 2.0944" actuatorfrcrange="-80 80" damping="2" frictionloss="2" />
        <geom type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="thigh-l_s" />
        <geom pos="-0.00313 -0.37 0" quat="0 0 1 0" type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="knee-l" />
        <body name="left_upper_thigh_drive_inertia">
          <inertial pos="0 0 0" mass="0.1" diaginertia="0.0001246 0.0001 0.0001" />
          <joint name="left_hip_joint_drive" type="hinge" pos="0 0 0" axis="1 0 0" actuatorfrcrange="-0.3265 0.3265" />
[... 73 lines cut ...]
      <site site="cuff_front"/>
    </spatial>
  </tendon> 

  <tendon>
    <spatial name="feet_connection_center" width="0.009" stiffness="1000000" damping="10">
      <site site="human_knee_btm_center"/>
      <geom geom="left_human_shank"/>
      <site site="feet_pad_center"/>
    </spatial>
  </tendon>

  <sensor>
      <tendonpos tendon="knee_connection_front" />
      <tendonlimitfrc tendon="knee_connection_front"/>
  </sensor>

</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/144244355/aa1018c4-b697-4ebb-a296-418c1a41028a

yuvaltassa, 2024-07-09:
You're not using [sidesite](https://mujoco.readthedocs.io/en/latest/XMLreference.html#spatial-geom-sidesite), how could it not glitch?

Fix that and then we'll discuss if there are any issues remaining.

NuCapybara, 2024-07-09:
Thank you so much for your kind help, Yuval!
For the sidesite, I tried the points between cuff and knee but it's still glitch. But a sidesite inside of the human leg(the cylinder) makes it stable, I notice the documentation said that it will make the tendon goes into the body. Does that make sense if I would like to simulate a spring?
But the force sensor reading looks much better than before. Thank you so much Yuval!
Another question: If I would like to validate the force sensor reading by calculation, the force formula would be F = stiffness * x + damp * xdot, where x represents the distance the spring is stretched. It seems with the limited tendon, the tendon will not exceed the limited range and

NuCapybara, 2024-07-09:
Thank you so much for your kind help, Yuval!

For the sidesite, I tried the points between cuff and knee but it's still glitch. But a sidesite inside of the human leg(the cylinder) makes it stable, I notice the documentation said that it will make the tendon goes into the body. Does that make sense if I would like to simulate a spring-damper?

Another question: If I would like to validate the force sensor reading by calculation, the force formula would be F = stiffness * x + damp * xdot, where x represents the distance the spring is stretched. With the limited tendon, the tendon will not exceed the limited range, [0 0.05], and the force will only create after the tendon reaching its limit. How could I calculate the force if x cannot be found with the limited tendon?

But the force sensor reading looks much better than before. Thank you so much Yuval!

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="x2">
  <option gravity="0 0 -9.81" />
  <option integrator="RK4">
    <flag energy="enable"/>
  </option>
  <compiler angle="radian" />
  <visual>
    <headlight diffuse="0.6 0.6 0.6" ambient="0.1 0.1 0.1" specular="0.9 0.9 0.9"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <global azimuth="140" elevation="-20"/>
  </visual>
  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>
  <actuator>
    <motor name="left_hip_joint_motor" joint="left_hip_joint" ctrlrange="-1000 1000" ctrllimited="true" />
    <motor name="left_knee_joint_motor" joint="left_knee_joint" ctrlrange="-1000 1000" ctrllimited="true" />
    <!-- Define more actuators for other joints as needed -->
  </actuator>
  <asset>
    <mesh name="back_pack" file="back_pack.stl" scale="0.001 0.001 0.001" />
    <mesh name="hip_sa" file="hip_sa.stl" scale="0.001 0.001 0.001" />
    <mesh name="thigh-l_s" file="thigh-l_s.stl" scale="0.001 0.001 0.001" />
    <mesh name="knee-l" file="knee-l.stl" scale="0.001 0.001 0.001" />
    <mesh name="shank-l" file="shank-l.stl" scale="0.001 0.001 0.001" />
    <mesh name="ankle" file="ankle.stl" scale="0.001 0.001 0.001" />
    <mesh name="foot-l" file="foot-l.stl" scale="0.001 0.001 0.001" />
    <mesh name="hip-r" file="hip-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="thigh-r" file="thigh-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="knee-r" file="knee-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="shank-r" file="shank-r.stl" scale="0.001 0.001 0.001" />
    <mesh name="foot-r" file="foot-r.stl" scale="0.001 0.001 0.001" />
  </asset>
  <worldbody>
    <geom name="floor" pos="0 0 -0.5" size="0 0 0.05" type="plane" material="groundplane"/>
    <body name="backpack" pos="0 0 1.05" quat="0.5 0.5 0.5 0.5">
      <inertial pos="0.00139519 -0.0708429 0.0143736" quat="0.155308 0.69016 0.160036 0.688438" mass="8.063" diaginertia="0.275688 0.187219 0.144619" />
      <joint name="world_to_backpack" pos="0 0 0" axis="-1 0 0" range="-1.5708 1.5708" actuatorfrcrange="-40 40" damping="1" frictionloss="2" />
      <geom type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="back_pack" />
      <geom pos="0.191605 -0.211166 0.13831" quat="0 0 0.707107 0.707107" type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="hip_sa" />
      <geom pos="-0.191605 -0.211166 0.13831" quat="0.707107 -0.707107 0 0" type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="hip-r" />
      <body name="left_upper_thigh" pos="0.182275 -0.211166 0.13831" quat="-1 0 0 0">
        <inertial pos="-0.0044319 -0.254625 0.000368545" quat="0.713939 0.671779 -0.127372 0.150933" mass="2.1844" diaginertia="0.0398119 0.0387337 0.00242996" />
        <joint name="left_hip_joint" type="hinge" pos="0 0 0" axis="-1 0 0" range="-0.698132 2.0944" actuatorfrcrange="-80 80" damping="2" frictionloss="2" />
        <geom type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="thigh-l_s" />
        <geom pos="-0.00313 -0.37 0" quat="0 0 1 0" type="mesh" contype="1" conaffinity="1" group="1" density="0" mesh="knee-l" />
        <body name="left_upper_thigh_drive_inertia">
          <inertial pos="0 0 0" mass="0.1" diaginertia="0.0001246 0.0001 0.0001" />
          <joint name="left_hip_joint_drive" type="hinge" pos="0 0 0" axis="1 0 0" actuatorfrcrange="-0.3265 0.3265" />
        </body>
        <body name="left_upper_shank" pos="-0.01687 -0.37 0" quat="0 0 1 0">
          <!-- <site name="s1" pos="0.15 0.06 0" size="0.02"/> -->
          <site name="knee_l" pos="0.01 0 0" size="0.005" rgba="0 .7 0 1"/>
          <site name="cuff_side" pos="0.02 -0.1 0" size="0.005" rgba="0 .7 0 1"/>
          <site name="cuff_front" pos="0.07 -0.1 -0.07" size="0.005" rgba="0 .7 0 1"/>
[... 103 lines cut ...]
  <!-- <tendon>
    <spatial name="feet_connection3" range="0 .0001" width="0.005" stiffness="20000" damping="5000">
      <site site="human_knee_btm3"/>
      <geom geom="left_human_shank"/>
      <site site="feet_pad3"/>
    </spatial>
  </tendon>

  <tendon>
    <spatial name="feet_connection4" range="0 .0001" width="0.005" stiffness="20000" damping="5000">
      <site site="human_knee_btm4"/>
      <geom geom="left_human_shank"/>
      <site site="feet_pad4"/>
    </spatial>
  </tendon> -->

</mujoco>

```

</details>

Here is the video after adding the sidesite

Uploading exo_human-2024-07-09_15.23.37.mp4…

NuCapybara, 2024-07-09:
reopen the issue ;)

yuvaltassa, 2024-07-09:
It's difficult to see the path of the tendon, maybe make the cylinder translucent?
The sidesite is supposed to be **outside** the cylinder, on the side that you want the tendon to wrap on. You can also put it insode bu that is a different kind of "wrapping", as explained in the docs.

I think what you're asking for is a sensor for the passive forces in the tendon (stiffness+damping), rather than the force generated by the constraint solver, is that correct?

NuCapybara, 2024-07-09:
> It's difficult to see the path of the tendon, maybe make the cylinder translucent? The sidesite is supposed to be **outside** the cylinder, on the side that you want the tendon to wrap on. You can also put it inside, but that is a different kind of "wrapping", as explained in the docs.
-  Hi Yuval, thank you so much for your following up. Here is my current cylinder trabslucent view. The sidesite is the red dot.(I made it next to the site, with 0.005 location difference)
[image]
I tried to put the sidesite outside of the cylinder, between the cuff & knee sites, it will lead to the glitch again? I wish the spring directly connects between the cuff site and knee site without wrapping on other directions. Would you suggest any good site location in this case?

https://github.com/google-deepmind/mujoco/assets/144244355/e3efbdd4-3a38-420c-9634-c2c15369ae3d

> I think what you're asking for is a sensor for the passive forces in the tendon (stiffness+damping), rather than the force generated by the constraint solver, is that correct?
- Yes, that's correct. I'm asking for a sensor for the passive forces in the tendon. 

Thank you so much for answering my questions and for your kind support and patience, Yuval! :)

NuCapybara, 2024-07-16:
> I still don't understand what is going on with the sidesite...
> 
> IIRC [this model](https://github.com/google-deepmind/mujoco/blob/main/test/testdata/model.xml) has a tendon wrapping around a cylinder, maybe load it and figure out what's happening there? [This model](https://github.com/google-deepmind/mujoco/blob/main/test/engine/testdata/island/tendon_wrap.xml) is even simpler.
> 
> I'll look into adding the tendon-passive-force sensor.

Hi Yuval,

Thank you so much for your help! I just looked into the first IIRC model with a tendon wrapping around a cylinder, it seems the same issue happens again, where the tendon does not stick with the side-site but it swings around. Does that happen on your end as well? 

Thank you again for your kind support and help :)

https://github.com/user-attachments/assets/79917b18-1876-4434-bad8-31859fceca85

yuvaltassa, 2024-07-17:
Ya but that's just bad design, look at the other model. I'll fix this specific model soon. Look at the other model.

closed 2024-07-17 by yuvaltassa

ZnMnCr, 2025-04-25:
Hello, I am currently trying to model a cable-driven parallel robot and aim to determine the tension forces in the cables. I have simulated the cables using tendons, but I am unclear about what type of actuators to use for applying specific tension to the tendons and what sensors to employ for measuring the tension in the cables.：
`
<mujoco model="FAST_5m_CDPR">
	<visual>
		<rgba haze=".3 .3 .3 1" /> <quality shadowsize="16384" numslices="28" offsamples="4" />
	</visual>
	
	<compiler angle="radian" coordinate="local" inertiafromgeom="true" />
	<option timestep="0.01" iterations="50" solver="Newton" tolerance="1e-10" gravity="0 0 -9.81"/>
	<asset>
		<texture type="skybox" builtin="gradient" rgb1="1 1 1" rgb2="0.6 0.8 1" width="256" height="256"/>
		<texture name="texplane" type="2d" builtin="checker" rgb1=".25 .25 .25" rgb2=".3 .3 .3" width="512" height="512" mark="cross" markrgb=".8 .8 .8" />
		<material name="matplane" reflectance="0.3" texture="texplane" texrepeat="1 1" texuniform="true" />
	</asset>
	
	<default>
		
		<joint armature="0.01" damping="0.01" limited="true" />
		<geom friction="0.5 0.1 0.1" density="1000" margin="0.001" rgba="0.8 0.8 0.8 1" />
	</default>
	
	<worldbody>
		<geom name="floor" pos="0 0 -1" size="0 0 1" type="plane" material="matplane" />
		<light directional="true" diffuse=".8 .8 .8" specular=".2 .2 .2" pos="0 0 5" dir="0 0 -1" />

		
		
		<body name="tower1" pos="-2.5 0 2.5">
			<geom type="sphere" size="0.05" rgba="1 0 0 1" />
			
			<site name="tower1_site" pos="0 0 0" size="0.01" />
		</body>
		
		<body name="tower2" pos="-1.25 -2.165 2.5">
			<geom type="sphere" size="0.05" rgba="1 0.5 0 1" />
			
			<site name="tower2_site" pos="0 0 0" size="0.01" />
		</body>
		
		<body name="tower3" pos="1.25 -2.165 2.5">
			<geom type="sphere" size="0.05" rgba="1 1 0 1" />
			
			<site name="tower3_site" pos="0 0 0" size="0.01" />
		</body>
		
		<body name="tower4" pos="2.5 0 2.5">
			<geom type="sphere" size="0.05" rgba="0 1 0 1" />
			
			<site name="tower4_site" pos="0 0 0" size="0.01" />
		</body>
		
		<body name="tower5" pos="1.25 2.165 2.5">
			<geom type="sphere" size="0.05" rgba="0 0 1 1" />
			
			<site name="tower5_site" pos="0 0 0" size="0.01" />
		</body>
		
		<body name="tower6" pos="-1.25 2.165 2.5">
			<geom type="sphere" size="0.05" rgba="0.6 0 0.8 1" />
			
			<site name="tower6_site" pos="0 0 0" size="0.01" />
		</body>
		
		<body name="end_effector" pos="0 0 0.98">
			<joint name="free_joint" type="free" />
			<geom type="cylinder" size="0.21 0.1" rgba="0 0 1 1" mass="9" />
			
			<site name="rope_attach1" pos="-0.21 0 0" />
			<site name="rope_attach2" pos="-0.006 -0.014 0.21" />
			<site name="rope_attach3" pos="0.105 -0.182 0" />
			<site name="rope_attach4" pos="0.015 0.002 0.21" />
			<site name="rope_attach5" pos="0.105 0.182 0" />
			<site name="rope_attach6" pos="-0.009 0.012 0.21" />
		</body>
	</worldbody>
	
	<tendon>
		<spatial name="rope1" limited="true" range="0 3" width="0.01" >
			<site site="tower1_site" />
		
			<site site="rope_attach1" />
		</spatial>
		
		<spatial name="rope2" limited="true" range="0 3" width="0.01"  >
			<site site="tower2_site" />
			<site site="rope_attach2" />
		</spatial>
		<spatial name="rope3" limited="true" range="0 3" width="0.01"  >
			<site site="tower3_site" />
			<site site="rope_attach3" />
		</spatial>
		<spatial name="rope4" limited="true" range="0 3" width="0.01" >
			<site site="tower4_site" />
			<site site="rope_attach4" />
		</spatial>
		<spatial name="rope5" limited="true" range="0 3" width="0.01" >
			<site site="tower5_site" />
			<site site="rope_attach5" />
		</spatial>
		<spatial name="rope6" limited="true" range="0 3" width="0.01"  >
			<site site="tower6_site" />
			<site site="rope_attach6" />
		</spatial>
	</tendon>
	
	<actuator>
	<muscle name="motor_rope1" tendon="rope1" ctrlrange="0 10"/>
	<muscle name="motor_rope2" tendon="rope2" ctrlrange="0 10"/>
	<muscle name="motor_rope3" tendon="rope3" ctrlrange="0 10"/>
	<muscle name="motor_rope4" tendon="rope4" ctrlrange="0 10"/>
	<muscle name="motor_rope5" tendon="rope5" ctrlrange="0 10"/>
	<muscle name="motor_rope6" tendon="rope6" ctrlrange="0 10"/>
	
	</actuator>
	

	
	<sensor>
		
		<tendonlimitfrc name="rope1_force" tendon="rope1" />
		<tendonlimitfrc name="rope2_force" tendon="rope2" />
		<tendonlimitfrc name="rope3_force" tendon="rope3" />
		<tendonlimitfrc name="rope4_force" tendon="rope4" />
		<tendonlimitfrc name="rope5_force" tendon="rope5" />
		<tendonlimitfrc name="rope6_force" tendon="rope6" />
		<framepos name="end_effector_pos" objtype="body" objname="end_effector" />
		<tendonlimitpos name="rope1_pos" tendon="rope1" />
		<tendonlimitpos name="rope2_pos" tendon="rope2" />
		<tendonlimitpos name="rope3_pos" tendon="rope3" />
		<tendonlimitpos name="rope4_pos" tendon="rope4" />
		<tendonlimitpos name="rope5_pos" tendon="rope5" />
		<tendonlimitpos name="rope6_pos" tendon="rope6" />
	</sensor>
</mujoco>
`

ZnMnCr, 2025-04-25:
Hello, I am currently trying to model a cable-driven parallel robot and aim to determine the tension forces in the cables. I have simulated the cables using tendons, but I am unclear about what type of actuators to use for applying specific tension to the tendons and what sensors to employ for measuring the tension in the cables.

[fastCDPR.txt](https://github.com/user-attachments/files/19908088/fastCDPR.txt)

commit a2fd34ca (2024-07-17): Modify test/testdata/model.xml so the wrapping is stable. Fixes #1795

PiperOrigin-RevId: 653202132
Change-Id: I90ccff840a0b98d44dbee8888cb37ff5195caaa6