google-deepmind/mujoco #1797: Deformable elastic bodies

Shubh0802, 2024-07-10:
### Discussed in https://github.com/google-deepmind/mujoco/discussions/1789

<div type='discussions-op-text'>

<sup>Originally posted by **shubhamt2897** July  7, 2024</sup>
I have been trying to simulate objects like hoses which deform during robot's motion, in the default code damping and material properties is what I am looking into right now. I saw the old video of 2.3 for an elasticity plugin but couldn't find it . Can anyone direct me in the right direction, if they have worked on similar applications?</div>

Balint-H, 2024-07-11:
Hello, check out the following plugin's docs:
https://github.com/google-deepmind/mujoco/tree/main/plugin/elasticity

And here are some example models:
https://github.com/google-deepmind/mujoco/tree/main/model/plugin/elasticity

Shubh0802, 2024-07-12:
> Sie erhalten nicht oft eine E-Mail von ***@***.*** Erfahren Sie, warum dies wichtig ist <https://aka.ms/LearnAboutSenderIdentification>

Hello,

 

Thanksalot fort he help!

 

Mit freundlichen Grüßen/Best regards

Shubham Pradip Tangadpalliwar

Dürr Systems AG
Application Technology
Planning/Simulation
Carl-Benz-Str. 34
74321 Bietigheim-Bissingen 
Germany

MS Teams  ***@***.***> Chat/Call
E-Mail  ***@***.***> ***@***.***
Internet www.durr.com 

Vorsitzender des Aufsichtsrats: Dietmar Heinrich
Vorstand: Dr. Jochen Weyrauch (Vors.), Jaroslaw Baginski, Dr. Lars Friedrich, Dr. Hannes Schmüser, Bruno Welsch
Sitz der Gesellschaft: Stuttgart; eingetragen im Amtsgericht Stuttgart HRB 757705 

Datenschutzhinweise:  <https://www.durr.com/de/datenschutz> hier klicken  // Privacy Notice:  <https://www.durr.com/en/data-protection> click here

Dürr & More - aktuelle Informationen über unsere Produkte und Serviceleistungen in unserem Onlinemagazin:  <https://www.durr-group.com/de/duerrmore/> hier klicken
Dürr & More - current information about our products and services in our online magazine:  <https://www.durr-group.com/en/duerrmore/> click here
   
 Bitte denken Sie an die Umwelt, bevor Sie diese E-Mail drucken / Please think about our Environment before printing this e-mail 

https://github.com/google-deepmind/mujoco/tree/main/plugin/elasticity

https://github.com/google-deepmind/mujoco/tree/main/model/plugin/elasticity

closed 2024-07-12 by quagla