google-deepmind/mujoco #1798: `def_map` is not copied correctly when copying a mjCModel

JewelryForge, 2024-07-10:
A bug is introduced by revision [Remove mjCBase::def and use mjCBase::classname instead](https://github.com/google-deepmind/mujoco/commit/e0d33eb1d08c5030adaed8873af1b92e0aee92e7). The `def_map` has only a key "main", ignoring the keys of the copied `def_map`.

https://github.com/google-deepmind/mujoco/blob/e0d33eb1d08c5030adaed8873af1b92e0aee92e7/src/user/user_model.cc#L151

This bug can be replicated with the following code
```c++
  const char *filename = "mujoco_menagerie/unitree_go1/go1.xml";
  char error[1000];
  auto *spec0 = mj_parseXML(argv[1], nullptr, error, 1000);
  auto *spec1 = mj_copySpec(spec0);
  mj_compile(spec1, nullptr);
```

quagla, 2024-07-10:
Any idea why RecompileCompare in user_api_test didn't pick up the bug? I would have expected to do what the MRE above does.

JewelryForge, 2024-07-10:
I'm not sure yet. If the "mujoco_menagerie/unitree_go1" folder is put in "mujoco/model", it does report a segmentation fault.

quagla, 2024-07-10:
Well done, thanks! I am still a bit surprised it requires a default for the assets while it works for other defaults?

closed 2024-07-10 by copybara-service[bot]

JewelryForge, 2024-07-11:
@quagla I'm happy that my pr has been accepted! Currently, I find that your revision [Add attach tag to MJCF](https://github.com/google-deepmind/mujoco/commit/27b9ddda5bdd12345abdfbc0a6efce121d40cd6c) has correctly handled def_map and now my code is redundant.

commit 5bd13b49 (2024-07-10): Copy all keys of def_map when copying a mjCModel. Fixes #1798