google-deepmind/mujoco #1801: Where is the reference point of `fullinertia` in `body/inertia`?

barikata1984, 2024-07-10:
Hi,

I'm a student and I'm trying to use MuJoCo for rigid body dynamics simulation. I'm looking for some help with how to set `fullinertia` in `body/inertia`. 

I have checked [the documentation](https://mujoco.readthedocs.io/en/stable/XMLreference.html#body-inertial). First of all, the inertial frame of a body is noted that 
> The inertial frame is such that its center coincides with the center of mass of the body

Next, `diaginertia` is described that 
> Diagonal inertia matrix, expressing the body inertia relative to the inertial frame. 

Okay, it's clear and straightforward. Then, `fullinertia` is explained that 
> Full inertia matrix M. Since M is 3-by-3 and symmetric, it is specified using only 6 numbers in the following order: M(1,1), M(2,2), M(3,3), M(1,2), M(1,3), M(2,3). The compiler computes the eigenvalue decomposition of M and sets the frame orientation and diagonal inertia accordingly. If non-positive eigenvalue

As you can see, there is no description about where is the reference point of the full inertia matrix. 

As you may know, if that point is expected to be the center of mass, non-diagonal elements should not appear in the inertia matrix, and that is exactly the case for `diaginertia`. The `fullinertia` attribute does not make sense in this case. I believe another point would be the reference point for 'fullinertia'. But I cannot find where it is.

Actually, the values for `fullinertia` I am trying to set are defined relative to the origin of the body frame. Can I recognize that the body origin is the reference point of `fullinertia` and set my values to that attribute? Besides, how should I set the `pos` parameter in this case? The position of the center of mass with reference to the body origin?

Thanks in advance!

yuvaltassa, 2024-07-10:
It is with respect to `body/inertial/pos`. I'm almost sure of this but you should be able to verify with some simple experiments.

> As you may know, if that point is expected to be the center of mass, non-diagonal elements should not appear in the inertia matrix

Is this true? I don't think it's true. It's true if the frames are also aligned, but it's not sufficient that the position is the same.

closed 2024-07-10 by yuvaltassa

barikata1984, 2024-07-11:
Oh, I see. My Bad. Thanks for pointing out my misunderstanding