google-deepmind/mujoco #1805: Difference between using motor and position actuators in mujoco

Yingfan99327, 2024-07-11:
Hi,

I'm a student and I'm trying to use MuJoCo for robotic simulation

I'm looking for some help with the actutaor settings used in mujoco.

I want to implement some learning algorithm in panda robot in mujoco, the thing is that if I set the actuators of the panda robot by setting the type as "position" as follows, if I load the model into mujoco.viewer, the robot will stay at its initial configuation.

<actuator>
        <position name="panda0_joint1" joint="panda0_joint1" class="panda" kp="870" forcerange="-87 87" ctrlrange="-2.9671 2.9671"/> 
        <position name="panda0_joint2" joint="panda0_joint2" class="panda" kp="870" forcerange="-87 87" ctrlrange="-1.8326 1.8326"/> 
        <position name="panda0_joint3" joint="panda0_joint3" class="panda" kp="870" forcerange="-87 87" ctrlrange="-2.9671 2.9671"/> 
        <position name="panda0_joint4" joint="panda0_joint4" class="panda" kp="870" forcerange="-87 87" ctrlrange="-3.1416 0.0"/> 
        <position name="panda0_joint5" joint="panda0_joint5" class="panda" kp="120" forcerange="-12 12" ctrlrange="-2.9671 2.9671"/> 
        <position name="panda0_joint6" joint="panda0_joint6" class="panda" kp="120" forcerange="-12 12" ctrlrange="-1.66 2.1817"/> 
        <position name="panda0_joint7" joint="panda0_joint7" class="panda" kp="120" forcerange="-12 12" ctrlrange="-2.9671 2.9671"/> 
    </actuator>

however, if I set the actuators of the robot with the type "motor" as follows, if I load the model into mujoco.viewer, the robot will fall down automatically which I guess I need to do some gravity compensation.

<actuator>
        <motor name='panda_motor1' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint1' gear="1"/>
        <motor name='panda_motor2' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint2' gear="1"/>
        <motor name='panda_motor3' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint3' gear="1"/>
        <motor name='panda_motor4' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint4' gear="1"/>
        <motor name='panda_motor5' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint5' gear="1"/>
        <motor name='panda_motor6' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint6' gear="1"/>
        <motor name='panda_motor7' ctrllimited="true" ctrlrange="-500 500" joint='panda0_joint7' gear="1"/>
 </actuator>

Because I hope to set the robot at some initial configuration in the beginning of each trainning round by using the reset() function, I don't want to see the robot falling down automatically. So, may I ask how to solve this problem if I set the actuator type as "motor"?

And another question is about my understanding of the mujoco.data.ctrl, if I use the "position"-type actuator, does it mean that every time I call mj_step function, the control signal in data.ctrl should be the (joint) position command, if I use the "motor"-type actuator, the control signal used for data.ctrl should be the (joint) torque command?

Thanks very much and wishing you all the best.

Yingfan99327, 2024-07-11:
And I just tried to add gravcomp ton each body, I found that if I set gravcomp=1 for each link body, the robot will also fall down, however, if I set gravcomp=5 or higher, it seems to work a bit but also a bit strange compared to using "position"-type actuators.

yuvaltassa, 2024-07-12:
The answer to all your questions is "yes"

closed 2024-07-12 by yuvaltassa

Yingfan99327, 2024-07-18:
Hi, Yuval,

Thanks so much for your reply!

Just one more question about the actuator type in mujoco.

From the official document, I noticed that different actuator types could be combined together and defined in the xml files. And I noticed from the official document that, it says that _“Note that in order create a PD controller, one has to define two actuators: a position servo and a velocity servo. This is because MuJoCo actuators are SISO while a PD controller takes two control inputs (reference position and reference velocity). ”_  I want to check whether I understand this sentence in a right way.

My understanding is as follows,

If I defined the actuator type actuator type for one joint like this, namely, I incorporated three kinds of actuator types in one joint, does it mean that I have two ways of sending joint torque signals to each joint to control the robot based on PD control? Here are my questions as follows,

```
<actuator>
        <position name="position_servo1" joint="panda0_joint1" class="panda" kp="870" forcerange="-87 87" ctrlrange="-2.8973 2.8973"/> <!-- velocity="2.1750" -->
        <velocity name="velocity_servo1" joint="panda0_joint1" class="panda" kv="100" forcerange="-87 87" ctrlrange="-2.8973 2.8973"/>
        <motor name='motor1' ctrllimited="true" ctrlrange="-87 87" joint='panda0_joint1' gear="1"/>-->
<actuator>
```

1) The first one is that I can define the kp, kd matrix on my source code and then caculate the joint torque with q_pos_ref and q_vel_ref and then use data.ctrl = torque to control the robot, which to some extend means that only the "motor"-type actuator works.

2) The second one is that I found mujoco already has realized a PD controller internally and what I need to do is to set the kp and kv parameters for the "position"-type and "velocity"-type in the xml file, however, how should I set the data.ctrl? I am not very clear about how to feed two inputs into the "position-velocity" type actuator, maybe like this? If I understand this right, could you tell me how to define the data.ctrl if I use this kind of "position-velocity" type actuator

```
data.ctrl = qpos_ref - data.qpos + qvel_ref - data.qvel 
```
How does mujoco separates these the position errors or velocity erros and then calculate the torque?

3) Which option is better, I mean, simply use the motor-type actuator and define the Kp and Kv on my own to calculate the joint torque or use the position-velocity type actuator and set kp and kv in the xml files?

Thanks very much for your time to answer my "rookie" questions about mujoco and wishing you all the best.

Yingfan99327, 2024-07-18:
I guess I know the answer for the second question, because once I add a actuator, the size of data.ctrl would also increase, hahah, I just need to define two groups of actuators for the robot, namely, one group all position-type actuators, and one group velocity-type actuators as follows,
```
data.ctrl[0:7] = qpos_ref
data.ctrl[7:14] = qvel_ref
```
right?

And what about question 3, which is better

yuvaltassa, 2024-07-18:
Regarding Q 3: up to you, they are equivalent. Maybe try both, just to see that you understand what is happening.

The only real difference is that for velocity-type actuators (and damping via position kv), you'll get better integration with implicit or implicitfast.

Read this: https://mujoco.readthedocs.io/en/latest/computation/index.html#numerical-integration

Yingfan99327, 2024-07-19:
Thanks very much for your reply! I got it!

Wishing you all the best.