google-deepmind/mujoco #1806: Frames rendered from MJX-derived MjData objects are darker than they should be

7oponaut, 2024-07-12:
Frames rendered from MJX-derived MjData objects are darker than they should be.

The issue is present if `mujoco.mj_forward()` is not called on an `MjData` object before using it for rendering.

There are two workarounds:
- call `mujoco.mj_forward()` on the `MjData` object before rendering the first frame and read data into it after each `mjx.step()` using `mjx.get_data_into()`
- after each `mjx.step()`, use `mjx.get_data()` to move data to the CPU, followed by `mujoco.mj_forward()` on the resulting `MjData` object before using it to produce the next frame

The issue is reproduced in the [MJX tutorial notebook](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb#scrollTo=Pr29xq0-JRQv): the first video in the tutorial is made using CPU-based simulation, and the second video is made using MJX. The second video is lit differently.

Depending on whether this is intended behavior, this is either a mujoco bug or a bug in the MJX tutorial notebook.

CPU video

https://github.com/user-attachments/assets/90836789-c64d-4816-9990-34236b386395

MJX video without `mujoco.mj_forward()`

https://github.com/user-attachments/assets/af2b8bef-d357-44a9-bb03-21d8eea02c60

MJX video with `mujoco.mj_forward()`

https://github.com/user-attachments/assets/a7b82332-ec8f-4ac3-8484-9e21fc530019

I made the videos using this code, derived from the MJX tutorial notebook:
```
import cv2
import jax
import mujoco
import mujoco.mjx as mjx

xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""

width, height = 320, 240
duration = 3.8  # (seconds)
framerate = 60  # (Hz)

mj_model = mujoco.MjModel.from_xml_string(xml)

# enable joint visualization option:
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True

codec = cv2.VideoWriter_fourcc(*"mp4v")

# simulate on the CPU
renderer = mujoco.Renderer(mj_model, height=height, width=width)
video_writer = cv2.VideoWriter("mj.mp4", codec, framerate, (width, height))

mj_data = mujoco.MjData(mj_model)
mujoco.mj_resetData(mj_model, mj_data)

frame_count = 0
while mj_data.time < duration:
    mujoco.mj_step(mj_model, mj_data)
    if frame_count < mj_data.time * framerate:
        renderer.update_scene(mj_data, scene_option=scene_option)
        frame = renderer.render()
        video_writer.write(frame)
        frame_count += 1

video_writer.release()
renderer.close()

# simulate using MJX
for i in range(2):
    renderer = mujoco.Renderer(mj_model, height=height, width=width)
    video_writer = cv2.VideoWriter(f"mjx{i}.mp4", codec, framerate, (width, height))

    mj_data = mujoco.MjData(mj_model)
    mujoco.mj_resetData(mj_model, mj_data)
    mjx_model = mjx.put_model(mj_model)
    mjx_data = mjx.put_data(mj_model, mj_data)
    if i:
        mujoco.mj_forward(mj_model, mj_data)

    jit_step = jax.jit(mjx.step)

    frame_count = 0
    while mjx_data.time < duration:
        mjx_data = jit_step(mjx_model, mjx_data)
        if frame_count < mjx_data.time * framerate:
            mjx.get_data_into(mj_data, mj_model, mjx_data)
            renderer.update_scene(mj_data, scene_option=scene_option)
            frame = renderer.render()
            video_writer.write(frame)
            frame_count += 1

    video_writer.release()
    renderer.close()
```

OS: Arch Linux x86_64
Kernel: 6.9.7-arch1-1
mujoco version: 3.1.6

Also reproducible by running the MJX tutorial notebook on colab

yuvaltassa, 2024-07-12:
IIRC correctly, MJX does not have the equivalent of mujoco.mj_camlight, so the lights are not being updated. Can you confirm by calling only that function (instead of all of mj_forward)?

cc @erikfrey

7oponaut, 2024-07-12:
Tested, calling `mujoco.mj_camlight()` only does not fix the issue.

`mujoco.mj_camlight()`

https://github.com/user-attachments/assets/3f86ab24-f0a4-4883-952e-df5ac388ae82

`mujoco.mj_forward()`

https://github.com/user-attachments/assets/09862d4a-4acc-472a-957a-b35193ac5815

Code
```
import cv2
import jax
import mujoco
import mujoco.mjx as mjx

xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""

width, height = 320, 240
duration = 3.8  # (seconds)
framerate = 60  # (Hz)

mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)

codec = cv2.VideoWriter_fourcc(*"mp4v")

# enable joint visualization option:
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True

codec = cv2.VideoWriter_fourcc(*"mp4v")

# simulate using MJX
for i in range(2):
    renderer = mujoco.Renderer(mj_model, height=height, width=width)
    video_writer = cv2.VideoWriter(f"mjx{i}_.mp4", codec, framerate, (width, height))

    mj_data = mujoco.MjData(mj_model)
    mujoco.mj_resetData(mj_model, mj_data)
    mjx_model = mjx.put_model(mj_model)
    mjx_data = mjx.put_data(mj_model, mj_data)
    match i:
        case 0:
            mujoco.mj_camlight(mj_model, mj_data)
        case 1:
            mujoco.mj_forward(mj_model, mj_data)

    jit_step = jax.jit(mjx.step)

    frame_count = 0
    while mjx_data.time < duration:
        mjx_data = jit_step(mjx_model, mjx_data)
        if frame_count < mjx_data.time * framerate:
            mjx.get_data_into(mj_data, mj_model, mjx_data)
            renderer.update_scene(mj_data, scene_option=scene_option)
            frame = renderer.render()
            video_writer.write(frame)
            frame_count += 1

    video_writer.release()
    renderer.close()
```

yuvaltassa, 2024-07-12:
You need `mj_kinematics` followed by `mj_camlight` if you want to call it where you call it.

**But** this is the wrong place to call it. You're basically setting the lights in the `mj_data` once. If you were to attach the light to a moving body it would never move, right? If you want to do the "right" thing (given the current MJX limitation of not computing camera and light poses), it should be enough to call `mj_camlight` (no kinematics) after `get_data_into` but before `update_scene`.

Please confirm that the above is correct. 

Moving forward, I think MJX should compute light and camera poses at some point, but int the meantime the above solution should be fine. `mj_camlight` is super cheap.

BTW [this model](https://youtu.be/cE3s_IfO4g4) (from the introductory tutorial) has some moving lights I believe.

7oponaut, 2024-07-12:
Calling `mj_camlight` where you mentioned works when tested with a modified version of my code.

I will do a test on the tutorial scene you linked later to check light source movement.

I tested with this:
```
import cv2
import jax
import mujoco
import mujoco.mjx as mjx

xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""

width, height = 320, 240
duration = 3.8  # (seconds)
framerate = 60  # (Hz)

mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)

codec = cv2.VideoWriter_fourcc(*"mp4v")

# enable joint visualization option:
scene_option = mujoco.MjvOption()
scene_option.flags[mujoco.mjtVisFlag.mjVIS_JOINT] = True

codec = cv2.VideoWriter_fourcc(*"mp4v")

jit_step = jax.jit(mjx.step)

# simulate using MJX
for i in range(3):
    renderer = mujoco.Renderer(mj_model, height=height, width=width)
    video_writer = cv2.VideoWriter(f"mjx{i}_.mp4", codec, framerate, (width, height))

    mj_data = mujoco.MjData(mj_model)
    mujoco.mj_resetData(mj_model, mj_data)
    mjx_model = mjx.put_model(mj_model)
    mjx_data = mjx.put_data(mj_model, mj_data)

    frame_count = 0
    while mjx_data.time < duration:
        mjx_data = jit_step(mjx_model, mjx_data)
        if frame_count < mjx_data.time * framerate:
            mjx.get_data_into(mj_data, mj_model, mjx_data)
            match i:
                case 0:
                    pass
                case 1:
                    mujoco.mj_camlight(mj_model, mj_data)
                case 2:
                    mujoco.mj_forward(mj_model, mj_data)

            renderer.update_scene(mj_data, scene_option=scene_option)
            frame = renderer.render()
            video_writer.write(frame)
            frame_count += 1

    video_writer.release()
    renderer.close()
```

closed 2024-07-14 by yuvaltassa