google-deepmind/mujoco #1808: Numpy==2 breaks cam.lookat array item-assignment (stride=0)

joshuasmith44, 2024-07-13:
OS: (Asahi Linux Ubuntu 22.04)
Mujoco: 3.0.0

Found that when changing the lookat array in the MjvCamera object updating a single element would update all elements in the array possibly because array stride was set to 0.
Minimal Example (same conda env except numpy install is either 1.26 or 2.0):

(broken)
(mjtest) ubuntu@localhost:~/Research/mujoco_test$ python
Python 3.10.14 (main, May  6 2024, 19:36:58) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco as mj
>>> cam = mj.MjvCamera()
>>> cam
<MjvCamera
  azimuth: 90.0
  distance: 2.0
  elevation: -45.0
  fixedcamid: -1
  lookat: array([0., 0., 0.])
  trackbodyid: -1
  type: 0
>
>>> cam.lookat[0] = 1
>>> cam
<MjvCamera
  azimuth: 90.0
  distance: 2.0
  elevation: -45.0
  fixedcamid: -1
  lookat: array([1., 1., 1.])
  trackbodyid: -1
  type: 0
>
>>> cam.lookat.__array_interface__
{'data': (533993184, False), 'strides': (0,), 'descr': [('', '<f8')], 'typestr': '<f8', 'shape': (3,), 'version': 3}
>>> import numpy as np
>>> np.__version__
'2.0.0'
________________________________________________________________________________________________
(working)
(mjtest) ubuntu@localhost:~/Research/mujoco_test$ python
Python 3.10.14 (main, May  6 2024, 19:36:58) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mujoco as mj
>>> cam = mj.MjvCamera()
>>> cam
<MjvCamera
  azimuth: 90.0
  distance: 2.0
  elevation: -45.0
  fixedcamid: -1
  lookat: array([0., 0., 0.])
  trackbodyid: -1
  type: 0
>
>>> cam.lookat[0] = 1
>>> cam
<MjvCamera
  azimuth: 90.0
  distance: 2.0
  elevation: -45.0
  fixedcamid: -1
  lookat: array([1., 0., 0.])
  trackbodyid: -1
  type: 0
>
>>> cam.lookat.__array_interface__
{'data': (710077008, False), 'strides': None, 'descr': [('', '<f8')], 'typestr': '<f8', 'shape': (3,), 'version': 3}
>>> import numpy as np
>>> np.__version__
'1.26.0'

yuvaltassa, 2024-07-14:
Sorry, I can't reproduce.

The last cell in the introductory tutorial makes [this video](https://youtu.be/cE3s_IfO4g4), right?

yuvaltassa, 2024-07-14:
Maybe see if you can reproduce with latest MuJoCo?
If not, then I suppose it's been fixed 🙂

yuvaltassa, 2024-07-14:
Oh hold on, this is specifically for numpy version 2.0?

Looking at our [requirements.txt](https://github.com/google-deepmind/mujoco/blob/22a10fd2ffedad2de43eeeb1da5670db56b9d3a8/python/build_requirements.txt#L17), it doesn't look like 2.0 is supported yet, though perhaps we should be forward fixing something here...

Reopening. Someone more knowledgeable than me will take a look.

yuvaltassa, 2024-07-14:
I did a bit of reading on this and looks like numpy 2.0 (released June 2024) has lots of breaking changes ([example](https://numpy.org/devdocs/numpy_2_0_migration_guide.html#the-pyarray-descr-struct-has-been-changed)). 
IIUC from reading internal discussions, numpy 2.0 support is unlikely to happen in the near future.
We will update this issue if/when that is planned.
In the meantime, please downgrade to 1.x
Re-closing.

closed 2024-07-14 by yuvaltassa

joshuasmith44, 2024-07-14:
I tested this a bit further. I replicated the issue using Numpy 2.0.0 and Mujoco 3.1.3. Updating to the next mujoco 3.1.4 fixed the issue. I would hazard a guess that this has to do with the pybind version being updated to 2.12 from 2.11.1.
Changing commit: https://github.com/google-deepmind/mujoco/commit/930d7b0487dd9d3c06b480c74f289d5859db11e3 
PyBind 2.11.1 not supporting Numpy 2.0: https://github.com/pybind/pybind11/issues/5009

joshuasmith44, 2024-07-14:
Only remaining concern is that if I install just doing "pip install mujoco==3.1.3" it installs numpy 2.0.0 and this issue remains. I am stuck on older mujoco versions unfortunately :(

traversaro, 2024-07-16:
> I did a bit of reading on this and looks like numpy 2.0 (released June 2024) has lots of breaking changes [...]

What is exactly the failure mode with numpy 2? I just installed pip install mujoco==3.2.0 numpy==2 and running with `pytest --pyargs mujoco` all the tests seems to pass without problems:

~~~
(mujocopy312) C:\Users\icub>pip list
Package             Version
------------------- --------
absl-py             2.1.0
colorama            0.4.6
etils               1.9.2
exceptiongroup      1.2.2
fsspec              2024.6.1
glfw                2.7.0
importlib_resources 6.4.0
iniconfig           2.0.0
mujoco              3.2.0
numpy               2.0.0
packaging           24.1
pip                 24.0
pluggy              1.5.0
PyOpenGL            3.1.7
pytest              8.2.2
setuptools          70.3.0
tomli               2.0.1
typing_extensions   4.12.2
wheel               0.43.0
zipp                3.19.2

(mujocopy312) C:\Users\icub>pytest --pyargs mujoco
================================================= test session starts =================================================
platform win32 -- Python 3.12.4, pytest-8.2.2, pluggy-1.5.0
rootdir: C:\Users\icub
collected 170 items

bindings_test.py ............................................................................................... [ 55%]
...........                                                                                                      [ 62%]
memory_leak_test.py .                                                                                            [ 62%]
minimize_test.py ........                                                                                        [ 67%]
msh2obj_test.py .                                                                                                [ 68%]
render_test.py ...                                                                                               [ 70%]
renderer_test.py ......                                                                                          [ 73%]
rollout_test.py ....................................                                                             [ 94%]
specs_test.py .......                                                                                            [ 98%]
usd\exporter_test.py .                                                                                           [ 99%]
viewer_test.py .                                                                                                 [100%]

================================================= 170 passed in 8.41s =================================================
WARNING: Nan, Inf or huge value in QPOS at DOF 0. The simulation is unstable. Time = 0.0000.

WARNING: Nan, Inf or huge value in QACC at DOF 3. The simulation is unstable. Time = 14.4380.

WARNING: Nan, Inf or huge value in QACC at DOF 3. The simulation is unstable. Time = 14.4380.
~~~

JoshMSmith44, 2024-07-16:
In my tests, using numpy==2 and mujoco==3.1.3 and earlier, then updating a single element in cam.lookat (MjvCamera) updates all elements to that value. Also, runing pip install mujoco==3.1.3 installs/allows numpy==2.

JoshMSmith44, 2024-07-16:
Doesn't seem to be an issue with recent mujoco versions, I just think pip package should be configured such that pip install mujoco<=3.1.3 should only allow numpy<2

traversaro, 2024-07-16:
Yes, sorry for the confusion, by question was for recent mujoco versions (I tested 3.2.0), not for older versions.

traversaro, 2024-09-04:
From what I understand investigating a bit more (but I may be missing something) mujoco is not using `PyArray_Descr` directly, only via pybind11. pybind11 2.12.* supports NumPy 2.0, while pybind11 2.11.* does not support numpy 2.0 (see https://github.com/pybind/pybind11/issues/5009). So it all boils down to the pybind11 version used by mujoco (as long as `MUJOCO_PYTHON_USE_SYSTEM_PYBIND11` is not used).

mujoco 3.1.3 used pybind11 2.11.1 (see https://github.com/google-deepmind/mujoco/blob/3.1.3/python/mujoco/CMakeLists.txt#L194), while mujoco 3.1.4 used pybind11 2.12.0, meaning that mujoco <= 3.1.3 does not work with numpy 2. This is coherent with what was found by @JoshMSmith44 .

> Doesn't seem to be an issue with recent mujoco versions, I just think pip package should be configured such that pip install mujoco<=3.1.3 should only allow numpy<2

I am not an expert on PyPI, but I do not think you can change the metadata for already published packages (see https://github.com/pypa/packaging-problems/issues/97#issuecomment-372516469), only publish new packages, i.e. something like 3.1.3.1 .

nimrod-gileadi, 2024-09-04:
Thanks, @traversaro, that's right.