google-deepmind/mujoco #1810: unpin a flexcomp at runtime

xkiixkii, 2024-07-15:
Hi,

I'm a student and I'm trying to use MuJoCo for creating a gymnasium environment for grasping deformable objects.

I'm looking for some help with unpinning flexcomp at runtime. 

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="insole">
	<compiler strippath="true" meshdir="meshes/" />
	<extension>
		<plugin plugin="mujoco.elasticity.solid" />
	</extension>
	<worldbody>
		<light pos="0.3 0 1.5" dir="0 0 -1" directional="true" />
		<!-- <geom name="floor" pos="0 0 -0.1" size="0 0 0.05" type="plane"/> -->
		<body name="insole">
			<flexcomp name="insole" pos=".12 -.05 .462" mass="1" type="gmsh" file="insole.msh" rgba=".1 .9 .1 1">
				<edge equality="true" />
				<pin id="22 25 27 26" />
				<plugin plugin="mujoco.elasticity.solid">
					<config key="young" value="200000.0" />
					<config key="poisson" value="0.2" />
					<config key="damping" value=".005" />
				</plugin>
			</flexcomp>
		</body>
	</worldbody>
</mujoco>

```

</details>

In this environment, a robot arm is used to grab a deformable insole. A few vertices in the latter part of the insole is initially fixed on a platform. After a certain simulation time, the platform is removed and the insole is allowed to fall freely for a while.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                               
I would like to ask if there is any way to unpin these vertices after the platform is removed at runtime. 

Here is a screenshot of the current environment:
[image]

MuJoCo Version: 3.1.6
OS: Ubuntu 22.04.4 x86_64
Using Python binding

quagla, 2024-07-16:
Hi, you can't really unpin since pins prevent the creation of DOFs during model compilation. I can see two ways to achieve this:

- Not pinning to start with, but having equality constraints that you switch off
- Use the new model editing API and change the flex from pinned to unpinned in mjsFlex, then call mj_recompile

xkiixkii, 2024-07-17:
Thank you for your response, that helps a lot.
I have tried the first approach that works pretty good for my application now.
As for the second one, in [mjsFlex](https://github.com/google-deepmind/mujoco/blob/a2fd34caee777f55446e2dffc7f46e6f803ed1d6/include/mujoco/mjspec.h#L388) I could not find the pinning property.

quagla, 2024-07-23:
You are right, pinning is an option of FlexComp, so it's not part of mjsFlex. The equivalent operation would be to remove the joint from the bodies that correspond to the pinned vertices.

closed 2024-07-23 by quagla