google-deepmind/mujoco #1814: How to incoporate Motion Capture(MoCap) data ground truth into the system?

coder-rakeshpaul, 2024-07-15:
Hi,

I want to add the MoCap data ground truth to my environment. Similar to what has been incorporated in the[ dm_control environment. ](https://github.com/google-deepmind/dm_control/tree/main/dm_control/locomotion/mocap)Where they have added to ground truth of the CMU MoCap data. 

I find that they download the data from the google server and not sure how they actually mapped the model. 

When we import the cmu_humanoid xml into the environment. We see that the model is floating in the air, so mapping any the joint angles the model does the motion (but it's feet is always above the ground)
[image]

I want the feet to be on the ground so as to get the proper ground truth before training a RL policy. 

How do I go about doing this?

PS: I don't just want to use the CMU Mocap data and want to add other mocap data hence not using the already existing ground truth of the CMU Mocap.

yuvaltassa, 2024-07-16:
`data.qpos[2] -= height_offset` ?

coder-rakeshpaul, 2024-07-16:
yes that was my 1st thought too, but what if at the start of the motion the person is not on his feet rather on his toes? This will cause the foot to go below the ground (since I was talking about a dataset there would be multiple clips and a common offset would have been nice), yes this seems like a specific case and could be manually tuned.

Do you think the dataset in dm_control's MoCap was done the same way (with the height offset you mentioned)? or could they have used a better more efficient method (which I maybe not aware off)?

Also another question, how do I find out how far above the ground the feet are? (to get the height offset) is there a function or something which gives the location of the feet wrt ground?

Balint-H, 2024-07-18:
You can do a preprocessing step checking the height of the toes and heel throughout the animation. Then, subtract the minimum of that from the root height (+- some fixed offset to account for the foot thickness)